pimoroni/hyperpixel4 #2: GPIO Pinout

charlieprynn, 2018-07-24:
I'm looking to use this screen, as well as some other GPIO pins.

Is this possible? If so, do you have a pinout of how to wire it up manually.

Thanks

LongLane, 2018-07-26:
Would be good to get this header included in the package: https://shop.pimoroni.com/products/2x20-pin-gpio-header-for-raspberry-pi-2-b-a?variant=1132812269

LongLane, 2018-07-26:
Regarding GPIO Pinout, see this link: https://pinout.xyz/pinout/hyperpixel
Looks like the HyperPixel uses all available GPIO Pins, hence the inclusion of the alternate I2C bus!

charlieprynn, 2018-07-26:
@LongLane Thanks for the reply. I wondered if the 4inch display might differ to the 3.5 that is listed on pinout.xyz.

I also have the 4 inch non-touch, so I assumed that it might use fewer GPIO pins.

Does that mean its impossible to connect a Teensy board too?

stuartm, 2018-07-27:
According to https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-hyperpixel-4 (last section) there are no GPIO pins available.

Gadgetoid, 2018-07-27:
Without the touch overlay it's possible you could reclaim BCM9 and BCM10, but BCM18, BCM26 and BCM27 are used for SPI communications to initialise the display and BCM19 controls the backlight.

By default BCM9 and BCM10 are set up as an alternate i2c bus- normally i2c3- and used to communicate with the touch overlay. You can use this i2c bus for other i2c devices.

If you need multiple GPIO pins (more than two) then a basic IO expander hooked to this i2s bus would be sufficient. An MCP23017 would, for example, give you 16 pins configurable as inputs/outputs.

Gadgetoid, 2018-09-06:
This is now documented on Pinout.xyz - https://pinout.xyz/pinout/hyperpixel4

Roaders, 2020-05-11:
Hi
I am very new to a lot of this stuff. I want to have a hyperpixel touchscreen connected to my pi zero but also want to be able to drive one or 2 relays using gpio. Reading this thread it seems that this is not possible by default as the hyperpixel uses all the gpio pins.

Am I correct in thinking that I can attach the display and use all the gpio pins but then attach an IO expander using the i2c interface and then drive replays from those pins? Do those pins appear as normal pins to software? i.e. I want to address them from node red.

Thanks

baesek, 2021-11-21:
> Without the touch overlay it's possible you could reclaim BCM10 and BCM11, but BCM18, BCM26 and BCM27 are used for SPI communications to initialise the display and BCM19 controls the backlight.
> 
> By default BCM10 and BCM11 are set up as an alternate i2c bus- normally i2c3- and used to communicate with the touch overlay. You can use this i2c bus for other i2c devices.
> 
> If you need multiple GPIO pins (more than two) then a basic IO expander hooked to this i2s bus would be sufficient. An MCP23017 would, for example, give you 16 pins configurable as inputs/outputs.
> 
> Edit: Corrected i2c pins to BCM10 and BCM11- thanks @sandyjmacdonald

How could I reclaim BCM10 or BCM11 to to use them as a power button pin for the raspberry pi ? I tried both already by putting the following in /boot/config.txt, but nothing happend when i connected BCM11 to GND, same for 10.
`dtoverlay=gpio-shutdown,gpio_pin=11,active_low=1,gpio_pull=up`

baesek, 2021-11-23:
I just found this page https://forums.pimoroni.com/t/using-button-pcbs-with-the-hyperpixel-hat/5779/4
Does it mean that just removing the pin for gpio 11 and 10 from the pin header makes them available as gpio pins on the raspberry pi again?

closed 2026-08-25 by Gadgetoid