pimoroni/hyperpixel4 #3: Is it possible to disable backlight when shutdown?

stuartm, 2018-07-27:
When the Raspberry Pi is shutdown, but still connected to power, the backlight remains on. Is this behaviour configurable? Or should it be possible to toggle the backlight off by pulling the backlight  GPIO high/low during the shutdown process?

Any documentation that we can refer to?

Gadgetoid, 2018-07-27:
It's possible you could pull this pin using a technique similar to the one here: https://github.com/pimoroni/clean-shutdown/blob/master/daemon/lib/systemd/system-shutdown/gpio-poweroff

This drops a script into systemd that runs upon shutdown and asserts a GPIO pin.

Gadgetoid, 2018-10-12:
New, easier, method, add the following to `/boot/config.txt`:

```
dtoverlay=gpio-poweroff,gpiopin=19,active_low=1
```

Roaders, 2020-05-11:
Can I turn the screen off using software when the pi is still turned on? i.e. on timeout turn the screen off then turn it back on when someone walks in front of the screen?
I want my pi to be a control panel on the wall but not lit up all the time.

sjtoik, 2020-05-23:
I have similar project going on. I have the touch version, and would like tap the screen to wake it up and turn of with a timer.

sjtoik, 2020-05-24:
It seems that there is a supporting device tree function built in in the [pi4 branch](https://github.com/pimoroni/hyperpixel4/blob/pi4/src/hyperpixel4.dts), but the overlay is not loaded.

Running ``dtoverlay hyperpixel4`` drops error messages to ``dmesg`` as follows 

```
OF: overlay: find target, node: /fragment@4, phandle 0xcd not found
OF: overlay: init_overlay_changeset() failed, ret = -22
```

sjtoik, 2020-05-24:
I stand corrected, the above error happens possibly, because loading of overlays is a new thing. The debug log ``sudo vcdbg log msg`` shows no errors. The .dts indicates compatibility with gpio-backlight overlay. It also has references of gpio pin 19 being the control for this feature.

After installing the [python module](https://github.com/linusg/rpi-backlight) for the official 7" display, I quickly verified that the control works:
``` $ python3 
>>> from rpi_backlight import Backlight
>>> backlight = Backlight()
>>> backlight.power = False
>>> backlight.power = True
``` 

I don't know why this device is not responding to xset dkms commands, and what should be done to make it work, but this resolves to control issue for now. Next step is to hook it up to the screensaver functionality.

fquirin, 2020-10-03:
This was working for a long time:
`dtoverlay=gpio-poweroff,gpiopin=19,active_low=1`

But now I installed the newest Raspbian Lite (aka Raspberry Pi OS, August 2020) on the same RPi4 and the screen will stay black :-(

I've tried to but different things in a shutdown script located at `/etc/rc0.d` but the screen won't stay off. Following commands are working as long as the system is active:
- `raspi-gpio set 19 op dl`
- `DISPLAY=:0.0 xset dpms force off`
- `echo '1' > /sys/class/backlight/rpi_backlight/bl_power`
But at the very last state the backlight turns on again.

Any known issues with the latest RPi OS? Any solutions?

fquirin, 2020-10-05:
> The latest RPi OS - or rather the latest Linux kernel - seems rather more forceful about pins being mutually exclusive so I'm not surprised it's hit the backlight too.
> 
> The previous and rather contrived method of using a script in `/lib/systemd/system-shutdown/gpio-poweroff` should still work but it's far from optimal.
> 
> Hacking on the script I linked above, something like this will work: [...]

Thanks for the info.

I've created the script and put it into `/lib/systemd/system-shutdown/gpio-poweroff` (and made it executable if that's relevant), but ultimately the screen stays on. The usual behavior is:

`sudo shutdown -h now` -> screen turns black for ~3s -> screen turns back on and you see a bit of RPi console log -> RPi turns off and backlight stays on

Did I miss something? Do I have to install anything to make this script work?

Gadgetoid, 2020-10-06:
Okay upon further investigation the backlight *should* be turning off with no additional changes since there's a service in `/etc/systemd/system/poweroff.target.wants/rpi-display-backlight.service` that effectively runs `echo 1 | sudo tee -a /sys/class/backlight/rpi_backlight/bl_power` on shutdown.

It appears to shut down the backlight... but then it comes right back on again at the last stage. I'm guessing this is the point at which the `gpio-backlight` module is unloaded, releasing the GPIO back to its original state- which would be a high impedance input- and letting the pull-up on the Hyperpixel do its work so the backlight returns to "on." This would explain the weird backlight turning off and then on again on shutdown.

The only solution I can find is to hot-swap the module that owns the GPIO pin-

```
sudo rmmod gpio-backlight
sudo dtoverlay /boot/overlays/gpio-poweroff.dtbo gpiopin=19 active_low=1
```

This seems to allow the backlight to turn off and stay off. I'll try and throw together a systemd unit for this.

Gadgetoid, 2020-10-06:
This is pretty clunky, but create the following as `/etc/systemd/system/hyperpixel4-backlight.service`:

```
[Unit]
Description=Sets up gpio-poweroff to handle Hyperpixel backlight upon shutdown/reboot
ConditionPathExists=/usr/bin/hyperpixel4-init
ConditionPathExists=/boot/overlays/gpio-poweroff.dtbo
ConditionPathExists=/usr/bin/dtoverlay
DefaultDependencies=no
Before=umount.target

[Service]
Type=oneshot
ExecStart=/bin/sh -c '/sbin/rmmod gpio-backlight;/usr/bin/dtoverlay /boot/overlays/gpio-poweroff.dtbo gpiopin=19 active_low=1'

[Install]
WantedBy=reboot.target halt.target poweroff.target
```

And then run `sudo systemctl enable hyperpixel4-backlight.service`

fquirin, 2020-10-06:
> It appears to shut down the backlight... but then it comes right back on again at the last stage. I'm guessing this is the point at which the gpio-backlight module is unloaded, releasing the GPIO [...]

This pretty much sums what I had puzzled together I guess in a technically more precise way :sweat_smile: 

Its working with the `hyperpixel4-backlight.service` described above! Thanks a lot! :star_struck:

sjtoik, 2021-01-13:
I wrote a holiday project [hyperpixel4-blu](https://github.com/sjtoik/hyperpixel4) and embedded @Gadgetoid 's systemd service to be part of if. The package monitors XScreenSaver events and toggles the backlight on and off accordingly. You can find the .deb from the releases.

fhb, 2021-05-16:
This makes my b

> This is pretty clunky, but create the following as `/etc/systemd/system/hyperpixel4-backlight.service`:
>
> ```
> [Unit]
> Description=Sets up gpio-poweroff to handle Hyperpixel backlight upon [...]

This disables my backlight control in /etc/rc.local:
/usr/bin/gpio -g mode 19 pwm
/usr/bin/gpio -g pwm 19 40

Any way to dim the light with this service?

jeffThompson, 2022-05-05:
@fhb – using those commands worked great for me as part of a bash script too, thanks so much!

briwil, 2022-07-21:
> @fhb – using those commands worked great for me as part of a bash script too, thanks so much!
> 
> Edit: nope, both the service and the gpio commands still have the screen coming back on about three seconds after shutdown

Same here, still comes on after 3 seconds.

coollx, 2024-04-15:
> > @fhb – using those commands worked great for me as part of a bash script too, thanks so much!
> > Edit: nope, both the service and the gpio commands still have the screen coming back on about three seconds after shutdown
> 
> Same here, still comes back on after 3 seconds.

Same here. I encountered this issue today. Have you been able to solve it?

gwenhael-le-moine, 2024-10-27:
Here's a script that works for me:

```shell
#!/bin/bash

# set gpio 19 as output
pinctrl set 19 op

case $1 in
    "off")
	pinctrl set 19 dl
	;;
    "on")
	pinctrl set 19 dh
	;;
    *)
	echo "Usage: $0 on|off"
	;;
esac
```

closed 2026-08-25 by Gadgetoid