pimoroni/hyperpixel4 #6: Hyperpixel4 on aarch64 (RPi3+)

psnszsn, 2018-08-29:
I ran the install script, then compiled the hyperpixel4-init.c and hyperpixel4.dts files and placed them in the appropriate directories. The hyperpixel4-init.service seems to start, but the display remains blank.

Any ideas how to do this?

Gadgetoid, 2018-08-30:
I haven't attempted to get it working on aarch64 (or really much more than just booting aarch64 for attempts to compile rpi_ws281x), so I don't know the lay of the land.

Are you sure you're using a HyperPixel4 and not a HyperPixel?

You can attempt to run `hyperpixel4-init` directly, to see if it produces any errors. Generally it will fire out the SPI data whether the display is there or not, so it's not really helpful from a debugging perspective.

You should also check the contents of `/boot/config.txt` to make sure the installer has made the changes successfully. It might be worth pasting your `/boot/config.txt` here also.

psnszsn, 2018-08-30:
Yes, it is a HyperPixel4. The HyperPixel is not sold anymore as far as i know.
`hyperpixel4-init` runs, but it does not return any message. The exit code is `0`.
This is the content of `/boot/config.txt`:
```
dtoverlay=hyperpixel4
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
framebuffer_width=800
framebuffer_height=480
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
display_rotate=3
hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

Gadgetoid, 2019-11-07:
Sorry I'm a bit late - but (if you haven't got things up and running by now) could you post the contents of your `/boot/config.txt`? During my testing I've found that some interfaces being enabled will cause a blank screen. I wonder if this could be affecting you.

closed 2020-03-17 by Gadgetoid