pimoroni/hyperpixel4 #9: Combining the Hyper Pixel with a Pi DAC Pro?

LongLane, 2018-09-05:
I was wondering if it's possible to mount the Pi DAC Pro in-between a RPI and the HyperPixel display? It would create a very nice and compact streaming solution running e.g. Volumio.

Thanks

Gadgetoid, 2018-09-05:
Not possible I'm afraid, or at least not practical- you'd need some sort of USB-based audio IO device. HyperPixel uses all the pins.

For reference, here's what PCM uses: https://pinout.xyz/pinout/pcm

And HyperPixel (4 uses the same pinout as the original): https://pinout.xyz/pinout/hyperpixel

drewcovi, 2018-09-05:
Hypothetically, could this be used? http://www.ti.com/product/TLV320DAC3100-Q1 

Thanks for all the hard work @Gadgetoid. Really loving the hyperpixel4

LongLane, 2018-09-08:
Thanks for the clarification. Appreciate your good work!

closed 2018-09-08 by LongLane