pimoroni/hyperpixel4 #13: Touchscreen not working

trunksy, 2018-12-07:
Hi,
Just installed from the one line install - displays perfectly, but has no touchscreen capability.
Any ideas?
Thanks

Gadgetoid, 2019-08-29:
Sorry for the very delayed response- did you have any luck getting touch up and running?

malcolmholmes, 2019-09-15:
@Gadgetoid I'm in this situation. It *has* worked, but even with a completely fresh install, using `curl https://get.pimoroni.com/hyperpixel4 | bash`, the screen works, but no touch. Let me know if I can debug anything. This is a Hyperpixel4 on a RaspberryPi 4.

bolsoncerrado, 2019-09-24:
I'm on a Raspi3B+ with Jessie and my Touch is not working either, display is fine.

Previously I had the official RaspiTouch 7" display working fine, so not sure if there's some remmants anywhere that are deviating the touch functionality to the wrong device or something :?

Anyone?

lincsanders, 2019-09-25:
Just wanted to chime in that I am experiencing this too using the pi4 branch to install on a fresh Buster image. 

For what it's worth, my output when rotating the screen (rotation works perfectly) contains "unable to find device pointer:Goodix Capacitive TouchScreen"

bolsoncerrado, 2019-09-25:
Just tried a DIFFERENT HP4 on a DIFFERENT pi3B+ in a DIFFERENT SD (with Raspbian Jessie too) and the issue is the same so it must be something on the software...

bolsoncerrado, 2019-09-25:
@Gadgetoid are you copying?

bolsoncerrado, 2019-09-25:
OK

$ ls /dev/input
mice

Something’s f*cked up here…

Touch is not being loaded on bootup as it seems… (event0)

WHAT is it preventing it? IDK.

Gadgetoid, 2019-10-15:
Sorry for missing this- it's been a manic few.... weeks, months... years!?

I've just updated the Pi 3B+ support branch for the HyperPixel4 which may help get things smoothed out and working. I've tweaked a variety of things in the DTS file and backported the `hyperpixel4-rotate` utility from the Pi 4 branch to make it easier to rotate touch/LCD.

Additionally I've had some weird issues with HyperPixel4 on the Pi 3B+ versus the Pi 4 which I have, as of yet, been unable to diagnose. On each of these Pi's the touch IC appears with a completely different, yet wholly valid and functional i2c address.

On the Pi 3B+ it tends to appear with: `0x5d`
On the Pi 4 it tends to appear with: `0x14`

You can run `sudo i2cdetect -y 3` (the number 3 might differe depending on what you see with `ls /dev/i2c-*`) to confirm which i2c address your HyperPixel4 touch is using on your Pi.

My best guess is that some change of pin state during boot between the Pi 3B+ and Pi 4 is what's causing this disparity- the touch IC family does have a selectable i2c address and the two addresses are so wholly different that it's impossible they are a bit flip or similar glitch. Additionally it's sometimes possible to get the HyperPixel4 on the Pi 4 to show up with the other address- usually via hotplugging.

Up until now this difference has been pretty stable and I've maintained separate Pi 3B+ and Pi 4 branches in the repository to address it, but since I still don't know exactly *why* it's happening it's at least possible it could be causing obscure issues.

TLDR: Non-working touch could be an indication that your Pi is seeing the touch IC on a different address. Run: `sudo i2cdetect -y 3` to see which.

lincsanders, 2019-10-15:
@Gadgetoid I will give this a shot as soon as I get home and report back with findings. 

You are an absolute legend for the writeup and investigation, you have my undying gratitude 🙏

Gadgetoid, 2019-11-05:
I believe this issue ties in with #41 - but only for users having problems specifically with the HyperPixel 4 Rectangular on a Pi 4. It looks like there's a thermally-dependent race-condition during start-up that can change the i2c address on the Pi 4. I've proposed a software workaround and super simple resistor bodge for this here: https://github.com/pimoroni/hyperpixel4/issues/41#issuecomment-549759417

If you're having issues with Pi 3B+ or earlier, make sure to `ls /dev/i2c-*` and `sudo i2cdetect -y 7` or `sudo i2cdetect -y 3` depending on which bus you see. Knowing which Pi you're using and which i2c address you're seeing (if any) could let me know if I need to port this software workaround to the Pi 3B+ and earlier branch of installer also.

Gadgetoid, 2020-07-15:
Closing for now- but feel free to re-open if you have further findings.

closed 2020-07-15 by Gadgetoid