pimoroni/hyperpixel4 #18: Is there a way to disable touch?

farinspace, 2019-03-04:
During operation, phantom-touch event happens, moving the cursor, some times this behavior happens frequently where it interferes with typing.

Gadgetoid, 2019-03-05:
How is your hyperpixel set up? Is anything contacting the display, or could it be in an electrically noisy environment? I've not seen phantom touch events before so I'm inclined to believe it's an external problem.

The touch driver is configured by the "hyperpixel4" dtoverlay, so you can drop a `#` before `dtoverlay=hyperpixel4` in `/boot/config.txt` to turn it off completely. It shouldn't be generating phantom touch events, though.

petaflot, 2022-08-23:
you should also be able to use `xinput` to disable it

closed 2023-01-14 by farinspace