pimoroni/hyperpixel4 #20: Odroid (C2, UX4) Tinker Board S, Rock Pi 4 Support

cedrickzepp, 2019-04-01:
I have several of the SBC mentioned above, I would like to know if there is the possibility of compatibility of your beautiful screen with the others.https://guides.github.com/features/mastering-markdown/
[image]
[image]
[image]
[image]

mikekscholz, 2019-04-03:
I can’t comment on the driver side of things but these boards would have to support the alternate dpi pinout configuration, it changes the default functions of the pins. https://pinout.xyz/pinout/dpi

Gadgetoid, 2019-07-09:
I believe the chances of first-party support on anything but the Pi are basically zero. Ignoring the DPI outputs potentially being non-existent on other boards, as (currently) the sole wrangler of software I simply don't have the time to get up to speed enough with any other board to support it in any useful way.

closed 2019-07-09 by Gadgetoid