pimoroni/hyperpixel4 #21: Support other distributions and aarch64

0x0f0f0f, 2019-04-14:
I have correctly set up the display on Void Linux armv7 for the RPI3 B+.

If run on aarch64 the display remains black.

Works fine on armv7 though when manually installing the overlay and configuring `/boot/config.txt`.
Also, I'm launching `hyperpixel4-init` in `/etc/rc.local`.

The current installation script is quite messy and only supports few linux distributions that include systemd, which is not really needed by the Hyperpixel 4.

A great improvement could be giving installation details for platforms other than Raspbian, some other users recently demanded this on your website.

Gadgetoid, 2019-04-15:
Due to time constaints we generally only support Raspbian- there are too many unknowns with alternate distributions. The installer should probably check `/etc/os-release` and fail deliberately on anything else.

It might be worth providing step-by-step manual instructions for more advanced users to follow, though.

KodeIn, 2020-06-15:
I'm using a dirty hack to get the square HyperPixel running on a rpi4b under manjaro.
Using BerryBoot I first install and launch raspbian, that allows me to run the install.sh from the git.
Once it's done and that I set the orientation I want, since raspbian and manjaro are using the same boot partition, I can boot on manjaro and have the screen working.

I know it's far from perfect, but at least, I can get things more or less working...

0x0f0f0f, 2020-06-16:
@KodeIn that's a very dirty hack. See how I manually compiled the updated driver on a RPI3 running Void Linux [here](https://0x0f0f0f.github.io/posts/2019/08/building-a-raspberry-pi-3b-full-keyboard-handheld.-part-1/)

0x0f0f0f, 2020-06-19:
@KodeIn I have now published updated instructions at the same link. It is now quite easier to run an hyperpixel on different distributions. It might actually be easy now to decouple this driver from systemd or apt.

etothex23, 2021-09-17:
@0x0f0f0f  I am trying to install on Manjaro arm. Do you have instructions? Your link is dead.

Gadgetoid, 2021-09-18:
Might help that I have some outstanding PRs to add save-to-Xorg.conf functionality to the rotate utility. This should- in theory- work where there's no screen configuration (arandr) utility - https://github.com/pimoroni/hyperpixel4/pull/145/files

closed 2026-08-25 by Gadgetoid