pimoroni/hyperpixel4 #22: Provide touchscreen calibration instructions.

0x0f0f0f, 2019-04-14:
I'm having an hard time calibrating the touchscreen with `xinput_calibrator`.
The touchscreen is really miscalibrated. When I touch the right part of the screen the cursor never gets further than 3/4 of the display width. 

Xorg does not load the calibration settings I've put in `/etc/X11/xorg.conf.d/99-calibration.conf`

`xinput list-props [hyperpixel's id]`  tells me that the Hyperpixel supports a libinput Calibration Matrix. I have also enabled xorg's `40-libinput.conf` config file to load libinput but no luck yet.

0x0f0f0f, 2019-04-14:
I have calibrated the touchscreen successfully by installing evdev

Here are the steps needed to calibrate the touchscreen on Void Linux. May differ slightly on raspbian (uses apt)

```sh
# install evdev driver and xinput_calibrator
sudo xbps-install -S xf86-input-evdev xinput_calibrator xinput
# create xorg config folder
sudo mkdir /etc/X11/xorg.conf.d/
# enable 10-evdev.conf
sudo cp /usr/share/X11/xorg.conf.d/10-evdev.conf /etc/X11/xorg.conf.d/
# list input devices and remember the id number next to "Goodix Capacitive Touchscreen"
xinput list
# now calibrate the touchscreen and save the config to a file.
# replace HYPERPIXEL_ID with the number you found in the previous step
xinput_calibrator --misclick 0 --device HYPERPIXEL_ID > 99-calibration.conf.temp
# if the display was calibrated correctly, open the config file
# in a text editor and remove the leading lines until the one beginning with "Section"
# and remember to save the file
nano 99-calibration.conf.temp
# move the configuration file under its correct directory
sudo mv 99-calibration.conf.temp /etc/X11/xorg.conf.d/99-calibration.conf
```

t0b3, 2019-05-19:
Your steps didn't work for me on raspbian, Xorg wasn't happy with these options.

However I was lucky to calibrate my hyperpixel4 successful with [these steps](https://wiki.archlinux.org/index.php/Talk:Calibrating_Touchscreen#Libinput_breaks_xinput_calibrator)

It works fine now with `/etc/X11/xorg.conf.d/99-calibration.conf`
```
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "Goodix Capacitive TouchScreen"
        Option  "TransformationMatrix"  "1.69 0 -0.68 0 0.63 0.38 0 0 1"
EndSection
```

NB: Seems like the default calibration confuses 800px vs 480px: 
5/5th of the touching width address only the rightmost 3/5th of the screen, 
while the lower 3/5th of the touch cover the full height.

mikekscholz, 2019-05-20:
On raspbian you wanna make sure you install xserver-xorg-input-evdev then you want to edit your 40- file inside of xorg.conf.d and find the goodix and change the driver line from evdev to xinput. It should then obey your 99-calibration. Also even though xinputcalibrator gives an output to use swapXY, change that to say swapAxes.

mikekscholz, 2019-05-20:
I can be way more specific later when I get off work, I’m nowhere near a pi right now to give exact code:

t0b3, 2019-05-21:
> I can be way more specific later when I get off work, I’m nowhere near a pi right now to give exact code:

That would help to clarify and understand your configuration, cause I feel more confused as I didn't have any 40-evdev file, only 10-evdev or 40-libinput, neither I've seen xinput driver, only evdev or libinput.

mikekscholz, 2019-05-21:
Yea I couldn’t remember the name offhand, in your 40-libinput under the entry for goodix, change the driver line from libinput to evdev

mikekscholz, 2019-05-23:
https://www.raspberrypi.org/forums/viewtopic.php?t=179477 Here is the full write up I found as to what’s actually going on here, final solution is about the 4th post down I think.

blumberg, 2019-05-23:
@0x0f0f0f I have the same issue with my HyperPixel. After a lot of calibration and testing, the best, and easiest, solution for me was to modify a little the driver.

After you clone the driver from GitHub, go to file `src/hyperpixel4.dts` and modify the following lines:

                 touchscreen-size-x = <800>;
                 touchscreen-size-y = <480>;

to 

                touchscreen-size-x = <480>;
                touchscreen-size-y = <800>;

then just run `./install.sh` and everything should work nicely.

t0b3, 2019-05-23:
@blumberg hey works great here!!! please make a pull request to get this nasty issue fixed by smart default values.

mikekscholz, 2019-05-23:
This got me wondering, as to why the actual display is configured as 480x800 then rotated to position 3, as if it’s natively a portrait display. This causes issues playing video in vlc with the same sort of 2/3 of the screen cutoff/offset. I might have to try changing the video timing to 800x480 and then the screen rotation to 0.

blumberg, 2019-05-23:
> @blumberg hey works great here!!! please make a pull request to get this nasty issue fixed by smart default values.

I believe it's a manufacturing issue, a few screens are assembled in one way and others in another way. If that is the case, make no sense to modify this as you will create an issue for those that don't have it now.

What could be done is to recognize this issue and offer an easy way to switch between those two drivers.

those are the commands that I've been using:

```
git clone https://github.com/pimoroni/hyperpixel4.git
cd hyperpixel4
sed -i 's/touchscreen-size-x\ =\ <800>/touchscreen-size-x\ =\ <480>/g' src/hyperpixel4.dts
sed -i 's/touchscreen-size-y\ =\ <480>/touchscreen-size-y\ =\ <800>/g' src/hyperpixel4.dts
./install.sh
```
It could be suggested on the main page: _**If you are experiencing calibration issues, try procedure below:**_

TeCoEd, 2019-06-10:
I was having an issue with the calibration and then I got Garrus (Mass Effect) on board! Basically I used a clean OS install and then used the Pi preferences from the main menu and set the '**OVER SCAN** to off.  Then I installed the HyperPixel 4 software / drivers.  Works now, touch is calibration is accurate.

Also rotated the screen and works fine too.  Only needed to changed the **rotation** to 1

[image]

Gadgetoid, 2019-06-14:
First of, all, I've been in a black hole of 32blit development work along with trying to keep up with my longtail of software maintenance, so this had largely slipped me by. Sorry for the delay!

@mikekscholz because it's a 480x800 pixel portrait display (presumably intended for some kind of handheld computer)
@blumberg as near as I can tell, it's related to something having changed with recent Raspbian updates. I don't know what yet, but I've taken a perfectly-functional HyperPixel 4 off a shelf, updated, tested it and run into this same problem.

Right now the swapping 800/480 touchscreen sizes fix is not working for me and I need to get to the bottom of why.

If I can find an approach that works, then it should be possible to include both the flipped and non-flipped coordinates in the dtoverlay and give the user a parameter to switch to the one that works for them.

I'm somewhat relieved that our touchscreen is not the *only* one having issues.

Thank you everyone for your input here.

Gadgetoid, 2019-06-14:
I've been working on some tweaks to the dtoverlay which get us half of the way there, but calibration is still an absolute nightmare. The best approach I've found thus far is the manual calculation one linked by @t0b3 which gives me a line like:

```
DISPLAY=:0.0 xinput set-prop "pointer:Goodix Capacitive TouchScreen" "libinput Calibratio n Matrix" 0.946372239748, 0.0, 0.125, 0.0, 1.64383561644, -0.676369863014, 0.0, 0.0, 1.0
```
Which I then refined manually by trial and error, and a little use of `evtest` to get the min/max X coordinates to get:

```
DISPLAY=:0.0 xinput set-prop "pointer:Goodix Capacitive TouchScreen" "libinput Calibratio n Matrix" 0.779220779221, 0.0, 0.424025974026, 0.0, 1.69, -0.69, 0.0, 0.0, 1.0
```

This still isn't perfect, the horizontal acceleration of the cursor is off such that it speeds ahead of my finger as I approach the right. But it's better than the half-a-screen problem I had before.

Gadgetoid, 2019-06-14:
I've created a new branch to track the libinput tweaks- https://github.com/pimoroni/hyperpixel4/tree/libinput

Notably it includes changes to the dts file and a udev rule to persist the transformation matrix settings: https://github.com/pimoroni/hyperpixel4/commit/56b0ad487079e9166a910c50cb781eb2d2f03ac7

mmattocks, 2019-07-03:
Hello, I tried to get the libinput branch to work on my hyperpixel 4 but the input seems to be mapped.. strangely (it's flipped in parts of the screen and not in others). Does this issue presently affect raspbian only? My project is pretty simple, mouse input into an ncurses interface, so if there is some alternative that is presently working, I will switch over to that.

Gadgetoid, 2019-07-03:
@mmattocks I can't say I've encountered anything like that. Which OS/version are you running?

markmcgookin, 2019-07-04:
> @0x0f0f0f I have the same issue with my HyperPixel. After a lot of calibration and testing, the best, and easiest, solution for me was to modify a little the driver.
>
> After you clone the driver from [...]

So I tried this... and I thought I was missing something and forked the repo, changed those lines, and also changed the ones below regarding the size in mm.

```
touchscreen-size-x = <480>;
touchscreen-size-y = <800>;
touchscreen-x-mm = <51>;
touchscreen-y-mm = <85>;
```
But on a fresh boot I still have portrait mode and inverted touch. (I grabbed the script from curl https://get.pimoroni.com/hyperpixel4 and stuck it in my repo too to see if it was working some magic that my local version was missing, I changed it to point it at my repo, and used it for the one line install but still ended up with the exact same thing)

QuantumStar, 2019-07-12:
> I believe it's a manufacturing issue, a few screens are assembled in one way and others in another way. If that is the case, make no sense to modify this as you will create an issue for those that [...]

Thank god this worked!!! Since I already had the software installed, I couldn't clone the current repository (it said something about the directory not being empty). After deleting the hyperpixel4 folder, and running the three lines you provided (and changing the rotate from 3 to 1 in the config.txt), it works perfectly!

rho1140, 2019-08-05:
I bought a hyperpixel4 touch off sparkfun recently. Had the weird touch issue described here after running the instructions here,
https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-hyperpixel-4

Downgraded to a raspbian distro from 2018-06, did not update after it asked, it works now. Hope this helps.

markmcgookin, 2019-08-05:
Were you using n00bs or a full raspbian build?

rho1140, 2019-08-05:
I believe it was full raspbian build. I downloaded raspbian-2018-06-29 image from here,
http://downloads.raspberrypi.org/raspbian/images/

It was this thread that led me to it,
https://forums.pimoroni.com/t/hyperpixel-4-0-mouse-offset/10606/5

markmcgookin, 2019-08-06:
> I believe it was full raspbian build. I downloaded raspbian-2018-06-29 image from here,
> http://downloads.raspberrypi.org/raspbian/images/
> 
> It was this thread that led me to it,
> https://forums.pimoroni.com/t/hyperpixel-4-0-mouse-offset/10606/5

Just trying this now, but not sure it will work. Don't you HAVE to use buster if you are using a Raspberry PI 4?

rho1140, 2019-08-06:
I'm using Raspberry pi 3+.

markmcgookin, 2019-08-06:
@rho1140 
> I'm using Raspberry pi 3+.

Yeah as far as I can tell, the hyperpixel4 touch is just unusable with the PI4. Something that Pimoroni just seem to be ignoring now

mikekscholz, 2019-08-06:
> I believe it was full raspbian build. I downloaded raspbian-2018-06-29 image from here,
> http://downloads.raspberrypi.org/raspbian/images/
> 
> It was this thread that led me to it,
> https://forums.pimoroni.com/t/hyperpixel-4-0-mouse-offset/10606/5

Yep. The 2018-09 build is the one that broke everything. Unfortunately not an option for Pi4 users.

Noia, 2019-08-20:
This needs a resolution soon. I'm having these issues on the latest Rasbian on a 3B+ with a brand new Hyperpixel 4. Those should be taken off the store honestly until this is resolved, otherwise it's sale of a known defective device.

markmcgookin, 2019-08-21:
100% agree. It simply does not work with pi4 and the OS that requires. And it’s been troublesome for people on pi3 too

Gadgetoid, 2019-08-21:
Sorry for not being more attentive to this thread.

I've been keeping an eye on this problem, but haven't had a day spare to look into it again yet. I've already had multiple- evidently unsuccessful- shots at trying to figure out why HyperPixel 4 touch is causing so many headaches and have come out a little at a loss. The rift between how Raspbian Stretch and Raspbian Buster treat touchscreens certainly doesn't help.

So far I've been unable to replicate the hypothesis that our touchscreens have a manufacturing variation causing my tested/working calibration to fail for certain users, but it sounds at least plausible. Particularly since I've have had two users who I've talked through the same steps of adjusting the "Transformation Matrix" on Pi 4 whose experiences are consistent with this.

I'll endeavour to get my hands on a variant so that I can build a solution into our installer/helper scripts.

HyperPixel 4 is not - as far as I'm aware - defective, but I recognise that completely wonky touch, coupled with the infuriatingly arcane methods by which it must be calibrated, may be as good as broken to some.

If you're not completely sick of HyperPixel 4 yet, then I'd appreciate a summary of the following from anyone who's currently having problems:

1. Pi Model
2. Raspbian version
3. A brief summary of how your touch is behaving
4. The default orientation of your display relative to the PCB- IE: is the header normally on the top or bottom

Thank you!

Gadgetoid, 2019-08-23:
From more testing it looks like HyperPixel4 works pretty much as expected* out of the box on a freshly downloaded and updated (using the prompt that you get on first boot) Raspbian Buster on Pi 4. As near as I can tell, Pi 4 does not exhibit the weird touch input issue. *However* having an external screen connected will mess up your touch as the calibration matrix applies to the whole desktop area, not just the one screen.

On the Pi 3 the story is much different, and I've got an SD card that demonstrates the 3/4 touch issue that kicked off this thread. HyperPixel4 historically worked with Stretch on Pi 4, but some change in Buster has- for the Pi 3's display/input stack only- completely messed it up.

It seems I got a bit of tunnel-vision with respect to the Pi 4/HyperPixel4/Buster and overlooked earlier Pi's in the same situation.

Now... as for the Pi 3 weirdness. It feels like a bug in the Goodix touch driver, or a bug I was previously (unwittingly) relying upon that's since been fixed.  This is something of an educated guess, but it looks like the `touchscreen-size-x` and `touchscreen-size-y` values are actually being recognised now, whereas previously they were not. I changed these values for Pi 4, but didn't backport to Pi 3. Credit to @blumberg for recognising this and posting a fix.

Anyway the long and short of this is that I have a working HyperPixel 4,  Buster, Pi 3 combination on my desk and I'm attempting to distill this into a fixed installer which you can find here - https://github.com/pimoroni/hyperpixel4/tree/pi3-buster-fix

# Fix for the swapped/inverted touchscreen (Pi 4/Buster)

I've reverted my invert/swap change to the Pi 4. I swear I made this change for a reason and tested it at the time, but it broke for me today, thus the reversion.

You can reinstall for Pi 4 to get this fix and then touch/and the `hyperpixel4-rotate` utility should work as expected.

# Fix for original issue that started this thread (Pi 3/Buser)

If you're using a Pi 3, Buster and a HyperPixel 4 and are experiencing the weird 3/4ths touch issue that was described by the original poster then you should use - https://github.com/pimoroni/hyperpixel4/tree/pi3-buster-fix

Currently this uses the Pi 4 device-tree overlay and doesn't include backlight control (one step at a time!).

First back up your `/boot/config.txt` and `/boot/overlays/hyperpixel4.dtbo` files.

Then you can apply this by cloning it:

```
git clone https://github.com/pimoroni/hyperpixel4 -b pi4-buster-fix
```

And installing the tweaked dtoverlay:

```
cd hyperpixel4/src
make
sudo make install
```

And then below `dtoverlay=hyperpixel4` in `/boot/config.txt` adding the lines:

```
dtparam=touchscreen-swapped-x-y
dtparam=touchscreen-inverted-x
```

Note: These values will depend upon your chosen display rotation.

markmcgookin, 2019-08-23:
For the Pi4 fix… is this a version of Noobs or a version of Raspbian burned directly to the card? 

If it is n00bs… how do you setup a n00bs install and see a prompt without attaching an external display at some point before the drivers for hp4 have been installed? Or should you install, unplug display, restart pi?

Can we have the EXACT steps you have gone through to try and get this resolved please? Were you connected over SSH, which exact OS build were you using etc… I’ve never had success with a clean install on Pi4 4GB HP4.

Noia, 2019-08-23:
The above fix helped me. Using Pi 3B+ w/ Buster.
I did some minor changes due to the rotation of my display, my /boot/config.txt looks like this:
```
dtoverlay=hyperpixel4
dtparam=touchscreen-swapped-x-y
dtparam=touchscreen-inverted-y
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
framebuffer_width=800
framebuffer_height=480
enable_dpi_lcd=1
display_default_lcd=1
display_rotate=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```
Now my cursor actually follows my finger and I can reach all four corners of the screen :)

Gadgetoid, 2019-08-27:
@markmcgookin 

1. Started with Raspbian Buster with desktop and recommended software (2019-07-10) - direct link:  2019-07-10
2. Wrote to SD card and booted in Pi 4
3. Waited for full boot, and then proceeded through the onboarding popup
4. Allowed the onboarding popup to install updates
5. Ran `curl get.pimoroni.com/hyperpixel4 | bash`
6. Made sure I chose option 1 - "Rectangular    - Pi 4"
7. Waited for installer to finish

Note: until I made this fix (https://github.com/pimoroni/hyperpixel4/commit/baa0510e161c906956ec67089fdc0fa87b4b35a7), my touchscreen wasn't working correctly. This is contradictory to several tests I did earlier, but digging up the old HyperPixel4 boards I used in these tests yields the same results. I'm at a loss as to what changed.

@Noia great news! Thanks for letting me know.

alledingenmaken, 2019-09-01:
Fix on Pi3 Buser almost works, but after cold boot it doesn't anymore, your dist/hyperpixel4-init file has CR-return characters in it, which generate a "/usr/bin/env: 'python\r': No such file or directory" error when initializing the systemd hyperpixel4 service, after applying the fix. When I fixed the file by removing the CR's it works.

rcrosado, 2019-09-16:
> The above fix helped me. Using Pi 3B+ w/ Buster.
> I did some minor changes due to the rotation of my display, my /boot/config.txt looks like this:
>
> ```
> dtoverlay=hyperpixel4 [...]

Works perfectly for me as well on Pi3 with Buster using these parameters, thanks!

bolsoncerrado, 2019-09-30:
> I believe it's a manufacturing issue, a few screens are assembled in one way and others in another way. If that is the case, make no sense to modify this as you will create an issue for those that [...]

30th September 2019 and I had to apply the above in order for my Hp4 to work with my brand new SD stretch image to work properly.

actually, I had to:

git clone https://github.com/pimoroni/hyperpixel4 -b pi3
cd hyperpixel4/
sed -i 's/touchscreen-size-x\ =\ <800>/touchscreen-size-x\ =\ <480>/g' src/hyperpixel4.dts
sed -i 's/touchscreen-size-y\ =\ <480>/touchscreen-size-y\ =\ <800>/g' src/hyperpixel4.dts
sudo ./install.sh

Gadgetoid, 2019-10-14:
@bolsoncerrado these changes have now been merged into the Pi 3 branch of HyperPixel 4, along with a new `hyperpixel4-rotate` utility backported from the Pi 4/Buster version.

For usage/install instructions see: https://github.com/pimoroni/hyperpixel4/tree/pi3

I've tested this on Buster and Stretch on a Pi 3B+ and- as long as the latest updates are installed- it appears to work.

The new `hyperpixel4-rotate` utility from Pi 3 favours dropping a file in `/etc/udev/rules.d/` instead of using dtparams to invert/rotate the display.

closed 2019-10-14 by Gadgetoid