pimoroni/hyperpixel4 #23: Screen content rotated by 90 degrees and backwards

raju-bitter, 2019-05-09:
I did a fresh install of Raspian with the Hyperpixel display. The script for installing the drivers went through without any problems, but once I booted, all the the screen content was rotated by 90 degrees, the height of the screen was incorrect (lower 40% showed pixel noise), and all the pixels were backward. See the screenshot.
[image]
Before the upgrade the display was working fine with a version of the software I had installed in August 2018.
Hardware: Raspberry Pi 3 Model B
Raspbian version used: 2019-04-08-raspbian-stretch.zip

mikekscholz, 2019-05-10:
That looks like the older hyperpixel to me, not the 4inch one. theres a different firmware on here just called hyperpixel not hyperpixel4, I dont believe theyre cross-compatible.

Gadgetoid, 2019-06-14:
This is correct- you need the driver from here: https://github.com/pimoroni/hyperpixel

closed 2019-06-14 by Gadgetoid