pimoroni/hyperpixel4 #24: Pi4

Super-Tom, 2019-06-26:
Any plans for updates to allow this to work with a RPi4?

procount, 2019-06-27:
@Gadgetoid is already working on it -see the pi4 branch.

(Some strange changes for Pi4 required..... I wonder if it will require separate installations for Pi4 and other RPis?)

Gadgetoid, 2019-06-28:
A separate installation will likely be the case- although I've updated the one-line-installer to prompt the user to pick "Pi4" if they're installing on that.

The installers are somewhat cross-compatible, and may become more so in future. I think the benefits added by the Raspbian Buster/Pi 4 combination are more than sufficient to make maintaining a couple of separate install branches worthwhile, though. A newly setup Pi4 with HyperPixel4 can now support HDMI *and* HyperPixel4 side-by-side, and the HyperPixel4 can be rotated on the fly into both portrait and landscape orientations.

Pretty cool stuff!

But, yes, Pi4 branch it is- see: https://github.com/pimoroni/hyperpixel4/tree/pi4

Gadgetoid, 2019-10-15:
Hopefully it's now clear that the Pi 4 is supported (although I'm sure some people who've had issues might beg to differ 😢).

closed 2019-10-15 by Gadgetoid