pimoroni/hyperpixel4 #25: Possible to dim screen?

BamBamF16, 2019-06-29:
I am using this with an app that doesn't use X.  It runs its own touch and screen handlers.  However, the normal method for other screens of writing dimming isn't working.  For now we are passing those lines of code, but I am wondering if there is anything non-standard relating to screen dimming.  

For example at night I want the brightness to be 50% of normal.  Where/how do I set this?

AnthonyDiGirolamo, 2019-06-29:
You need to output a PWM signal to the backlight control pin (which is BCM 19). [See pinout here](https://pinout.xyz/pinout/hyperpixel4). Here's [an article about using hardware PWM](https://www.codecubix.eu/linux/hardware-pwm-with-raspberry-pi-zero/) with the rpi zero. 

I gave it a shot and it seems to work. Here's a proof of concept repo you can try: https://gitlab.com/anthonydigirolamo/rpi-hardware-pwm

BamBamF16, 2019-06-29:
I can verify this works and dims the screen correctly.  

Jeff

closed 2019-06-29 by BamBamF16

cpietsch, 2024-03-10:
for me `sudo ./pwm 19 1000000 380000` is already 100%