pimoroni/hyperpixel4 #26: Screen on Raspberry Pi 4 is rotated incorrectly, y-axis is inverted

dstaley, 2019-07-02:
I just installed a Hyperpixel 4 on a Raspberry Pi 4 running a fresh install of Raspbian, and this is the result:

[image]

I'm assuming this isn't the correct orientation, and was expecting landscape. Also the touchscreen's y-axis is inverted. For example, if I touch the Wastebasket icon in the top left, the cursor moves to the bottom left.

Gadgetoid, 2019-07-03:
I've somehow managed to get two separate Pi setups which require different calibration matrix settings, and I've no idea why, or how to reconsile these. Hopefully some more people will weigh in on this issue with their experience and we can find a pattern.

Portrait is actually the default orientation for HyperPixel4, since it's a 480x800 pixel portrait display.

wildestpixel, 2019-07-03:
I can also confirm that my y axis is inverted on a new HyperPixel 4 delivered to me this week. Doing as suggested above run to 'hyperpixel4-rotate left' and set my calibration matrix to '0 1 0 1 0 0' makes good the display but its an ugly solution

Gadgetoid, 2019-07-04:
I think it's related to this commit - https://github.com/pimoroni/hyperpixel4/commit/d4b1e7dc4504d2f2dd7452690fe0e37c006ab267

In some setups there seems to be an interaction between "Coordinate Transformation Matrix" and "libinput Calibration Matrix" that inverts an otherwise perfectly functional touchscreen. I think my second test setup might have been the exception to the rule, so I've borked up the rotate script for everyone else.

It might be worth trying to change the "Coordinate Transformation Matrix" since that's the one that remains constant when `hyperpixel4-rotate` is used.

Try running:

```
hyperpixel4-rotate your-desired-rotation
```

Followed by:

```
xinput set-prop "pointer:Goodix Capacitive TouchScreen" "Coordinate Transformation Matrix" -1 0 1 0 -1 1
```

If that works I can patch `hyperpixel4-rotate` to use this as the default "Coordinate Transformation Matrix" and, perhaps, add a flag to use the inverted one... just in case.

@wildestpixel there used to be a way to trigger right-click using a utility called "twofing" which - iirc - catches a two-finger tap and converts it into a right-click. I don't know if this will work on Buster, though, or if the touch driver for HyperPixel4 is compatible with it. There's an old, old tutorial for setting it up here- https://www.raspberrypi.org/forums/viewtopic.php?t=138575

markmcgookin, 2019-07-04:
Copying here for visibility as discussed in [issue 22](https://github.com/pimoroni/hyperpixel4/issues/22)

So I tried editing `src/hyperpixel4.dts` ... and I thought I was missing something and forked the repo, changed those lines, and also changed the ones below regarding the size in mm.

```
touchscreen-size-x = <480>;
touchscreen-size-y = <800>;
touchscreen-x-mm = <51>;
touchscreen-y-mm = <85>;
```
But on a fresh boot I still have portrait mode and inverted touch. (I grabbed the script from curl https://get.pimoroni.com/hyperpixel4 and stuck it in my repo too to see if it was working some magic that my local version was missing, I changed it to point it at my repo, and used it for the one line install but still ended up with the exact same thing)

wildestpixel, 2019-07-04:
@gadgetoid, that gives me 

**pi@raspberrypi:~ $ hyperpixel4-rotate right
Rotating display
Saving display settings to /usr/share/dispsetup.sh
Setting matrix: -1 0 1 0 -1 1
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
pi@raspberrypi:~ $ xinput set-prop "pointer:Goodix Capacitive TouchScreen" "Coordinate Transformation Matrix" -1 0 1 0 -1 1
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  131 (XInputExtension)
  Minor opcode of failed request:  57 ()
  Value in failed request:  0x80
  Serial number of failed request:  20
  Current serial number in output stream:  21**

And input is broken naturally

markmcgookin, 2019-07-04:
> @Gadgetoid, that enters okay, but the touch is still broken I’m afraid to the degree I now can’t get it back with the working parameters of
> 
> `0 1 0 1 0 0`
> 
> alone in the executable /usr/bin/hyperpixel4-rotate
> 
> This, however
> 
> `xinput set-prop "pointer:Goodix Capacitive TouchScreen" "Coordinate Transformation Matrix" 0 1 0 1 0 0 0 0 1`
> 
> Gets me back okay when I rotate left or even right( I added 0 0 1 to make 9 digits as I saw you tried and edited from 6 to 9 earlier), it doesn’t however work for the vertical orientations.

So is your touch working now with 1 0 1 0 0 0 0 1?

wildestpixel, 2019-07-04:
Yes I’ve got it working perfectly with left and right orientations. Just need work on the verticals

Gadgetoid, 2019-07-05:
To make this stick on reboot you need to edit the udev rules that are otherwise automatically created by `hyperpixel4-rotate`. In fact it's probably the "libinput Calibration Matrix" you want to change, letting "Coordinate Transformation Matrix" retain whatever its defaults might be- assuming those are in any way consistent and unaffected by some buried system setting I've overlooked.

These rules are saved at `/etc/udev/rules.d/98-hyperpixel4-calibration.rules` and basically contain the first 6 numbers from a full transformation matrix.

Again- be wary that "libinput Calibration Matrix" and "Coordinate Transformation Matrix" appear to be two completely sepearate things.

I'm setting up a new system now to see what state I get it into.

It would be nice to have a GUI Utility to handle touch remapping now we can rotate displays on the fly. I'd argue that this is the remit of "Screen Configuration" itself, but I don't know how willing the foundation are to bend its codebase to their will, and I can't see any evidence that they've forked it at all.

Gadgetoid, 2019-07-05:
Okay! I think the problem mostly stems from the fact that - by default - the screen is portrait with a flipped x/y touch input. While this is easily fixed with `hyperpixel4-rotate` it really throws people off, and probably ends up getting some setups into bad states with attempted fixes.

On my fresh setup I simply ignroed this initial glitch and ran `hyperpixel4-rotate left` to get a functional, properly set up touchscreen/screen combo.

With this in mind I've now patched the dtoverlay so that touch input matches the default portrait orientation, and I've also patched the `hyperpixel4-rotate` utility so that the transformations it set are compatible with this default.

I also came across this bug https://github.com/pimoroni/hyperpixel4/issues/28 which you should be aware of if you plan on using *portrait* mode.

## What you should avoid:

* Running any `xinput set-prop` commands manually
* Manually rotating the screen in Screen Configuration
* Using `display_rotate` or `lcd_rotate` in /boot/config.txt

## Steps for properly set up screen from a fresh copy of NOOBS Buster on Pi 4:

* `curl -sSL https://get.pimoroni.com/hyperpixel4 | bash`
* Answer `y` to "Are you installing on a Raspberry Pi 4" (uses the Pi4 branch on this repo)
* Let the installer run
* Reboot
* `hyperpixel4-rotate left` to set and save landscape orientation

The last step, alas, cannot be part of the installer because the Pi (arandr) wont rotate a screen it doesn't know about, and I'm using arandr to generate `/usr/share/dispsetup.sh` in the format that it expects.

## If your setup is working

If if ain't broke, don't fix it!

If you're tempted to reinstall `hyperpixel4` using the steps above, however, then please report back. I'm still not sure what effect any prior changes might have on an updated setup.

markmcgookin, 2019-07-14:
> Okay! I think the problem mostly stems from the fact that - by default - the screen is portrait with a flipped x/y touch input. While this is easily fixed with `hyperpixel4-rotate` it really throws [...]

So I've just got back and tried this with the latest noobs... downloaded noobs 3.2.0 - Installed, added to wifi (all attached to TV) then ran the installer (I cloned the pi4 branch and ran install.sh) and rebooted. I left the HDMI in, so when I restarted the TV display was at 90 degrees and the screen wasn't on. Unplugged HDMI, rebooted, screen not coming on now.

markmcgookin, 2019-07-18:
Has there been any more movement on this? I tried again last night, and the latest noobs, and latest bash script still leave everything inverted. 

Annoying as it's sold as a product that "works out of the box with one install script" and is currently unusable, for a fairly premium price.

SirPottinger, 2019-07-20:
Hi all, 

after sucessfully installing (one line isntaller "curl -sSL https://get.pimoroni.com/hyperpixel4 | bash") and rebooting the sytem I tried to rotate the display by using "hyperpixel4-rotate left".
The following error message appears:

[image]

May that help you fixing the problem?

markmcgookin, 2019-07-20:
@SirPottinger, were you SSH’d in or was this a native session on the pi itself?

SirPottinger, 2019-07-20:
Yes. I am using the Pi via SSH and your solution worked. After that the display rotated correctly, but the touch orientation didn't rotate accordingly. It stays the same as before rotating. "hyperpixel4-rotate left" doesn't affect the still screwed up touch screen axis.

markmcgookin, 2019-07-20:
Glad that helped you get a little further. Yes the axis seem screwed up for most of us, and pimoroni just seem to be ignoring it. We bought a fairly expensive component off them that basically doesn’t work. There was some chat on another issue about it being the actual component makers fault as it’s natively a portrait display, but my axis are even screwed in portrait. It’s frustrating, as we purchased a product from pimoroni and regardless of the source it simply doesn’t function as intended. All marketing material shows it in landscape, it’s clearly sold like that.

mikekscholz, 2019-07-23:
Welcome to the wonky world of touchscreens on the Pi, where all touchscreens collectively lost their minds with the april 2019 release of raspbian, with every touchscreen I own needed a different kind of medication to get under control. Like 4 different kinds/brands of displays all went haywire in unique ways.

markmcgookin, 2019-07-23:
@mikekscholz can you find the medication for this one please? I’d hate to have to have it put down

mikekscholz, 2019-07-23:
As an aside I finally got a simple fix figured out, but im using a 3b+ on the latest version of stretch and dont know if it will carry over to buster or not... its looking like pimoroni has updated the drivers as well since then.

mikekscholz, 2019-07-23:
I wrote up the basics and linked to the documentation I used in issue #22, may 20th entry. Its a bit broken up tho. Im downloading a buster image right now before I head home from work and Ill work up the exact process I use if it still works.

markmcgookin, 2019-07-26:
Thanks @mikekscholz that's awesome. I'll have a look if I get a chance today some time myself

markmcgookin, 2019-07-28:
Going to bump this issue again, as we don't seem to have a simple concrete fix from the people who provided the screen and installation script. I have done a clean raspbian install today to get docker working, everything fresh and updated. Ran the installation, and while the display rotates, the touch axis are still all off. @Gadgetoid is there any update on a fix for this from Pimoroni? We haven't heard anything from yourself/them in over 3 weeks and those of us who bought pi4 and this screen to use for touch are basically left with a non-functional product.

mikekscholz, 2019-07-29:
I haven’t abandoned you, im still working on replicating my fix on buster. My company just switched me back to morning shift a week ago and everythings a little fuzzy at this point.

markmcgookin, 2019-07-29:
@mikekscholz how dare they? I shall write them a strongly worded letter and get you off that awful shift.

Jokes, but thank you so much for your help. I read the docs you linked to but it was a bit over my head

mikekscholz, 2019-07-29:
It was over mine too when I started, in my case it was a 7in hdmi with usb touch that was a portrait mode touchscreen controlling a 3d printer that was integrated into the enclosure, and a software update rendered it completely useless outside of landscape mode, took me like 2 months. And happened to apply to my hyperpixel too

markmcgookin, 2019-08-13:
@mikekscholz any chance you got a few notes together mate?

koshimazaki, 2019-09-20:
@Gadgetoid i get this error below, once following steps above. 
And had to do it as my Pi is stuck on splash screen after changing config file and dtbo.
I tried reinstalling number of times but Pi is still stuck. 

> 

Rotating display
Can't open display 
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in __init__
    version_output = self._output("--version")
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _output
    raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: Can't open display 

Setting matrix: 0 -1 1 1 0 0
Unable to connect to X server
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules

>

koshimazaki, 2019-09-20:
Also after running **make** I get this

> dtc -I dts -O dtb -o hyperpixel4.dtbo hyperpixel4.dts
> hyperpixel4.dtbo: Warning (unit_address_vs_reg): /fragment@2/__overlay__/i2c@0: node has a unit name, but no reg property
> hyperpixel4.dtbo: Warning (gpios_property): /__fixups__:gpio: property size (21) is invalid, expected multiple of 4

are these expected/standard warnings?

koshimazaki, 2019-09-21:
i managed to get it to work. @markmcgookin @Gadgetoid 
Got passed splash screen after deleting `display_rotate=1` from config.txt
rotated screen left and it seem to work fine. Tried with two Pis as had same issue with splash screen so thats consistent. Have touch working on one in landscape and other in portrait mode.

and I also still have issue as described [here](https://github.com/pimoroni/hyperpixel4/issues/42)

Gadgetoid, 2019-10-01:
`display_rotate=1` seems to completely break, quite catastrophically on Pi 4/Buster. I've been looking for a way to rotate the console, given that the screen is portrait and cannot be rotated by changing any other settings but no dice yet. Pi 4 has really changed the whole game with respect to DPI displays.

R0B3RDV, 2019-10-01:
what also breaks the installation of the screen is if you execute raspi-config or if you enable the input possibilities like remote GPIO etc. and escpecially screensettings. What helped with me was walking trough the config.txt after installation and commented out all the non essensial commands

Gadgetoid, 2019-10-08:
Indeed. Enabling interfaces can make subtle things go wrong, since the screen requires pretty much all-the-GPIO to function.

Gadgetoid, 2019-11-05:
I think - hope - this original issue here is solved. As we've segued into console rotation/display_rotate discussion around this should continue here: https://github.com/pimoroni/hyperpixel4/issues/39

Otherwise, closing for now! As always feel free to re-open if this rears its head again.

closed 2019-11-05 by Gadgetoid

tuna-f1sh, 2019-12-06:
I know this is closed but I'm going to throw in something relevant I think. 

After a morning head scratching why an exact cloned setup (Balena application), resulted in one with inverted touch and one with out I've come to the conclusion there are different hardware variants of the Hyperpixel4 that result in different touch driver behaviour! 

By swapping only the screens, the touch is rotated 180 deg. This means one has to apply different overlay fragments for different screens! As far as I can tell, the only identifying difference is the batch serial on FPC?

```
FPC serial 2018.06.1 Overlay: `"hyperpixel4,touchscreen-inverted-x"`
FPC serial 2018.06.14 Overlay: `"hyperpixel4"`
```

Should help others.

The only different I can spot