pimoroni/hyperpixel4 #27: `hyperpixel4-rotate` should probably set DISPLAY=:0.0 for use over SSH

Gadgetoid, 2019-07-04:
I'm not entirely sure what the etiquette is in Linux for targeting a specific display when one is not set, but in the case of the Pi 4 I don't believe there's any harm in always assuming `DISPLAY=:0.0`, or setting this variable only when it's not otherwise set in th ecalling env. This would allow `hyperpixel4-rotate` to run over SSH without users having to worry about prepending a display variable.

svanstrom, 2019-12-27:
As someone that just ran into this problem (and still feeling stupid about it), I'd suggest that instead of silently fixing this the user is instead hit with an appropriate, educational, error message.

locki-cz, 2021-04-07:
problem solved: https://youtu.be/OJ59hXSyBoI?t=674

closed 2021-05-20 by Gadgetoid