pimoroni/hyperpixel4 #29: Portrait mode

procount, 2019-07-09:
Since the Hyperpixel 4 is a bit non-standard with it being natively portrait, is there any merit in the drivers, dt overlays, rotation scripts etc, faking its orientation to make it look natively in landscape?

Or would this mess up the desktop drivers (especially if they bypass theses drivers)?

Gadgetoid, 2019-07-09:
I think this is - at least on Pi 4 - explciitly the domain of xrandr and the underlying graphics driver, which should handle the rotation from native orientation to whatever the user desires.

I appreciate this isn't much cop when there's no x desktop or other facility to do this, however. I do wonder if there's a way to poke the right filesystem node or something to handle rotation.

Since the DPI output happens entirely within the videocode, controlled only by config.txt parameters we're entirely at the behest of those for what's achievable. I think `lcd_rotate` still works for what it's worth, and the touch driver gives a reasonable amount of control over axis mirroring and swapping.

I'm guessing your perspective on this is being able to run a usable pre-desktop environment?

To be fair, landscape orientation does feel like a more sensible and less surprising default.

PinkSnake, 2019-10-08:
Is it possible to have more informations about the correct way to configure portrait mode on Raspi 3 B ?
Thx!

Gadgetoid, 2019-10-14:
@PinkShake this has changed as of the new improvements to the Pi 3B+ support branch- you should start from a fresh Raspbian Buster install a nd follow the install instructions here: https://github.com/pimoroni/hyperpixel4/tree/pi3

Finally use `hyperpixel4-rotate left` or `hyperpixel4-rotate right` to pick the landscape orientation you desire.

@procount I've, thus far, sound no way to rotate the console on a Raspberry Pi. While fbcon supports a "rotate" parameter, it doesn't seem to accomplish much. On the Pi 4 the "display_rotate" command doesn't work, either, meaning no rotation occurs until X fires up and does its magic.

closed 2019-10-14 by Gadgetoid

procount, 2019-10-14:
>  I've, thus far, sound no way to rotate the console on a Raspberry Pi

PINN works purely in the console/framebuffer world (No X-windows, xrandr etc), so I did all my rotations using my new driver controlled via the DTB settings and it seemed to work fine in all 4 orientations, at least with Stretch (apart from the screen blanking when a certain part of the screen is touched). I'm not sure I've revisited it since the change to Buster. I'll need to take a look at this once I've finished my current project.....