pimoroni/hyperpixel4 #30: Analog audio output conflicts?

fquirin, 2019-07-12:
Hi Hyperpixel team,

I'm not able to get any audio output via the analog jack on my Raspberry Pi 4 (Raspian Buster with desktop) after installing the Hyperpixel 4.0 Touchscreen. I can choose "Analog" in the desktop volume settings (right click on the speaker) but the volume will be 0 and when I try to increase the volume it snaps back into "no selected source".

Could this be related to the backlight control issue mentioned in the Hyperpixel (1.0?) documentation?
If so is it possible to deactivate PWM and re-enable analog audio?

Gadgetoid, 2019-07-12:
There's no PWM  backlight baked into HyperPixel4 since the kernel driver was never made available, so this shouldn't be a problem from what I recall.

Do you have a `dtparam=audio=off` in your `/boot/config.txt` somewhere that might be interfering?

Admittedly I haven't tried a HyperPixel4 alongside analog audio on Pi4- if the above doesn't some how cure your woes then I'll give it a go.

fquirin, 2019-07-12:
The `/boot/config.txt` is basically untouched:
```
# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

I'll restart tomorrow with a fresh Raspian Buster, see if audio still works and then install the Hyperpixel 4 again with the latest bash script ... lets see.

fquirin, 2019-07-13:
I've made a fresh installation with newest version of Raspbian Buster and updated Hyperpixel 4.0 installation script and ... everything works now :-D

closed 2019-07-13 by fquirin