pimoroni/hyperpixel4 #31: XRANDR and XPDYINFO out different pixel sizes

neurogamer, 2019-07-22:
I have an issue with a Hyperpixel. I try to run Wine on a pi and it works flawlessly on an external HDMI display. But on the hyperpixel, the window is too streched vertically and no matter what settings I use, I can't seem to resolve it. Then, i dumped xpdyinfo and it detects the screen with a pixelsize of 96x96. But xrandr finds it at 0x0. Is there another driver or some way to make the screen behave correctly?

Gadgetoid, 2019-07-22:
There's no display driver for HyperPixel- it just relies on the Pi's native ability to output a picture via DPI on the header.

As far as I'm aware, xrandr only works on the Pi 4 with the 3D graphics driver enabled- what OS/Pi version are you using?

neurogamer, 2019-07-22:
Hi there.

I'm running a Raspberry Pi3, with Raspbian. I run Wine apps (mostly Diablo2). If i play via an external HDMI monitor, it scales absolutely fine. If I run it on the attached Hyperpixel, it stretches out a bit, and there's no way I can fullscreen it (via a bash script that I use to run the app).

If the fault in on the RPi, that would be great, if you think that the Pi4 would sort my issue.

Gadgetoid, 2019-07-23:
Could you possibly snap a picture of it happening? I'm not totally sure the Pi 4 setup would do things differently enough to fix it. I'm guessing it's some sort of edge case interaction between Diablo 2's resolution and the inability to really run custom modes on the screen.

What resolution are you running Diablo 2 in? Native pre LoD is 640x480 which should fit on HyperPixel 4 with black bars either side, wheres the LoD additional 800x600 might cause problems.

Have you tried a custom resolution patch to see if you can get 800x480?

(Sheesh I lost so many hours to Diablo 2 back in the day!)

neurogamer, 2019-07-23:
Omg. Another Diablo fan around.

So. Full disclosure: I'm making some portable pc's/consoles for me and my friends. And I want to run Wine (via Retropi, the only way it works fine). And the hyperpixel is the highest quality screen I can get for the RPi's. So this is why I'm in a pickle and start annoying people on pimoroni's discord/github.

I've added some photos with my issues (the way the game stretches, xrandr -q 's output + i piped the output of xpdyinfo into a text file to make it easier to read + the contents of the sh script that i try to run it with).  

PS: if i connect the Pi to an external hdmi screen, it works fine, fullscreen, automatically. It only starts acting up on the hyperpixel.

[image]
[image]
[image]
[image]
[image]

Gadgetoid, 2019-10-15:
Sheesh I can't believe I haven't replied since July- sorry, time really has a habit of blurring past for me.

Did you get anywhere with this? From your screenshot it looks like the 800x480 of the HyperPixel4 is cutting off the 800x600 resolution of the LoD expansion. Scaling from 800x600 to 640x480 might be possible, but not pixel perfect- I also don't think resolution patches work in the menu screen. It's been a while since I've played!

What resolution is your external HDMI display? Other than the external display being large enough to accomodate the Window, and WINE perhaps not being able to scale down, or scale to non-integer interverals, or non aspect-preserved intervals I can't fathom any reason why HyperPixel 4 would differ from an HDMI display.

I have the OpenPandora version of Diablo 2, which is natively-compiled for ARM and linked against WINE. It has scaling to 800x480 (the resolution of the OPs screen) and doesn't look too bad despite being squashed vertically. It must be possible to enable non-integer display scaling in WINE but since this port is closed source :( I can't figure out how it's done.

closed 2020-07-31 by Gadgetoid