pimoroni/hyperpixel4 #33: Future: Could HyperPixel 5 run the touchscreen data on the same pins used by the FanSHIM?

ColinNg, 2019-07-26:
I have both HyperPixel 4 and FanSHIM. While FanSHIM runs in default mode, there can be no communication to it, and its RGB LED just does whatever it feels like. Could a future HyperPixel run the touchscreen lines on the pins used by the SHIM series, so they can be operated as well?

monglet, 2019-07-30:
I believe the fanshim is what's being changed- to i2c control.

Gadgetoid, 2019-10-14:
The "4" is the diagonal size in inches, not the version 😆 

But, broadly, no, the HyperPixel is constrained by which pins the Raspberry Pi can do DPI over- https://pinout.xyz/pinout/dpi

Since  BCM2 and BCM3 (normally the i2c bus) are used for the vitally important V-Sync and H-Sync signals (assuming you don't consider all colour signals equally as important) it's not possible to use the regular i2c bus for anything useful.

While a revision of Fan SHIM that uses the regular i2c bus both for fan and LED control is likely- it *still* wont work with HyperPixel 4, or any future/past or present DPI-based display.

Unless- of course- you tap off the touchscreen i2c bus and route that to the board in lieu of BCM2/3. This would probably require some bodge wiring and trace cutting but it can be done.

If you're feeling particularly brave you could cut the trace to the fan control pin and re-wire it to an i2c GPIO extender hooked up to the touchscreen clk/dat lines. Some discussion of a similar hack here - https://github.com/pimoroni/fanshim-python/issues/48

closed 2019-10-15 by Gadgetoid