pimoroni/hyperpixel4 #34: Cannot install on rpi4

rasmuskreiner, 2019-07-30:
Hi All,

I get this error message when trying to install the software on a brand new rpi4. 

```jul 30 11:16:34 raspberrypi systemd[1]: Starting HyperPixel 4.0" LCD Display Initialization...
jul 30 11:16:34 raspberrypi hyperpixel4-init[748]: bcm2835_init: gpio mmap failed: Cannot allocate memory
jul 30 11:16:34 raspberrypi systemd[1]: hyperpixel4-init.service: Main process exited, code=exited, status=1/FAILURE
jul 30 11:16:34 raspberrypi systemd[1]: hyperpixel4-init.service: Failed with result 'exit-code'.
jul 30 11:16:34 raspberrypi systemd[1]: Failed to start HyperPixel 4.0" LCD Display Initialization.```

It has a fresh install of Raspbian Buster.

monglet, 2019-07-30:
Are you using the pi4 branch?

rasmuskreiner, 2019-07-30:
Ups...

closed 2019-07-30 by rasmuskreiner

AndreasBrauckmann, 2019-09-09:
i have the same error:  message when trying to install the software on a brand new rpi4.

I get an error: **bcm2835_init: gpio mmap failed: Cannot allocate memory**, does anyone have an explanation for this?

Installation:
pi@raspberrypi:~ $ git clone https://github.com/pimoroni/hyperpixel4
```
Cloning into 'hyperpixel4'...
remote: Enumerating objects: 105, done.
remote: Counting objects: 100% (105/105), done.
remote: Compressing objects: 100% (83/83), done.
remote: Total 166 (delta 59), reused 50 (delta 21), pack-reused 61
Receiving objects: 100% (166/166), 462.97 KiB | 344.00 KiB/s, done.
Resolving deltas: 100% (82/82), done.
```
pi@raspberrypi:~ $ cd hyperpixel4
pi@raspberrypi:~/hyperpixel4 $ sudo ./install.sh

```
Notice: building hyperpixel4.dtbo
Created symlink /etc/systemd/system/multi-user.target.wants/hyperpixel4-init.service → /etc/systemd/system/hyperpixel4-init.service.
*** Job for hyperpixel4-init.service failed because the control process exited with error code.
See "systemctl status hyperpixel4-init.service" and "journalctl -xe" for details.
Installed: /usr/bin/hyperpixel4-init
Installed: /etc/systemd/system/hyperpixel4-init.service
Installed: /boot/overlays/hyperpixel4.dtbo
Config: Added dtoverlay=hyperpixel4 to /boot/config.txt
Config: Added overscan_left=0 to /boot/config.txt
Config: Added overscan_right=0 to /boot/config.txt
Config: Added overscan_top=0 to /boot/config.txt
Config: Added overscan_bottom=0 to /boot/config.txt
Config: Added framebuffer_width=800 to /boot/config.txt
Config: Added framebuffer_height=480 to /boot/config.txt
Config: Added enable_dpi_lcd=1 to /boot/config.txt
Config: Added display_default_lcd=1 to /boot/config.txt
Config: Added dpi_group=2 to /boot/config.txt
Config: Added dpi_mode=87 to /boot/config.txt
Config: Added dpi_output_format=0x7f216 to /boot/config.txt
Config: Added display_rotate=3 to /boot/config.txt
Config: Added hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6 to /boot/config.txt
```
pi@raspberrypi:~/hyperpixel4 $ journalctl -xe
```
-- The job identifier is 789.
Sep 09 18:16:28 raspberrypi systemd[1]: Mounted Arbitrary Executable File Formats File System.
-- Subject: A start job for unit proc-sys-fs-binfmt_misc.mount has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit proc-sys-fs-binfmt_misc.mount has finished successfully.
-- 
-- The job identifier is 789.
Sep 09 18:16:35 raspberrypi sudo[1166]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/find / -iname *hyperdisplay*
Sep 09 18:16:35 raspberrypi sudo[1166]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Sep 09 18:16:48 raspberrypi sudo[1166]: pam_unix(sudo:session): session closed for user root
Sep 09 18:17:01 raspberrypi CRON[1178]: pam_unix(cron:session): session opened for user root by (uid=0)
Sep 09 18:17:01 raspberrypi CRON[1182]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Sep 09 18:17:01 raspberrypi CRON[1178]: pam_unix(cron:session): session closed for user root
Sep 09 18:17:40 raspberrypi sudo[1196]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/curl get.pimoroni.com
Sep 09 18:17:40 raspberrypi sudo[1196]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Sep 09 18:17:41 raspberrypi sudo[1196]: pam_unix(sudo:session): session closed for user root
Sep 09 18:19:19 raspberrypi sudo[1219]:       pi : TTY=pts/0 ; PWD=/home/pi/hyperpixel4 ; USER=root ; COMMAND=./install.sh
Sep 09 18:19:19 raspberrypi sudo[1219]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Sep 09 18:19:20 raspberrypi systemd[1]: Reloading.
Sep 09 18:19:20 raspberrypi systemd[1]: Reloading.
Sep 09 18:19:21 raspberrypi systemd[1]: Starting HyperPixel 4.0" LCD Display Initialization...

-- Subject: A start job for unit hyperpixel4-init.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit hyperpixel4-init.service has begun execution.
-- 
-- The job identifier is 793.
**Sep 09 18:19:21 raspberrypi hyperpixel4-init[1259]: bcm2835_init: gpio mmap failed: Cannot allocate memory**
Sep 09 18:19:21 raspberrypi systemd[1]: hyperpixel4-init.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit hyperpixel4-init.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 1.
Sep 09 18:19:21 raspberrypi systemd[1]: hyperpixel4-init.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit hyperpixel4-init.service has entered the 'failed' state with result 'exit-code'.
Sep 09 18:19:21 raspberrypi systemd[1]: Failed to start HyperPixel 4.0" LCD Display Initialization.
-- Subject: A start job for unit hyperpixel4-init.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit hyperpixel4-init.service has finished with a failure.
-- 
-- The job identifier is 793 and the job result is failed.
Sep 09 18:19:21 raspberrypi sudo[1219]: pam_unix(sudo:session): session closed for user root
```