pimoroni/hyperpixel4 #35: Hyperpixel Square inverted colours (BGR instead of RGB?)

orbrey, 2019-08-01:
[image]
Colours inverted as per picture. 

"I'm using the latest stock Raspbian with your install script and everything appeared to install fine. On reboot the screen works but as you can see that it looks odd (see image attachment). I've tried uninstalling and reinstalling the drivers but I get the same result each time." This is as of July 30th so should include the fix for wobbling.

Gadgetoid, 2019-08-01:
This is definitely far more in the realm of "horribly distorted" than simply inverted.

The closest I'm aware of is a similar problem that can allegedly be fixed by removing the line "gpio=0-25=a2" from `/boot/config.txt`

jankmammal, 2019-08-01:
[image]

Here's a photo that hopefully shows the inverted colours better!

"gpio=0-25=a2" doesn't seem to help I'm afraid.

arturo182, 2019-08-01:
Had the same problem, can confirm @Gadgetoid's fix worked for me.

yakovlevyuri, 2019-08-01:
@arturo182 @Gadgetoid Worked for me as well, but the touch screen is disabled after that

jankmammal, 2019-08-02:
Commenting out "gpio=0-25=a2" confirmed working :)

jose-cardoso, 2019-08-05:
I'm the original reporter of this issue via Pimoroni Support and can confirm @Gadgetoid's solution worked for me. Much appreciated. :)

sianisstruggling, 2023-03-19:
I have this same issue on my hyperpixelr, but I do not have "gpio=0-25=a2" in my boot config to start with.
Has there been any further documentation on this issue?

peregrine, 2023-09-18:
I realize this issue is pretty old at this point, but I'm also seeing the same problem with my Raspberry Pi 3A+ and Hyperpixel4 Square with Buster installed and like the above poster, I don't have `gpio=0-25=a2` in by boot config either.

Any suggestions would be much appreciated!

[image]

peregrine, 2023-09-18:
And of course as soon as I posted this question I ended up fixing it. I used the manual install [here](https://github.com/pimoroni/hyperpixel4/tree/pi3) instead of the "wizard" I had been using through this script: `curl -sSL get.pimoroni.com/hyperpixel4-legacy | bash`

closed 2026-08-25 by Gadgetoid