pimoroni/hyperpixel4 #36: dtoverlay only handle three of four given parameters

cultpenguin, 2019-08-02:
Hi
I am testing HyperPixel in a Raspperry pi, and have had trouble with the rotated/inverted touch interface. I need to rotate and invert the touch interface. It seems that (according to the Readme) if the following line in /boot/config.txt was properly handled, my problems with touch would vanish:

`dtoverlay=hyperpixel4,touchscreen-swapped-x-y,touchscreen-inverted-y,touchscreen-inverted-x`

But, it seems that only the first 3 of the 4  parameters are handled. That last (here `'touchscreen-inverted-x'`) is ignored.

If the fourth parameter is ignored, will it be possible to update the code such that it is not ignored?

Have a nice day
- Thomas

Gadgetoid, 2019-08-20:
I think this is a hard limit in the parsing of `/boot/config.txt` which - I imagine - has a line buffer with a hard character limit.

You can place parameters on the line underneath with `dtparam=`

From section 3.2 of the Pi device tree docs - https://www.raspberrypi.org/documentation/configuration/device-tree.md

If you have an overlay that defines some parameters, they can be specified either on subsequent lines like this:
```
dtoverlay=lirc-rpi
dtparam=gpio_out_pin=16
dtparam=gpio_in_pin=17
dtparam=gpio_in_pull=down
```
or appended to the overlay line like this:
```
dtoverlay=lirc-rpi:gpio_out_pin=16,gpio_in_pin=17,gpio_in_pull=down
```

cultpenguin, 2019-08-20:
Exactly what I needed. Thank you very much!
The hyperpixel 4 now works fine, with the following lines in /boot/config.txt

```
dtoverlay=hyperpixel4,touchscreen-swapped-x-y
dtparam=touchscreen-inverted-y
dtparam=touchscreen-inverted-x
```

Thank you and have a nice day
- Thomas

closed 2019-08-20 by cultpenguin