pimoroni/hyperpixel4 #38: Why is my display in portrait mode?

maciekish, 2019-09-03:
Hi, i have a Rapsberry Pi 4 and a Hyperpixel 4.0. I used the normal setup script and selected "1" - "Rectangular Hyperpixel and Pi 4". After reboot my display works but its in portrait mode. Whats wrong?

Gadgetoid, 2019-09-04:
Run the command `hyperpixel4-rotate left` or `hyperpixel4-rotate right` to rotate screen and touch.

It's actually a portrait screen but the Pi4-native commands to rotate it don't work until it's installed/rebooted, thus the weird default and the need for `hyperpixel4-rotate.`

frank26080115, 2019-09-28:
That command didn't work for me, I tried both with and without sudo, tried both rebooting and full power downs

The touchscreen directions are now reflecting the change, but the screen display itself is still portrait, I used "left" for "hdmi is down". In fact, none of the directions work

I did a `cat /var/log/syslog | grep -i hyperpixel`

    Sep 27 23:41:13 raspberrypi systemd[1]: Starting HyperPixel 4.0" LCD Display Initialization...
    Sep 27 23:41:13 raspberrypi hyperpixel4-init[338]: /usr/bin/hyperpixel4-init:52: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
    Sep 27 23:41:13 raspberrypi hyperpixel4-init[338]:   GPIO.setup([CLK, MOSI, CS], GPIO.OUT)
    Sep 27 23:41:13 raspberrypi hyperpixel4-init[338]: Setting up HyperPixel4...
    Sep 27 23:41:13 raspberrypi systemd[1]: hyperpixel4-init.service: Succeeded.
    Sep 27 23:41:13 raspberrypi systemd[1]: Started HyperPixel 4.0" LCD Display Initialization.

Please help

witemike, 2020-12-29:
Hi there. I know this thread is old and you've probably resolved the issue, but if you're using a headless set up over SSH the command just has to be prefixed with "DISPLAY=:0.0", so the full command would be (no need for sudo):-

_"DISPLAY=:0.0 hyperpixel4-rotate left"_ 

"left" can be replaced with "right", "inverted", "normal". can confirm I haven't yet found a way to boot into landscape. 

Hope this helps.

closed 2026-08-25 by Gadgetoid