pimoroni/hyperpixel4 #39: Pi4 and landscape in console (Raspbian Buster lite)

illydone, 2019-09-04:
I have a pi4 and just want to use the Hyperpixel 4 for a portable device in console mode on a raspbian buster lite. After installation everything is working in portrait mode.

I tried to use hyperpixel4-rotate but the script isn't for console/framebuffer (it asks for xinput and xrandr which aren't installed).

I tried to change directly in config.txt with display_rotate=1 but it doesn't work
-either I keep "dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6" and screen is locked in landscape: no cursor, no update when I type something.
-either I try to invert 480 and 800 in dpi_timings (I don't think it would work but it was worth a try) and display is strange: dark corners, display bugs, etc.

I think there should be correct dpi_timings in config.txt to get it to work, but I don't know which one.

I don't test touch in console as it is useless for me.

Gadgetoid, 2019-09-05:
The screen is a portrait display. The `dpi_timings` values relate to how the pixels are transferred over the DPI bus and don't have any effect on rotation etc.

Either `display_rotate` or `lcd_rotate` should work in `/boot/config.txt` since these are the standard method for rotating the video output on the Pi.

I wont be able to experiment with these myself until sometime next week though.

illydone, 2019-09-05:
Okay, thanks. I already supposed it was a portrait screen. 

To turn the screen, I already tried this:
-lcd_rotate doesn't have any effect
-display_rotate turn the text but I don't have any more access to the console nor any tty as if everything is locked

I'll try to access it using ssh next time to identify the problem. I'll also reinstall a fresh and updated Buster lite and reinstall the screen to be sure there wasn't any external parameter.

illydone, 2019-09-06:
I made another test with Raspbian Buster full and with graphical interface, everything is working and after installation, X11 is starting in portrait mode.

With hyperpixel4-rotate, you can turn it and touch is also turned. However, the console (when ctrl-alt-F1) is still in portrait. When trying to change framebuffer-width and _height and display_rotate, everything goes wrong and there are display bugs in X11 and console. If I try to invert width and height, it's the same problem.

If I get back to console only boot with display_rotate=1, the screen is incomplete: it as if the console layout/buffer is larger than what is displayed.

Gadgetoid, 2019-09-25:
For me with `display_rotate=1` I get the "Welcome to the Raspberry Pi Desktop" splash stuck permenantly on the display, and while I can hit "ctrl+alt+f1,2,3" etc to switch terminals they only momentarily flash up before they are overdrawn again by the splash.

This would appear to be a change/bug in the way that the Pi 4 handles DPI displays and the `display_rotate` option.

I wonder if there's an alternate way to rotate the display in console mode. The Pi 4 has both a `/sys/class/graphics/fb0/rotate` and a `/sys/class/graphics/fbcon/rotate` but echoing any values into these does nothing. They also both read 0 even with a `display_rotate` value set.

illydone, 2019-10-02:
I made other tests and the screen seems definitely a portrait screen in console... :/
I don't know if there's a way to change it. Do you know if pimoroni works on it?

PS: I also test a graphical Kali Linux and hyperpixel4-rotate doesn't work at all, so no console and even no desktop in lanscape.

amnesia0287, 2019-10-09:
I’ve also encountered the same thing on my pihole. It’s some sort of missing dependency. Gives an error about failing to import ScreenLayout. I just haven’t sat down to figure out what the dependency is. I can’t imagine it’s anything that can’t just be installed tho.

supersjimmie, 2019-10-25:
I landed here without a hyperpixel, with just some cheap 4 inch chinese hdmi display.
Same here, adding the rotate_display=3 (or 1) makes my rPI4 hang before starting the desktop.
Just adding this to let you know it's most probably not hyperpixel related but something between rap4 and the latest(?) raspbian.

xDSticker, 2019-10-27:
I have the same problem. I want the raspberry to work in landscape mode while in the console but stuck there.

Gadgetoid, 2019-11-04:
This may have something to do with the fact that there's one "pixel fountain" (pixel pipeline? pixel bus?) reserved for rotation on the Pi 4, and since the HDMI is still active when using DSI or DPI it's probably reserved for HDMI. This would- at a very tenuous guess- mean that the Pi 4 is not configured to support rotation over DPI...  but then surely since it's possible over DSI (the ribbon cable to the official Pi display) (I assume) it should be possible over DPI.

But yes, like @supersjimmie suggests this isn't a Pi 4 specific problem and we should probably narrow down the scope of the issue (works on DSI/DPI?), report it and cross our fingers for a fix.

Grabbing an official display to test now.

tusing, 2019-11-08:
Anyone have any luck with this? Same issue, when rotating to a landscape mode my display is overdrawn with a static image of a console.

The "desktop" behind this image still works, but is obscured by the image and this is useless.

Gadgetoid, 2019-11-09:
Has anyone tried `display_rotate` this without `dtoverlay=vc4-fkms-v3d` in `/boot/config.txt`?

It's clear this issue isn't confined to HyperPixel - https://www.raspberrypi.org/forums/viewtopic.php?t=246146
and: https://raspberrypi.stackexchange.com/questions/102182/display-rotate-3-only-filling-half-screen-raspberry-pi-4

Both suggesting that `vc4-fkms-v3d` seems to break this setting.

Zokol, 2019-11-10:
> Has anyone tried `display_rotate` this without `dtoverlay=vc4-fkms-v3d` in `/boot/config.txt`?

Tried this and it worked!
I'm running 2019-09-26 Raspbian Buster Lite

/boot/config.txt:
```
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
display_rotate=1
```

illydone, 2019-11-15:
> Has anyone tried `display_rotate` this without `dtoverlay=vc4-fkms-v3d` in `/boot/config.txt`?
>
> It's clear this issue isn't confined to HyperPixel - [...]

I tested it with my "no so" fresh install of Raspbian. I first did a full upgrade and eeprom upgrade but it doesn't work. In console it's okay in portrait mode however as soon as I use display_rotate, the system is stuck on the splash screen.
However I'll try to reinstall/refresh the config as I also have some problems with the touch function in X sessions (portrait and landscape).

winterelf, 2020-01-17:
the following combination worked issue:
dtoverlay=hyperpixel4:rotate
display_rotate=1

based on this :
(https://github.com/pimoroni/hyperpixel4/issues/8)

and also in the docs: https://github.com/pimoroni/hyperpixel4

mikekscholz, 2020-01-17:
Just to note, display rotate is depreciated now and has been for a while. Ive been using it with display_lcd_rotate just fine.

cannikin, 2020-02-06:
I got the screen to rotate with `display_lcd_rotate=3` (so that the power/hdmi ports are at the bottom) but the touchscreen doesn't respond. I don't think it's rotated or anything weird because no matter where I touch nothing happens.

I put the hyperpixel on a Pi4 running Octopi and [Octoscreen](https://github.com/Z-Bolt/OctoScreen). As part of the Octoscreen install I had to install dependencies `libgtk-3-0 xserver-xorg xinit x11-xserver-utils` is there anything else missing? I installed those AFTER the hyperpixel drivers, maybe I need to reverse the order?

mikekscholz, 2020-02-06:
Xserver-xorg-input-evdev for one... ive experimented with octoscreen but the codes not quite to my liking yet. But the way it runs ratpoison without having the raspbian desktop installed I’m not sure how well the Hyperpixel drivers may or may not respond. Id be interested to hear if you have any success though.

cannikin, 2020-02-06:
@mikekscholz Thanks for the suggestion! Installed `xserver-xorg-input-evdev` successfully but still no touch after reboot. :(

Is a foolproof fix to just install the full Raspian Desktop from scratch and then install Octoprint/Octoscreen after the fact?

mikekscholz, 2020-02-06:
Octoprint has a command listed in the console when it finally stops to where you can start typing that you type in to install the desktop environment. But that kind of defeats the purpose of using octoscreen which is meant to be as light as possible on system resources. Personally ive been leaning toward octodash however I’m using it on a touchscreen with usb interface. Ive not tried it on my hyperpixel

cannikin, 2020-02-06:
I started this whole journey from this tweet: https://twitter.com/jerware/status/1215445584305258496?s=21 Dude has a Pi4 and Hyperpixel4, used the official [OctoPi 0.17.0](https://octoprint.org/blog/2019/11/14/octopi-release-0-17-0/) image and everything was great for him. Not sure what I'm missing...

cannikin, 2020-02-06:
Octodash looks great, maybe I'll give that a try!

mikekscholz, 2020-02-06:
Its still in development, ive been doing test builds and tweaking it to my personal taste but its functional and its getting close to being nice and polished. I want to like octoscreen but I cant get it to not crash constantly. I hope that gets ironed out soon. Besides that take a look at octoprint-touchui and touchui autostart. Thats my main workhorse interface as these others come to completion

cannikin, 2020-02-06:
Looks like octodash also requires a desktop install? It's complaining about `xinit` not being found.

mikekscholz, 2020-02-06:
It requires some minimal software but nothing close to a full desktop. Its an electron app thats stand-alone that boots directly from the comandline

cannikin, 2020-02-06:
Just go it installed but touch still isn't working. Started with a fresh install of OctoPi, too. Even used that command that Octopi gave me to install the desktop, then re-installed the hyperpixel drivers but still nothing. UGH

cannikin, 2020-02-07:
Holy crap it was this: https://github.com/pimoroni/hyperpixel4/issues/41 Whether it worked or not depended on how warm the Pi was at bootup!

sepia-assistant, 2020-02-09:
Some more info about this issue can be found in #51 : 
https://github.com/pimoroni/hyperpixel4/issues/51#issuecomment-583848997

diracsbracket, 2020-10-27:
> The Pi 4 has both a `/sys/class/graphics/fb0/rotate` and a `/sys/class/graphics/fbcon/rotate` but echoing any values into these does nothing.

The following works for me on Pi4:
`
echo 1 | sudo tee -a /sys/class/graphics/fbcon/rotate > /dev/null
`

LordRembo, 2020-11-08:
So, you can rotate the screen in Pi 4 if you comment out `dtoverlay=vc4-fkms-v3d` but that breaks a bunch of stuff (like 3D acceleration and some stuff that RetroPie needs to work), so that's not a good enough option.
Has anyone figured out a way to rotate the screen without messing up the drivers?

I found a topic [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=247619&p=1516639#p1516639) that says you need to either:
- a GPU to set rotation via settings, but I don't have that (I have Raspbian Lite + RetroPie),
- or 'set rotation on the kernel command line', which is in `/boot/cmdline.txt` (but I don't understand how it works)

LordRembo, 2020-11-08:
Edit for the last commit: making changes to rotation via `cmdline.txt` did do something. It makes the content 'think' it's on a rotate screen, causing half of it to be cut off.
Any ideas or suggestions or alternative means of rotation WITHOUT removing `vc4-fkms-v3d` would be most helpful.

kalbanowski, 2021-04-11:
Sorry for being off-topic to fixing the hyperpixel4 rotation itself, but for what it's worth to the people specifically trying to get OctoDash rotated with the Hyperpixel 4.0 (rectangular) here a little bit in the future with OctoPi 0.18 and OctoDash 2.1.2, octodash _is_ using X11, just not running a desktop or window manager. So it's easy to edit /home/pi/.xinitrc to set the screen rotation as X is booting up:

```
#!/bin/sh

xset s off
xset s noblank
xset -dpms

ratpoison&
xrandr --output DSI-1 --rotate left      ### added
octodash
```

I haven't tried this from a clean install, but I believe you should be able to run `hyperpixel4-rotate left` from any prompt to set up the touch rotation, then the modified .xinitrc will take care of the display rotation.

To test whether display and touch are matching, it's easy to run `xeyes` from any shell once X has been started (you'll likely need to install the x11-apps package):
```
DISPLAY=:0.0 xeyes
```

The eyes will 'look' at any touch, so now you can see where the pointer is. Ctrl-C to kill the eyes.

illydone, 2021-04-18:
Thank you LordRembo for pointing this similar issue:
https://github.com/pimoroni/hyperpixel4/issues/111

Alphex gives a solution that worked for me: last RaspberrypiOS update on an old 1GB Pi4 (original install was a Raspbian lite).
Use branch "Rectangular - Pi 4 - Experimental Touch Fix"

With these parameters in /boot/config.txt, it works and the Pi is booting in landscape directly:
'
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
# dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
display_rotate=3

dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
dtoverlay=hyperpixel4-0x5d
`
I tried to add the desktop on this Raspbian Lite and change the boot options and I also get the UI in landscape mode with the touch working perfectly.

However hyperpixel4-rotate don't work any more and I can't change the orientation dynamically. However, if somebody confirms, I think this issue can be closed.

LordRembo, 2021-04-18:
This is again a fix that disables the original overlay,  which then breaks 3D-acceleration (as I mentioned in my comment before) . I don't agree that this counts as a proper fix  if you lose original functionality. A partial fix is not a fix. So the issue should remain open.

LordRembo, 2021-04-18:
Repost from the related thread:
The only fixes that seem work semi-consistently,  are the ones that involve first removing the original vc4-fkms-v3d overlay. Since that removes original functionality (eg. 3D acceleration) and the core problem is Pi4 hardware based (there's links in previous comments and related threads), I think the best course of action is to 1) document the suggested fixes in this project's install info (including a caveat about losing functionality) and 2) throw in the towel and admit this is just not fully fixable for this screen + device combo. You need to buy a different screen or use a Pi3 to use horizontal orientation is you absolutely need 3D acceleration or run things like RetroPie.

illydone, 2021-04-18:
You're right. I don't need 3D nor UI for my main use of this screen (as a nano laptop) but it is a real problem.

Too bad as this Hyperpixel4 is really a nice screen and with a 3D printed case (as there's no official solution for this neither), IMHO there's no real equivalent with this size and quality.

Gadgetoid, 2021-04-19:
It's like squeezing water from a stone, but I'll see if we can go up the chain and find any sign of a vflip/hflip register in the display controller. This might allow us to achieve rotation at the hardware level and bypass the limitations of the Pi 4's GPU.

DaveTSG, 2021-05-02:
> Sorry for being off-topic to fixing the hyperpixel4 rotation itself, but for what it's worth to the people specifically trying to get OctoDash rotated with the Hyperpixel 4.0 (rectangular) here a [...]

For what it's worth, this has solved the issue on my Pi 4 running Raspbian Lite (OctoPi v0.17.0), and I can now use OctoDash perfectly with my Hyperpixel 4 screen. (Previously, I was unable to rotate the screen at all, and the touches that I did were recognised as being in different places, so I would have to touch 2cm away from the button I wanted to press, for instance).
...I struggled with this for a good six weeks, so I am glad to have finally found an answer. :)

closed 2026-08-25 by Gadgetoid