pimoroni/hyperpixel4 #40: bcm2835_init: gpio mmap failed: Cannot allocate memory

AndreasBrauckmann, 2019-09-09:
i have the same errorin a closed issue ... what am I doing wrong? Mmessage when trying to install the software on a brand new rpi4

error: **bcm2835_init: gpio mmap failed: Cannot allocate memory**, does anyone have an explanation for this?

Installation:
pi@raspberrypi:~ $ git clone https://github.com/pimoroni/hyperpixel4
```
Cloning into 'hyperpixel4'...
remote: Enumerating objects: 105, done.
remote: Counting objects: 100% (105/105), done.
remote: Compressing objects: 100% (83/83), done.
remote: Total 166 (delta 59), reused 50 (delta 21), pack-reused 61
Receiving objects: 100% (166/166), 462.97 KiB | 344.00 KiB/s, done.
Resolving deltas: 100% (82/82), done.
```
pi@raspberrypi:~ $ cd hyperpixel4
pi@raspberrypi:~/hyperpixel4 $ sudo ./install.sh

```
Notice: building hyperpixel4.dtbo
Created symlink /etc/systemd/system/multi-user.target.wants/hyperpixel4-init.service → /etc/systemd/system/hyperpixel4-init.service.
*** Job for hyperpixel4-init.service failed because the control process exited with error code.
See "systemctl status hyperpixel4-init.service" and "journalctl -xe" for details.
Installed: /usr/bin/hyperpixel4-init
Installed: /etc/systemd/system/hyperpixel4-init.service
Installed: /boot/overlays/hyperpixel4.dtbo
Config: Added dtoverlay=hyperpixel4 to /boot/config.txt
Config: Added overscan_left=0 to /boot/config.txt
Config: Added overscan_right=0 to /boot/config.txt
Config: Added overscan_top=0 to /boot/config.txt
Config: Added overscan_bottom=0 to /boot/config.txt
Config: Added framebuffer_width=800 to /boot/config.txt
Config: Added framebuffer_height=480 to /boot/config.txt
Config: Added enable_dpi_lcd=1 to /boot/config.txt
Config: Added display_default_lcd=1 to /boot/config.txt
Config: Added dpi_group=2 to /boot/config.txt
Config: Added dpi_mode=87 to /boot/config.txt
Config: Added dpi_output_format=0x7f216 to /boot/config.txt
Config: Added display_rotate=3 to /boot/config.txt
Config: Added hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6 to /boot/config.txt
```
pi@raspberrypi:~/hyperpixel4 $ journalctl -xe
```
-- The job identifier is 789.
Sep 09 18:16:28 raspberrypi systemd[1]: Mounted Arbitrary Executable File Formats File System.
-- Subject: A start job for unit proc-sys-fs-binfmt_misc.mount has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit proc-sys-fs-binfmt_misc.mount has finished successfully.
-- 
-- The job identifier is 789.
Sep 09 18:16:35 raspberrypi sudo[1166]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/find / -iname *hyperdisplay*
Sep 09 18:16:35 raspberrypi sudo[1166]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Sep 09 18:16:48 raspberrypi sudo[1166]: pam_unix(sudo:session): session closed for user root
Sep 09 18:17:01 raspberrypi CRON[1178]: pam_unix(cron:session): session opened for user root by (uid=0)
Sep 09 18:17:01 raspberrypi CRON[1182]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Sep 09 18:17:01 raspberrypi CRON[1178]: pam_unix(cron:session): session closed for user root
Sep 09 18:17:40 raspberrypi sudo[1196]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/curl get.pimoroni.com
Sep 09 18:17:40 raspberrypi sudo[1196]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Sep 09 18:17:41 raspberrypi sudo[1196]: pam_unix(sudo:session): session closed for user root
Sep 09 18:19:19 raspberrypi sudo[1219]:       pi : TTY=pts/0 ; PWD=/home/pi/hyperpixel4 ; USER=root ; COMMAND=./install.sh
Sep 09 18:19:19 raspberrypi sudo[1219]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Sep 09 18:19:20 raspberrypi systemd[1]: Reloading.
Sep 09 18:19:20 raspberrypi systemd[1]: Reloading.
Sep 09 18:19:21 raspberrypi systemd[1]: Starting HyperPixel 4.0" LCD Display Initialization...

-- Subject: A start job for unit hyperpixel4-init.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
[... 216 lines cut ...]
       configuration: broadcast=yes driver=brcmfmac driverversion=7.45.154 firmware=01-4fbe0b04 ip=192.168.0.111 multicast=yes wireless=IEEE 802.11
  *-network:1
       description: Ethernet interface
       physical id: 3
       logical name: eth0
       serial: dc:a6:32:18:70:34
       size: 10Mbit/s
       capacity: 1Gbit/s
       capabilities: ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=bcmgenet driverversion=v2.0 duplex=half link=no multicast=yes port=MII speed=10Mbit/s
```

pi@raspberrypi:~ $ gpio -v
```
gpio version: 2.50
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Unknown17, Revision: 01, Memory: 0MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.
```

AndreasBrauckmann, 2019-09-10:
Solved, fixing parameters in /boot/config.txt: 

```
# display can output without overscan
disable_overscan=1

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
dtparam=spi=off

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# enable raspicam
gpu_mem=128

[all]
dtoverlay=hyperpixel4
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
display_rotate=3
gpio=0-25=a2
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```
pi@raspberrypi:~ $ cat /etc/udev/rules.d/98-hyperpixel4-square-calibration.rules 
```
ATTRS{name}=="generic ft5x06 (11)", ENV{LIBINPUT_CALIBRATION_MATRIX}="-1 0 1 0 -1 1"
ATTRS{name}=="EP0110M09", ENV{LIBINPUT_CALIBRATION_MATRIX}="-1 0 1 0 -1 1"
```

illydone, 2019-09-11:
It seems you didn't use the "pi4" branch. I didn't have any problem when using it instead of the "master" branch.

Gadgetoid, 2019-10-15:
I've since updated the "master" branch of this repository to document the different branches for Pi 3, Pi 4 and Square/Rectangular. Hopefully this will curb future issues where the wrong software gets installed.

closed 2019-10-15 by Gadgetoid