pimoroni/hyperpixel4 #41: Touchscreen fails to connect at boot

neutralinsomniac, 2019-09-16:
My touchscreen fails to init 9 times out of 10 on boot. diff between a bad boot and a successful boot below. Anecdotally, it seems to work more reliably after the pi has been turned off for a while, but I only have 2 successes to go by. Heat issue maybe? This is on a pi4.

```
pi@raspberrypi:~ $ diff -Nurd <(grep -i good nowork.dmesg) <(grep -i good worked.dmesg)
--- /dev/fd/63	2019-09-15 20:24:02.790295364 -0400
+++ /dev/fd/62	2019-09-15 20:24:02.790295364 -0400
@@ -1,3 +1,2 @@
-[    5.002008] Goodix-TS 7-0014: i2c test failed attempt 1: -6
-[    5.042385] Goodix-TS 7-0014: i2c test failed attempt 2: -6
-[    5.081746] Goodix-TS 7-0014: I2C communication failure: -6
+[    4.721578] Goodix-TS 7-0014: ID 911, version: 1060
+[    4.747144] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-7/7-0014/input/input0
```

Gadgetoid, 2019-09-16:
Does it work reliably when it *does* init? Looks like it could be a connectivity issue, but this isn't something I've seen before.

neutralinsomniac, 2019-09-16:
Yeah, works perfectly unless I reboot. I've used it handheld for at least a couple hours with zero issues. Reboot, and it fails to init again.

malcolmholmes, 2019-09-19:
I am seeing the same error, and touchscreen failure. Will retry once the device has cooled down.

I have tried both the installs via `./install.sh`, and via `curl | bash`. I have had the touchscreen work (but with touch offset in both X and Y), but whenever I reinstall, it stops working. I have not yet been able to identify the conditions for success/failure, although I have noted the precise above Goodix error.

malcolmholmes, 2019-09-19:
It didn't take long to cool down, and the touchscreen works on boot (well, touch events happen, they're just in completely the wrong place on the screen! But that's the next problem)

R0B3RDV, 2019-09-25:
I can confirm that, if the the screen is a bit hot, and you reboot it sometimes, you could be out of luck and have no touchscreen. Only solution is to cool it down and then cold boot ?

R0B3RDV, 2019-09-25:
ps. i assume now that the malfunction of touch function is heat related, because i tried to reboot some times with a machine that was just started and then it works fine

R0B3RDV, 2019-09-25:
I found this in the log:
Sep 25 22:20:48 rbp4 kernel: [    4.665872] Goodix-TS 7-0014: i2c test failed attempt 1: -6
Sep 25 22:20:48 rbp4 kernel: [    4.702365] Goodix-TS 7-0014: i2c test failed attempt 2: -6
Sep 25 22:20:48 rbp4 kernel: [    4.742673] Goodix-TS 7-0014: I2C communication failure: -6
Sep 25 22:25:23 rbp4 kernel: [    4.833540] Goodix-TS 7-0014: i2c test failed attempt 1: -6
Sep 25 22:25:23 rbp4 kernel: [    4.872829] Goodix-TS 7-0014: i2c test failed attempt 2: -6
Sep 25 22:25:23 rbp4 kernel: [    4.911787] Goodix-TS 7-0014: I2C communication failure: -6
Sep 25 22:32:31 rbp4 kernel: [    4.794848] Goodix-TS 7-0014: i2c test failed attempt 1: -6
Sep 25 22:32:31 rbp4 kernel: [    4.837589] Goodix-TS 7-0014: i2c test failed attempt 2: -6
Sep 25 22:32:31 rbp4 kernel: [    4.872225] Goodix-TS 7-0014: I2C communication failure: -6
Sep 25 22:40:30 rbp4 kernel: [    4.736812] Goodix-TS 7-0014: i2c test failed attempt 1: -6
Sep 25 22:40:30 rbp4 kernel: [    4.778585] Goodix-TS 7-0014: i2c test failed attempt 2: -6
Sep 25 22:40:30 rbp4 kernel: [    4.812514] Goodix-TS 7-0014: I2C communication failure: -6
Sep 25 22:43:07 rbp4 kernel: [    4.759841] Goodix-TS 7-0014: i2c test failed attempt 1: -6
Sep 25 22:43:07 rbp4 kernel: [    4.792305] Goodix-TS 7-0014: i2c test failed attempt 2: -6
Sep 25 22:43:07 rbp4 kernel: [    4.831887] Goodix-TS 7-0014: I2C communication failure: -6

R0B3RDV, 2019-09-26:
Hi, I did some testing: on RbPi 4 with the latest release of Rasp. full version: I've tried to find out why the display was working during boot and not restart, and I can conclude that it has definitly something to do with the heath of the display, because i've tried like 20 reboots with and without an extra fan (external powersupply , so just for airflow) and the number of times that i restarted the Pi with an airflow for cool air, was like 0 times that i could not access the system via touchscreen after an reboot, against 18 times that I could not access the system via the display even my pi was not warmer then 50 degrees celcius. I dont know if this is the case with rbpi 3 too ?

Gadgetoid, 2019-09-30:
@R0B3RDV would these latest findings conclude that it's less related to heat and more related to some other issue with the display? I run my tests on a Pi which I tore the heat spreader off and now have poorly coupled to one of our big metal heatsink cases. It runs hot, albeit our ambient air temperature is not particularly extreme, so I might have expected to see some indication of this failure.

I can definitely try to overheat a display and see what happens, though.

It might be worth probing the i2c bus during this failure condition and seeing what you can see. Touch runs over a soft i2c bus `i2c-gpio`  that usually shows up as `i2c-3` but you can check which by running `ls /dev/i2c*` and observing the result.

Based on the assumption that it'll be 3, you can then probe the bus with `i2cdetect`:

```
i2cdetect -y 3
```

R0B3RDV, 2019-10-01:
i will shortly update some more info if you like, but i've upgraded the raspberry pi 4 now with a ninja coupe, (i had a passive cooler but i now replaced it with the active fanshim - without installing the fanshim drivers offcouse), installed the display ontop and working as a breeze now! :) with the passive cooler, i was thinking that maybe a particular place on the bottom of the screen was getting to warm, but so, i wasnt even using the board...

R0B3RDV, 2019-10-01:
but now i am having no trouble rebooting etc even after some while, no touchscreen malfunctions etc. I even tried to reverse the fan, passive fan ontop of the chip so that the airflow is to the bottom side of the screen and the chips are getting the warm air sucked away via passive and partly active cooling.

Gadgetoid, 2019-10-02:
I've been running some tests of my own to see if I can replicate.

If you have another failure condition, can you tell me what you see if you run:

```text
i2cdetect -y 7
```

You should see something like:

```text
pi@raspberrypi:~ $ sudo i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

herodk, 2019-10-08:
I am having the same problem with a Hyperpixel 4.0 Square on a PI zero W running Rasbian stretch:

```
[   20.539692] Goodix-TS 3-005d: i2c test failed attempt 1: -6
[   20.574850] Goodix-TS 3-005d: i2c test failed attempt 2: -6
[   20.613675] Goodix-TS 3-005d: I2C communication failure: -6
```

Any update on this?

Gadgetoid, 2019-10-15:
@herodk what do you see, in this case, when you run: `sudo i2cdetect -y 3` (where 3 is the i2c bus number revealed by listing enabled busses with: `ls /dev/i2c-*`

There's an issue between the Pi 4 and Pi 3B+ where some pin initialisation on boot causes the i2c address to change between one and the other. So far I've only seen:

1. Pi 3B+ sees the touch on `0x5d`
2. Pi 4 sees the touch on `0x14`
3. Hot plugging on Pi 4 will show touch on `0x5d`

This is partly why we have a Pi 3 and Pi 4 branch for the HyperPixel 4- these respective addresses seem to be stable.

However - and this is something of a guess, since I don't know what's really at play here yet - It's possible that this same issue is somehow affecting you intermittently.

PinkSnake, 2019-10-17:
@neutralinsomniac 
Same issue, more info ?

PinkSnake, 2019-10-17:
> > @neutralinsomniac
> > Same issue, more info ?
> 
> yep. just posting what was requested by @Gadgetoid

Ok so fixed on my side:
0. Uninstall hyperpixel driver
1. double check if old dtbo are in orverlay folder, if yes remove them
1. reinstall driver (curl ... | bash)
2. Disable raspberry i2c bus (raspi-config)
3. Reboot
4. check:
```
pi@rpi4:~ $ dmesg | grep Goodix
[    4.636222] Goodix-TS 7-0014: ID 911, version: 1060
[    4.677918] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-7/7-0014/input/input0
```

```
pi@rpi4:~ $ sudo i2cdetect -y -r 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

neutralinsomniac, 2019-10-17:
Is goodix.dtbo one of the files installed by the hyperpixel script?

neutralinsomniac, 2019-10-17:
anyway, same issue even after disabling i2c in raspi-config. touchscreen fails to be found:
```
[    5.249019] Goodix-TS 7-0014: i2c test failed attempt 1: -6
[    5.283963] Goodix-TS 7-0014: i2c test failed attempt 2: -6
[    5.321788] Goodix-TS 7-0014: I2C communication failure: -6
```

PinkSnake, 2019-10-17:
> anyway, same issue even after disabling i2c in raspi-config. touchscreen fails to be found:
> 
> ```
> [    5.249019] Goodix-TS 7-0014: i2c test failed attempt 1: -6
> [    5.283963] Goodix-TS 7-0014: i2c test failed attempt 2: -6
> [    5.321788] Goodix-TS 7-0014: I2C communication failure: -6
> ```

Try that: Power down the board, remove the screen, plug again and boot

neutralinsomniac, 2019-10-17:
> Try that: Power down the board, remove the screen, plug again and boot

Tried it. No dice. I still think this is heat related, as myself and others have observed.

PinkSnake, 2019-10-17:
Could you please give us : **$ ls /dev/i2c*** because seems i2c issue :S

neutralinsomniac, 2019-10-17:
> Could you please give us : **$ ls /dev/i2c*** because seems i2c issue :S

```
pi@raspberrypi:~ $ ls /dev/i2c*
/dev/i2c-7
```

PinkSnake, 2019-10-17:
@neutralinsomniac 
Same on my side, i don't have any idea ... sorry

Gadgetoid, 2019-11-05:
Further to my earlier post I've - a couple of times - seen the same issue as @neutralinsomniac where the Pi 4 will start up with the touch on `0x5d` (the correct address in theory) rather than `0x14`. I still - despite some time with an oscilloscope and constant reboots of the Pi 4 - haven't been able to narrow down exactly what causes it, however.

My hypothesis is that the `0x14` -> `0x5d` issue appears to be related to the touchscreen driver chip having an i2c select feature that relies on the interrupt pin being pulled one way or another on init. It would seem that in some cases the Pi 4 affects this pin, whereas in other cases it does not. My gut feeling would be some sort of start-up race condition between the touch IC and the Pi... I suspect temperature could possibly affect this.

gitsh01, 2020-01-22:
Fix confirmed.

Just letting you know I had this exact issue - touch only working sometimes on a raspberry pi 4 running Octoprint Buster.

Your patch git clone https://github.com/pimoroni/hyperpixel4 -b pi4-i2c-fix fixed my issue perfectly.

Thank you for finding and fixing the problem. Like others above, the hyperpixel was showing up as a 0x5d on boot with no physical touch involved.

gitsh01, 2020-01-22:
If it helps, my pi is in an enclosure powering a Prusa MK3s, which can get very hot, although right now it's only reading 20C because I haven't started a print job yet. I was doing soft reboots as I tried to fix other issues. I control it via ssh, there's no physical keyboard or other monitor attached, and I have two USB devices attached to it: the Prusa MK3s and a Logitech C910.

mikekscholz, 2020-01-22:
I cant speak for the pi 4, but on a 3b+ with the latest buster release, my touch coordinates were all over and needed manual adjustment to the values obtained from xinput-calibrator, but installing the alternate goodix driver from the Hyperpixel4Touchscreen repository gave perfect touch points, like perfect to the point of needing to delete 99-calibration.conf to not screw up what was already correct.

jcansell, 2020-02-28:
Thanks - i've added the pull-up resistor.
Can you just add the current board version to this thread so that future versions of myself don't mod a later release of the product :-)

kkolk, 2020-11-29:
Just wanted to say I had this issue with a Raspberry Pi4 and a HyperPixel4 and couldn't for the life of me understand why on some reboots it was just failing but would work the next day. 

Had it constantly failing, found this, installed it, immediately worked on reboot.

thirdr, 2025-01-08:
Closed as resolved.

closed 2025-01-08 by thirdr