pimoroni/hyperpixel4 #42: Touchscreen does not work the 1cm area to the right

dustyhorizon, 2019-09-19:
Just got my hyperpixel 4 today for my raspberrypi 4 and had got it working. However I noticed that the touchscreen does not work about 1cm to the right after rotating to the left. Any ideas?

Gadgetoid, 2019-09-19:
You should check in Preferences -> Screen Configuration and see if the display is being overlapped at right angles to one of the HDMI desktops. The touch input is scaled to the size of the *entire* desktop and not just the screen itself.

If you see your (disconnected) HDMI-0 and HyperPixel screens overlapping at 90 degrees then rotate the HDMI-0 to match, make sure they cleanly overlap and that should- hopefully- fix the problem.

dustyhorizon, 2019-09-19:
Had a look under Screen Configuration but I do not see any other screens other than DSI-1 (suppose that is the hyperpixel 4).

Attached is a picture of the Screen Configuration window. Also in that picture my pointer is at the 'boundary' that I am hitting with the touchscreen.

[image]

Gadgetoid, 2019-09-19:
Well that's strange!

It may well be that your input calibration is somehow not scaled to the full width of the screen on that axis, but I'm not sure why that would be the case. Curiously that's about where the cursor might end up on a 720x720 Square Hyperpixel 4, coincidence?

Try:

```
xinput list-props "pointer:Goodix Capacitive TouchScreen"
```

And stick a screenshot of the output here.

dustyhorizon, 2019-09-19:
@Gadgetoid here you go!

```
pi@raspberrypi:~ $ DISPLAY=:0.0 xinput list-props "pointer:Goodix Capacitive TouchScreen"
Device 'Goodix Capacitive TouchScreen':
        Device Enabled (135):   1
        Coordinate Transformation Matrix (136): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Calibration Matrix (267):      0.000000, -1.000000, 1.000000, 1.000000, 0.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Calibration Matrix Default (268):      0.000000, -1.000000, 1.000000, 1.000000, 0.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Send Events Modes Available (269):     1, 0
        libinput Send Events Mode Enabled (270):        0, 0
        libinput Send Events Mode Enabled Default (271):        0, 0
        Device Node (272):      "/dev/input/event0"
        Device Product ID (273):        1046, 911
```

I installed hyperpixel 4 by first git cloning the repo followed by checking out the pi4 branch after which the steps as detailed on that branch.

Gadgetoid, 2019-09-20:
Looks like your ` libinput Calibration Matrix Default` differs from mine, but I'm not sure if/why that would cause the result you're seeing and, as far as I'm aware, it can't be changed using `xinput set-prop`

```
pi@raspberrypi:~ $ DISPLAY=:0.0 xinput list-props "pointer:Goodix Capacitive TouchScreen"
Device 'Goodix Capacitive TouchScreen':
        Device Enabled (127):   1
        Coordinate Transformation Matrix (128): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Calibration Matrix (259):      0.000000, -1.000000, 1.000000, 1.000000, 0.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Calibration Matrix Default (260):      -1.000000, 0.000000, 1.000000, 0.000000, -1.000000, 1.000000, 0.000000, 0.000000, 1.000000
        libinput Send Events Modes Available (261):     1, 0
        libinput Send Events Mode Enabled (262):        0, 0
        libinput Send Events Mode Enabled Default (263):        0, 0
        Device Node (264):      "/dev/input/event3"
        Device Product ID (265):        1046, 911
```

In Right/Normal/Inverted rotation can you reach all the edges of the screen?

Gadgetoid, 2019-09-20:
Nope- nothing to do with `libinput Calibration Matrix Default`, I rebooted and my output now matches yours.

I see `HDMI-1` in my Screen Configuration since I have `hdmi_force_hotplug=1` in my `/boot/config.txt`, will try changing that.

dustyhorizon, 2019-09-20:
Regarding `In Right/Normal/Inverted rotation can you reach all the edges of the screen?`: None of the modes allow me to touch that specific area.

I tried to set `hdmi_force_hotplug=1` too but I couldn't see the menu to enter Screen Configuration on the hyperpixel.

MikaelX, 2019-09-20:
I have this exact same problem, fresh install of buster. Installed hyperpixel4 through the bash download on my pi4, 4gb

[image]

While fiddling around with the matrixes I've found that there's no problem with the actual screen, I managed to get it to respond within the area (right side 1-2 cm) but then of course not calibrated properly.

Any ideas?

Gadgetoid, 2019-09-23:
Looks like I'm going to have to install a fresh Raspbian Buster again and see if I can replicate. Off the top of my head I can't think of anything that could have changed to cause this specific an problem with touch input.

Gadgetoid, 2019-09-24:
I can't seem to replicate this touch deadzone- I wonder if it's an issue with the screen itself.

@dustyhorizon from your photo it looks like the boundary is on the far side of the touchscreen from the ribbon cable. Can you see any damage to the smaller ribbon cable on your HyperPixel? The part of the cable that's curved, before it goes into the onboard driver IC has lots of very small electrode connections on it, and it's possible that some damage from the bend, or a tear in that part of the ribbon could disconnect some of the electrodes but not all.

@MikaelX I'm not convinced it's a software issue, but this touchscreen calibration is nothing if not confusing. I wonder if there's any way to determine if the screen can "see" a touch, even if the rest of the software is not working.

dustyhorizon, 2019-09-24:
> I can't seem to replicate this touch deadzone- I wonder if it's an issue with the screen itself.
>
> @dustyhorizon from your photo it looks like the boundary is on the far side of the touchscreen from [...]

Does not look like it is damaged in any way, could it be possible that this is a manufacturing defect?

Gadgetoid, 2019-10-08:
Sorry for the delay. Have you been in touch with support yet? Further attempts have yielded no success at re-creating this so I suspect it's a hardware fault. Software has a nasty habit of surprising me in situations like this, though.

dustyhorizon, 2019-10-09:
> Sorry for the delay. Have you been in touch with support yet? Further attempts have yielded no success at re-creating this so I suspect it's a hardware fault. Software has a nasty habit of surprising me in situations like this, though.

Ah not yet, was hoping if there are any outcome from your end, I guess I will drop support an email then. Thanks!

Gadgetoid, 2019-10-15:
Hopefully they'll get you sorted out! Sorry I couldn't find an easier fix. HyperPixel loves to keep me on my toes! Going to close this for now, but feel free to re-open this issue or create a new one if you have further problems.

closed 2019-10-15 by Gadgetoid

deanbayley, 2020-02-01:
Was there ever a resolution to this fault, i'm having the EXACT same issue.

Dean.

MikaelX, 2020-02-02:
Hi all

Sorry for the lack of response. I cancelled my efforts in this. I did try to reseat all the cables and connections but no luck. 

Might pick it up again later on.

deanbayley, 2020-02-02:
Yeah I tried the same with connectors also tried totally fresh installs of rasbian lite and raspian desktop, could only get the latter to rotate, tried several methods such as:

Formatted card, flashed the raspian desktop image, booted up and ran:

```
sudo apt-get update
sudo apt-get install git
git clone https://github.com/pimoroni/hyperpixel4.git -b pi4
cd hyperpixel4
sed -i 's/touchscreen-size-x\ =\ <800>/touchscreen-size-x\ \= <480>/g' src/hyperpixel4.dts
sed -i 's/touchscreen-size-y\ =\ <480>/touchscreen-size-y\ \= <800>/g' src/hyperpixel4.dts
sudo ./install.sh
```
_rebooted_ 

`hyperpixel4-rotate right`

right hand side still didn't work.

Then tried formatting card, flashed the raspian desktop image, booted up and ran:

`curl https://get.pimoroni.com/hyperpixel4 | bash`

_rebooted_ 

`hyperpixel4-rotate right`

right hand side still didn't work.

didn't even setup wifi or use SSH, all commands were typed manually, network running over ethernet.

Kawuezel, 2020-07-30:
Hi,

Is there still no solution for this? I'm having the exact same problem with one of my Hyperpixels. First it worked great, but now, no chance to get the touch working on the right side.

Should I check with support to get an exchange?

Regards

Gadgetoid, 2020-07-30:
Near as I can tell this is still a hard fault and there's nothing software can do to fix it. My working theory is that there's something wrong with the ribbon cables- perhaps one of the traces near the edge has broken and that just happens to correspond with the last vertical (or horizontal as it were) trace in the touch matrix. 

Unlike our older HyperPixel I can't get any low-level information out of the touch controller to figure out what might be at play here. I any case support@pimoroni.com should be better equipped to sort out replacements where they're needed.