pimoroni/hyperpixel4 #45: Could you please provide noob info ?

PinkSnake, 2019-10-08:
Is it possible to have a full updated README.md in order to be able to install and rotate the screen correctly with different raspi config ?

In my case I have spent one day on it and I can't have a correct portrait mode on rpi3b :(

Gadgetoid, 2019-10-14:
I think I've probably changed enough that some of the pointers/instructions here might now be obsolete.

That said, HyperPixel 4 is physically a portrait display, so it should work this way by default on a Pi 3B.

What were you aiming to do, what steps did you try and what happened?

Gadgetoid, 2019-10-14:
I'm making some headway in bringing the `hyperpixel4-rotate` script to Pi 3B. It has to work a different way, but should hopefully make things much easier to grok.

Gadgetoid, 2019-10-14:
As of merging #37 the steps to install on Pi 3B+ are detailed here: 

https://github.com/pimoroni/hyperpixel4/tree/pi3

This now includes the backported `hyperpixel4-rotate` command, which gives you a simple way to rotate both the screen/touchscreen in lockstep with each other.

PinkSnake, 2019-10-14:
@Gadgetoid 

If I Could Tell You Just One Thing: Thank you! 

After removing dtbo and /boot/config.txt entries, re-install is okay.
I tested the 4 options without any worry.

closed 2019-10-14 by PinkSnake