pimoroni/hyperpixel4 #46: White static/disturbance at bottom of Hyperpixel 4 Square

Dieseljesus, 2019-10-08:
Hi. Just upgraded from a normal Hyperpixel 4 (Which had huge problems with touch being offset). Now i just installed the new display and have a case of white static at the lower part of the screen. Its like the lowest line on the screen flashes every like 7-10 seconds. I have changed the boot config settings such as HDMI boost, tried to rotate it etc but to no avail.

The drivers are the standard drivers installed through curl. Im also running the latest Raspbian on the pi3b+.

Any tips to this? So far all products i´ve bought from you guys have been faulty in some way and i really want to love these products. I would love to get any solution to this. The screen is fantastic but this white static line is driving me insane!

Gadgetoid, 2019-10-10:
I haven't seen the likes of this- it could be a software conflict. Did you re-use your existing SD card (from Hyperpixel 4) and set up the new display on that?

Dieseljesus, 2019-10-10:
Hey, thanks for the help! :) I used a new installation and did the normal installation. Everything looks good. With the normal HP4 i had to change all scripts since the touch was off and the install after that. With this one it worked straight out of the box. I also think the disturbance might be gone.

One thing, im using it on top of the pijuice which might interfere? I have to run it some more to see if these problems are gone, but i had them for an hour or two and after that i think they dissapeared!

Once again, thanks for the help. Will see if i still have problems and get back to you!

Gadgetoid, 2019-10-14:
Good news! Thanks for updating me.

I'm still frustrated that the normal HyperPixel 4 is such a hassle, it seems for every proble I squash another two crop up. Sorry you had a hard time with it!

Gadgetoid, 2019-10-15:
Looks like PiJuice uses the i2c bus on BCM2 and BCM3. It's more likely that PiJuice itself would fail than HyperPixel, since HyperPixel uses these as VSYNC/HSYNC and disables the associated i2c bus.

Closing this for now- but re-open if it goes pear shaped again!

closed 2019-10-15 by Gadgetoid