pimoroni/hyperpixel4 #47: HyperPixel 4 Square Flickering White Line at Top

frank26080115, 2019-10-09:
HyperPixel 4 Square with Touch, installed on R-Pi 4 running Raspbian Buster

The top line of pixels is flickering white frequently, it is super annoying. I have a video of it here:

https://gfycat.com/madeupdarkcottontail

Gadgetoid, 2019-10-15:
Sorry for the delay responding- are you still seeing this problem? It's not something I've witnessed before.

Try rotating the display (using the Display Configuration Utility on Pi 4 or `display_rotate=2` on Pi 3) and seeing if the white flashing stays in the same place or moves with rotation. This would determine if it's a hardware or software glitch.

frank26080115, 2019-10-18:
I tried that today. This seems to only happen when in the "normal orientation", at the top. In any other orientation, it does not happen at all.

fjsantosb, 2019-12-18:
Hi,

I have the same issue when develop using Pygame. A white line start flickering at random as the video showed above (in this case at the bottom, but screen is rotated using display_rotate=2).

You can test it with this simple code:

`import pygame

pygame.init()
pygame.mouse.set_visible(False)
screen = pygame.display.set_mode((720, 720))

clock = pygame.time.Clock()

while True:
    screen.fill((0, 0, 0))
    for i in range(28):
        pygame.draw.rect(screen, (250, 40, 153), (i * 21, 0, 20, 20))
    pygame.display.flip()
    clock.tick(60)`

Btw, using HyperPixel 4 (squared - not touch) and rpi4.

svanstrom, 2019-12-27:
Just set up a RPI4 with the square HP; and noticed the annoying flashing white after already having rotated the screen. So to me the flashing is happening on the left side (ie pin-side).

svanstrom, 2020-01-05:
Very unscientifically researched… RPI0s don't show the same problem.

Always meant to use these display on zeros, and after getting some headers on to them and this morning doing the same basic install I noticed that I didn't get annoyed by a flashing white line.

So this problem is either unique to the RPI4s, or to the software used on them.

macaba, 2020-01-07:
Just confirming another report here - white line randomly appears every few seconds along the pin header side. Looking at it with a slow motion video shows the whole picture shifts by 1.

NODESPLIT, 2020-01-31:
Same here, with rotation 180ed I see the line at the bottom and get slight flicker. Definitely more prominent with terminal maximised (maybe because it's black).

ffbosh, 2020-02-11:
the same thing happens to me, it seems as if the screen were generated by adding a line on the top side and moving the entire screen 1 pixel down.
Pi4, Buster Lite, HyperPixel 4 Square

svanstrom, 2020-02-15:
Any official update on this from Pimoroni, @Gadgetoid ?

svanstrom, 2020-03-12:
Is this not treated as an official channel?

Gadgetoid, 2020-03-12:
It is but it's only monitored by me, and once issues are tagged "notice" they are filtered out of my GitHub issues view.

As near as I can tell from reading reports this is an issue with Raspbian and the Pi 4 display stack, and not specifically with HyperPixel 4. I'm not actually sure where to report this issue- perhaps https://github.com/raspberrypi/linux - but I will attempt to replicate it and try and reach out to some of our contacts.

Gadgetoid, 2020-03-12:
So far I've failed to reproduce the flicker, and haven't noticed it previously across multiple HyperPixel 4 / Pi 4 configurations. It must be something unique about the Raspbian/Firmware version on affected Pi's.

I'd appreciate it if anyone who's still encountering this problem could chime in with:

* Raspbian version Full/Lite
* UI- console, desktop, etc?
* Rotation in `hyperpixel4-rotate` terms (left, right, inverted, normal)

Plus copy and paste the output of:

* `cat /proc/cpuinfo | grep Revision`
* `uname -a`
* `vcgencmd version`
* `vcgencmd dispmanx_list`
* `cat /boot/config.txt`

And if your patience hasn't worn thin by now, also the output of:

* `dpkg -l rpi-eeprom-images`
* `dpkg -l libraspberrypi0`
* `dpkg -l raspberrypi-bootloader`

ffbosh, 2020-03-12:
**Raspbian Buster with desktop (2019-07-10-raspbian-buster)**

hyperpixel4-rotate does not work

Rotating display
xrandr: Failed to get size of gamma for output default
warning: output DSI-1 not found; ignoring
/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py:66: UserWarning: XRandR wrote to stderr, but did not report an error (Message was: 'xrandr: Failed to get size of gamma for output default\n')
  warnings.warn("XRandR wrote to stderr, but did not report an error (Message was: %r)"%err)
Saving display settings to /usr/share/dispsetup.sh
Setting matrix: 0 -1 1 1 0 0
unable to find device pointer:Goodix Capacitive TouchScreen
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules

**cat /proc/cpuinfo | grep Revision**
Revision        : b03111

**uname -a**
Linux raspberrypi4 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

**vcgencmd version**
Jan 30 2020 13:50:45
Copyright (c) 2012 Broadcom
version 8f792e011c8a0ea3c5d47e7cc10172cc10b93c09 (clean) (release) (start)

**vcgencmd dispmanx_list**
display:0 format:XRGB8888 transform:0 layer:-127 720x720 src:0,0,720,720 dst:0,0,720,720 cost:489 lbm:0
display:2 format:RGB565 transform:0 layer:-127 2x2 src:0,0,2,2 dst:80,0,480,480 cost:386 lbm:1024

**cat /boot/config.txt**
disable_overscan=0
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=4
dtparam=i2c_arm=off
dtparam=i2s=off
dtparam=spi=off
max_framebuffers=1
dtoverlay=hyperpixel4
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f226
dpi_timings=720 0 15 15 15 720 0 10 10 10 0 0 0 60 0 35113500 6
dtoverlay=gpio-poweroff,gpiopin=19,active_low=1
display_rotate=0
display_default_lcd=1

**dpkg -l rpi-eeprom-images**
||/ Nombre                      Versión      Arquitectura Descripción
ii  rpi-eeprom-images     4.0-1        all                   Raspberry Pi 4 boot EEPROM images

**dpkg -l libraspberrypi0**
||/ Nombre          Versión                Arquitectura     Descripción
ii  libraspberrypi0 1.20200205-1     armhf                EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV

**dpkg -l raspberrypi-bootloader**
||/ Nombre                            Versión                     Arquitectura           Descripción
ii  raspberrypi-bootloader     1.20200205-1          armhf                      Raspberry Pi bootloader

svanstrom, 2020-03-13:
> So far I've failed to reproduce the flicker, and haven't noticed it previously across multiple HyperPixel 4 / Pi 4 configurations. It must be something unique about the Raspbian/Firmware version on [...]

Sadly haven't got the time to reproduce it rn; but I can narrow down the setup a bit based on what I was doing at the time…

Most likely it was Rasbian with desktop (but without recommended software).

It was a brand new RPI4 (max config); and straight after installation I installed aptitude, and did "aptitude update; aptitude upgrade".

After that I installed HyperPixel stuff; and used chromium in kiosk mode to cover the display with a website (which is how I evaluated the colors/quality of the HyperPixel). I had rotated it; and the flickering was "pin-side".

It was the square one, with touch.

And… like I said before, this problem went away when I used Raspberry Pi Zeros instead.

fabiopolimeni, 2020-04-25:
I have the same exact issue.
A brand new RPi4, with Bruster (no additional software)

The whole screen flickers, not only shows the white line on top, but it is as if if affects the whole screen to flicker.

Rotating the screen doesn't work:
Rotating display
xrandr: Failed to get size of gamma for output default
warning: output DSI-1 not found; ignoring
/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py:66: UserWarning: XRandR wrote to stderr, but did not report an error (Message was: 'xrandr: Failed to get size of gamma for output default\n')
  warnings.warn("XRandR wrote to stderr, but did not report an error (Message was: %r)"%err)
Saving display settings to /usr/share/dispsetup.sh
Setting matrix: 1 0 0 0 1 0
unable to find device pointer:generic ft5x06 (11)
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
----------------------------------------------------------------------
**cat /proc/cpuinfo | grep Revision**
Revision	: c03111

**uname -a**
Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

**vcgencmd version**
Feb 12 2020 12:36:21 
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)

**vcgencmd dispmanx_list**
display:0 format:XRGB8888 transform:0 layer:-127 720x720 src:0,0,720,720 dst:0,0,720,720 cost:489 lbm:0 

**cat /boot/config.txt**
# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
dtparam=spi=off
# Enable audio (loads snd_bcm2835)
dtparam=audio=on
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
gpu_mem=128
#hdmi_enable_4kp60=1

dtoverlay=hyperpixel4
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f226
dpi_timings=720 0 15 15 15 720 0 10 10 10 0 0 0 60 0 35113500 6

Is there any new on this? I need to rotate the screen, and I would like not to have the screen flickering :(

fabiopolimeni, 2020-04-25:
> I have the same exact issue.
> A brand new RPi4, with Bruster (no additional software)
> 
> The whole screen flickers, not only shows the white line on top, but it is as if if affects the whole screen to flicker.
> 
> Rotating the screen doesn't work:
> Rotating display
> xrandr: Failed to get size of gamma for output default
> warning: output DSI-1 not found; ignoring
> /usr/lib/python2.7/dist-packages/screenlayout/xrandr.py:66: UserWarning: XRandR wrote to stderr, but did not report an error (Message was: 'xrandr: Failed to get size of gamma for output default\n')
> warnings.warn("XRandR wrote to stderr, but did not report an error (Message was: %r)"%err)
> Saving display settings to /usr/share/dispsetup.sh
> Setting matrix: 1 0 0 0 1 0
> unable to find device pointer:generic ft5x06 (11)
> Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
> 
> **cat /proc/cpuinfo | grep Revision**
> Revision : c03111
> 
> **uname -a**
> Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
> 
> **vcgencmd version**
> Feb 12 2020 12:36:21
> Copyright (c) 2012 Broadcom
> version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)
> 
> **vcgencmd dispmanx_list**
> display:0 format:XRGB8888 transform:0 layer:-127 720x720 src:0,0,720,720 dst:0,0,720,720 cost:489 lbm:0
> 
> **cat /boot/config.txt**
> disable_overscan=1
> dtparam=i2c_arm=off
> dtparam=spi=off
> dtparam=audio=on
> max_framebuffers=2
> gpu_mem=128
> dtoverlay=hyperpixel4
> enable_dpi_lcd=1
> dpi_group=2
> dpi_mode=87
> dpi_output_format=0x7f226
> dpi_timings=720 0 15 15 15 720 0 10 10 10 0 0 0 60 0 35113500 6
> 
> Is there any new on this? I need to rotate the screen, and I would like not to have the screen flickering :(

Just one note, I have been able to invert the screen by adding `display_lcd_rotate=2` to the config.txt. Now, the flickering white stripe is at the bottom of the screen (so its physical position didn't change, if this can be of any help, to whoever is investigating this).

ffbosh, 2020-04-25:
I can rotate the screen from the /boot/config.txt file by adding the line display_rotate.
ex:
rotate 90 degrees:
	display_rotate = 1
rotate 180 degrees:
	display_rotate = 2
rotate 270 degrees:
	display_rotate = 3
horizontal flip:
	display_rotate = 0x10000
vertical flip:
	display_rotate = 0x20000
(save and reboot)

With the blink I can't help you, 
I have a similar problem, my problem is that the screen moves a pixel from the connector side to the opposite side every so often and, depending on the position of the screen, it shows a white line, but not I was able to fix it.

fabiopolimeni, 2020-04-25:
> I can rotate the screen from the /boot/config.txt file by adding the line display_rotate.
> ex:
> rotate 90 degrees:
> display_rotate = 1
> rotate 180 degrees:
> display_rotate = 2
> rotate 270 degrees: [...]

Yeah, the flickering ,manifests exactly like that, a white line appears at the very top of the screen (from the side of the connector to be clear) and it makes the whole screen framebuffer to move rapidly.

Btw, I have fixed the rotation problem by setting `display_lcd_rotation`, but I'm wondering whether the failure of `hyperpixel-rotate` command is actually related to the flickering problem.

ffbosh, 2020-04-27:
Well, after trying
Raspbian Buster Lite, desktop, and desktop with recommended software (with no positive results), I was able to "fix" the problem by changing <h_sync_polarity> from dpi_timings to 1.
This solved the "flickering" that the screen had, but added a line on the upper side (connector side) that I have solved with overscan_bottom = 1
I don't think this is the right solution but it works for me.
**dpi_timings=720 1 15 15 15 720 0 10 10 10 0 0 0 60 0 35113500 6**
**overscan_bottom=1**
Sorry for my "google english"
PS: All this with Raspbian Buster Desktop.

fabiopolimeni, 2020-04-28:
@ffbosh well, your answer is better than a no answer. Thank you for the suggestion.
set the h_sync_polarity to 1 is the only thing that makes the screen stable, but unfortunately the bottom line of the buffer is bleeding to the top. I've tried values of 2, 3 and 4, but all of them bring back the flickering. So, because I don't know what I'm doing I didn't investigate further.

Also, `overscan_bottom=1` doesn't fix the bleeding for me, the overscan is ignored.

@Gadgetoid Do the latest findings on this issue ring any bell?

Thanks.

johnpeart, 2020-05-14:
Have had the same issue with flickering on my screen that I've used on a fresh Raspbian install today. Can confirm that changing `config.txt` from...

```
dpi_timings=720 0 15 15 15 720 0 10 10 10 0 0 0 60 0 35113500 6
```

To this...

```
dpi_timings=720 1 15 15 15 720 0 10 10 10 0 0 0 60 0 35113500 6
```

Fixed the flickering issue, and adding:

```
overscan_bottom=1
```

Removed the bleeding for me. (Just as @ffbosh suggests)

ashenshugarRET, 2020-10-30:
I had the exact same issue with a flickering line which is fixed by changing
`dpi_timings=720 0 15 15 15 720 0 10 10 10 0 0 0 60 0 35113500 6`
to
`dpi_timings=720 1 15 15 15 720 0 10 10 10 0 0 0 60 0 35113500 6`

however, this causes a solid white line to appear on the screen, this is fixed by changing
`overscan_top=0`
to
`overscan_top=1`

I can confirm that the fix holds regardless of the screen rotation setting

closed 2026-08-25 by Gadgetoid