pimoroni/hyperpixel4 #48: Black Screen on Rpi4

PinkSnake, 2019-10-16:
Issue: Black screen after fresh install

How to reproduce:
`pi@rpihack:~ $ sudo apt full-upgrade
pi@rpihack:~ $ curl -sSL https://get.pimoroni.com/hyperpixel4 | bash`

Raspberry informations: 
`pi@rpihack:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@rpihack:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage4
`
dmesg:
`
[    9.790258] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
[    9.882495] brcmfmac: F1 signature read @0x18000000=0x15264345
[    9.890699] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.891337] usbcore: registered new interface driver brcmfmac
[   10.128940] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   10.158239] Goodix-TS 7-0014: i2c test failed attempt 1: -6
[   10.181323] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: May  2 2019 02:46:17 version 7.45.189 (r714228 CY) FWID 01-e1db26e2
[   10.193818] Goodix-TS 7-0014: i2c test failed attempt 2: -6
[   10.231959] Goodix-TS 7-0014: I2C communication failure: -6
`

Gadgetoid, 2019-10-17:
The i2c communication failure doesn’t sound promising. How is your display connected to the Pi?

If you’ve got SSH access can you also try running “hyperpixel4-init”?

Ninic0c0, 2019-11-06:
Sorry for delay but I have tested the screen on several raspberry pi 4 for pre-studies and i'm not able to get the same result on all of them...

Have you tried on your side to install on rpi4 ?

Cheers

Gadgetoid, 2019-11-06:
I've since installed on Pi 4 and Pi 3B+ (during my i2c address weirdness testing for #41) and in instances where I get a black screen, it's usually from having some interface enabled in `/boot/config.txt` that interferes with the DPI pins. i2s, i2c, spi and UART all need to be disabled.

Do you have some more details on the steps you tried, the Pi's you tried on and what didn't work?

Ninic0c0, 2019-11-06:
Hi and Thx for replying,

It's really strange, same setup. Fresh install with raspbian Buster (full).

Board #1 -> black screen, no message in dmesg or /var/log/messages. 
Board #2 -> Display ok but touchscreens witched 
Board #3 -> I have do exactly the same (i made a script) 

I don't understand but :
`curl -sSL https://get.pimoroni.com/hyperpixel4 | bash`

seems different of:

'git clone .... -b rpi4'

Could you please provide some advises ? Because i'm stuck from 2 days :(
Cheers

Ninic0c0, 2019-11-06:
Same issue with Buster Light.

1. fresh install
2. curl method to install driver

--> Display is ok but x,x twisted :S

Gadgetoid, 2019-11-08:
Closing this in favour of #52 where I've posted the step-by-step that I just successfully tested using the latest `pi4-i2c-fix` branch.

closed 2019-11-08 by Gadgetoid