pimoroni/hyperpixel4 #49: Pi4 forgets I2C settings

Matze121, 2019-10-23:
Hi everyone,

I‘ve got a Pi4 with Hyperpixel 4.0.
Since the 2019-09-26 release of Raspbian the Display is actually working.
But wherever I touch the Display, the cursor moves only a little bit in the lower right corner of the desktop.

After using the command „sudo ln -s /dev/i2c-3 /dev/i2c-1“ and a reboot the touch works perfectly.
But as soon as I shutdown the Pi and start it again this setting seems to be lost and the touch behaves like before...

Thanks for any help ;)

sandyjmacdonald, 2019-10-23:
I reckon if you add that `sudo ln -s /dev/i2c-3 /dev/i2c-1` to your `/home/pi/.bashrc` file, then it'll persist every time you reboot. Give it a try, let us know how you get on. 🙂

Matze121, 2019-10-25:
I just tried it but unfortunately it doesn't make any difference.
If I reboot it works perfectly, the problem is the first start after a shutdown.

Probably this command has to be executed earlier but I don't know how to do this...

Gadgetoid, 2019-10-29:
What driver are you using for touch? You shouldn't need to touch the i2c nodes at all since the driver is bound to the correct i2c interface by the device tree overlay.

Matze121, 2019-10-29:
I‘m using pi4 driver and the RV3028 RTC module attached to the I2C breakout.

I just unplugged the RTC and the touch works perfectly.
The problem seems to be the RTC...

Are there any settings required after `sudo pip install rv3028 `?

Gadgetoid, 2019-11-01:
Ah, so you're linking the i2c bus so you can use the `rv3028` library.

It may not help in your case, but you can also use `/dev/i2c-3` directly:

```
from smbus import SMBus
from rv3028 import RV3028

rv3028 = RV3028(i2c_dev=SMbus(3))
```

It sounds like the rv3028 is conflicting with the touch somehow, but it shouldn't be doing anything much unless you're actively running code to talk to it. There would be nothing you could change in software if the hardware wont play ball.

I'll see if I can replicate this.

Gadgetoid, 2019-11-01:
Okay. That's really, really weird. I'm seeing the same problem. 

* If I hot-plug the rv3028 it's fine and I can see it on the i2c bus.
* If I reboot with it plugged in, my touch is restricted to the top left corner
* If I contrive a way to connect the rv3028 without the `interrupt` pin then it works fine

So one solution would be to connect the rv3028 using some jumper jerky and leaving out the interrupt pin. It might also be possible to configure the rv3028 with interrupt behaviour that doesn't mess up the touch.

I don't know *why* it messes up the touch, though...

Matze121, 2019-11-01:
Leaving out the interrupt pin works!

I also added `dtoverlay=i2c-rtc,rv3028` to config.txt
The output of `sudo i2cdetect -y 7`:
[image]

If I connect the RTC without the LCD the output of `sudo i2cdetect -y 1ˋ shows UU instead of 52...

Why does it switch to i2c-7 and from UU to 52?
Thanks a lot!

Gadgetoid, 2019-11-04:
`UU` is how an i2c address currently in-use by the kernel is denoted.

When you use `i2c-rtc,rv3028` you're telling the kernel to load a kernel driver for the rv3028. However, this kernel driver only knows about `/dev/i2c-1` or the primary user-accessible i2c bus of the Pi.

Since HyperPixel4 must used these i2c pins for something else (the vital vsync and hsync signals) it has to set up a software i2c bus on some free pins elsewhere. `/dev/i2c-7` is where (for you at least) this software i2c bus shows up.

So these are physically different i2s busses connected to different pins.

You can probably either modify the `i2c-rtc` dtoverlay, or combine it into the `hyperpixel4` dtoverlay to add support for the rv3028 kernel driver on the software i2c bus. This can be tricky.

You'd need to borrow from `i2c-rtc-gpio-overlay` (https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/i2c-rtc-gpio-overlay.dts), and merge this chunk into the HyperPixel4 dtoverlay:

```
		target = <&i2c_gpio>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			status = "okay";

			rv3028: rv3028@52 {
				compatible = "microcrystal,rv3028";
				reg = <0x52>;
				status = "okay";
			};
		};
```

This should t hen, in theory, give you a `/dev/i2c-7` with a `UU` on both `0x52` and `0x14`.

You can see where HyperPixel4's GPIO i2c bus is set up here: https://github.com/pimoroni/hyperpixel4/blob/c958d35ee2c91af49c9c8e41feb4ef8508f8a81b/src/hyperpixel4.dts#L35-L44

Gadgetoid, 2019-11-04:
Here are the changes I made to the dtoverlay: `src/hyperpixel4.dts` before building and installing with `make && sudo make install`:

```patch
index 3757a99..bbdc392 100644
--- a/src/hyperpixel4.dts
+++ b/src/hyperpixel4.dts
@@ -80,6 +80,20 @@
             touchscreen-inverted-y;
         };
     };
+    fragment@8 {
+        target = <&i2c_gpio>;
+        __overlay__ {
+            #address-cells = <1>;
+            #size-cells = <0>;
+            status = "okay";
+
+            rv3028: rv3028@52 {
+                compatible = "microcrystal,rv3028";
+                reg = <0x52>;
+                status = "okay";
+            };
+        };
+    };
     __symbols__ {
         dpi18_pins = "/fragment@1/__overlay__/dpi18_pins";
     };
```

And this gives me:

```
pi@raspberrypi:~/Documents/hyperpixel4 $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
And voila:

```
pi@raspberrypi:~/Documents/hyperpixel4 $ sudo hwclock
2019-11-04 15:44:21.906488+00:00
```

Matze121, 2019-11-04:
Thanks a lot for your help!
I can only try this during the weekend, then I‘ll let you know how it works.

But it sounds very promising..

Matze121, 2019-11-16:
So I modified `src/hyperpixel4.dts` after your instructions and `i2cdetect -y 7` shows 2x `UU`!
Also `sudo hwclock` prints the time.

But if I reboot without network connection `sudo hwclock` returns `hwclock: ioctl(RTC_RD_TIME) to /dev/rtc0 to read the time failed: Invalid argument.`.

If I install the library from [https://github.com/pimoroni/rv3028-python](url) and execute `rv3028-python/examples/set-time.py` I get the error:

`Traceback (most recent call last):
  File "set-time.py", line 15, in <module>
    rtc = rv3028.RV3028()
  File "/usr/local/lib/python3.7/dist-packages/rv3028-0.0.5-py3.7.egg/rv3028/__init__.py", line 248, in __init__
  File "/usr/local/lib/python3.7/dist-packages/i2cdevice-0.0.6-py3.7.egg/i2cdevice/__init__.py", line 166, in __init__
    self._i2c = smbus.SMBus(1)
FileNotFoundError: [Errno 2] No such file or directory`

Does it work in your setup?

Gadgetoid, 2019-11-24:
To use the `rv3028` library you'll have to change the example to connect to i2c bus 7 instead of 1 by importing `smbus` at the top:

```
import smbus
```

And supplying a `bus` argument when the `RV3028` class is instantiated:

```
rtc = rv3028.RV3028(bus=smbus.SMBus(7))
```

Matze121, 2019-11-24:
Now another error:

`Traceback (most recent call last):
  File "set-time.py", line 15, in <module>
    rtc = rv3028.RV3028()
  File "/usr/local/lib/python3.7/dist-packages/rv3028-0.0.5-py3.7.egg/rv3028/__init__.py", line 254, in __init__
  File "/usr/local/lib/python3.7/dist-packages/i2cdevice-0.0.6-py3.7.egg/i2cdevice/__init__.py", line 230, in get
    self.read_register(register)
  File "/usr/local/lib/python3.7/dist-packages/i2cdevice-0.0.6-py3.7.egg/i2cdevice/__init__.py", line 183, in read_register
    self.values[register.name] = self._i2c_read(register.address, register.bit_width)
  File "/usr/local/lib/python3.7/dist-packages/i2cdevice-0.0.6-py3.7.egg/i2cdevice/__init__.py", line 283, in _i2c_read
    for x in self._i2c.read_i2c_block_data(self._i2c_address, register, bit_width // self._bit_width):
OSError: [Errno 16] Device or resource busy
`

It's just frustrating...

Matze121, 2019-12-07:
I followed your instructions but there seems to be only one way that the RTC works:

- get hyperpixel driver,  edit src/hyperpixel4.dts, install the driver
- get rv3028 driver, edit set-time.py and  `/usr/local/lib/python3.7/dist-packages/i2cdevice-0.0.6-py3.7.egg/i2cdevice/__init__.py` in line 166 to `self._i2c = smbus.SMBus(7)`

- hotplug rv3028, run set-time.py, reboot
- edit `/lib/udev/hwclock-set` to 
`#if [ -e /run/systemd/system ] ; then`
`# exit 0`
`#fi`,
execute `sudo hwclock -w`

closed 2019-12-07 by Matze121