pimoroni/hyperpixel4 #50: DSI and Hyperpixel4 Pi3B+

Azkali, 2019-10-30:
# DSI and Hyperpixel4

I have a Hyperpixel 4" Touchscreen running fine on a Pi3B+.
I'm trying to setup a dual monitor with a 5" DSI touchscreen (from DFRobot)

But I can have either one or the other at the same time ( when enabling/disabling enable_dpi_lcd in /boot/config.txt ).

Is it even possible to achieve this configuration ? 
The finality is to use this config on a pi4 ( I haven't received mine yet so I'm testing it on a Pi3B+ for now )

Gadgetoid, 2019-10-31:
I believe DPI and DSI are mutually exclusive on all models of Pi. This certainly wont work on anything pre Pi4 and information on the Pi4 is still pretty scant- the closest I get is that, from a hardware perspective, the DSI0 and DPI connections "share a pixel valve"- source: https://www.raspberrypi.org/forums/viewtopic.php?t=243812 and that of the three "pixel valves" one is assigned to HDMI, one to DSI0/DPI and one as a writeback to memory to perform transpose/rotate operations.

On Pi 4 your best bet is to use an HDMI display alongside with DPI or DSI since these should work together.

Azkali, 2019-11-01:
Mmhh just checked out there is an open issue on the RPI : https://github.com/raspberrypi/firmware/issues/1094 so technically it should be possible

closed 2026-08-25 by Gadgetoid