pimoroni/hyperpixel4 #51: [hyperpixel4-rotate] - ImportError: No module named screenlayout

thomasdlap, 2019-11-03:
Hey Devs, 

Firstly thank you for awesome work!

Need help debugging an issue.

I have followed steps outlined in https://github.com/pimoroni/hyperpixel4/tree/pi4

and the display is stuck in portrait, trying to get to landscape 

running the following command over ssh: DISPLAY=:0.0 hyperpixel4-rotate left

```
Rotating display
xrandr: Failed to get size of gamma for output default
warning: output DSI-1 not found; ignoring
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ImportError: No module named screenlayout
Setting matrix: 0 -1 1 1 0 0
unable to find device pointer:Goodix Capacitive TouchScreen
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
```

thomasdlap, 2019-11-03:
Interestingly enough if I select as raspberry pi 3, curl -sSL https://get.pimoroni.com/hyperpixel4 | bash

Im able to run the rotate command and on boot I have landscape but freezes/hangs on

[    OK    ] Found Device /dev/serial1

thomasdlap, 2019-11-03:
seemsidentical to issue 

https://forums.pimoroni.com/t/hyperpixel4-display-rotate-freezes-screen-at-startup/11965

Gadgetoid, 2019-11-04:
What OS/Pi are you running?

You can find OS details with: `cat /etc/os-release` or `lsb_release -a`

thomasdlap, 2019-11-04:
@Gadgetoid 

Thanks for quick response and command as a bit of a n00b :)

`
pi@octopi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

`

Gadgetoid, 2019-11-04:
Were you running Buster lite or full?

thomasdlap, 2019-11-04:
@Gadgetoid 

im using buster lite from:
https://storage.googleapis.com/octoprint/2019-06-24_2019-06-20-octopi-buster-lite-0.17.0.zip

Guide I followed for doing install was 
https://blog.tfm.ro/running-octoprint-on-raspberry-pi-4-model-b-part-1/

Gadgetoid, 2019-11-04:
Ah- yes the `hyperpixel4-rotate` command requires an X desktop session, and additionally requires the "Screen Configuration" utility which ships with Buster Desktop. It leverages this to persist the screen rotation/layout settings in what's the de-facto way on Pi 4 (for better or wosrse).

If you're not running X then it is- as far as I'm aware- currently impossible to rotate just the Terminal. This is a Pi 4 issue and not something we can readily fix. If you're running X then you'll need to `sudo apt install arandr`

Ninic0c0, 2019-11-06:
Hi, 

Same issue with a fresh install on Rpi4:

Rotating display
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in __init__
    version_output = self._output("--version")
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _output
    raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: No protocol specified
Can't open display :0.0

Setting matrix: 0 -1 1 1 0 0
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules

Is it possible to have a fix ? Because same installation on 2 rpi4 and one doesn't work at all 🈂️ 
Thx

meodai, 2020-01-24:
I also have this issue with an fresh PI4 instal. Works fine for the pi3 though.
Steps to reproduce. Install latest stable raspbian -> `sudo apt update && sudo apt full-upgrade` -> reboot -> `curl -sSL https://get.pimoroni.com/hyperpixel4 | bash` -> reboot -> `hyperpixel4-rotate right` (does not matter if I reboot in between or not).

Gadgetoid, 2020-01-24:
If you're using Raspbian Lite then this utility will never work- it's designed to hook into the GUI screen configuration and rotate the touch to match.

For command-line rotation it should suffice just to use `display_lcd_rotate` in `/boot/config.txt`, see: https://github.com/pimoroni/hyperpixel4/issues/39

meodai, 2020-01-24:
@Gadgetoid its not raspbian light. Its the full fledged one with UI and Office apps and everything.

sepia-assistant, 2020-02-09:
I've been trying to fix my Hyperpixel4 touchscreen for a few hours now and thought I should share some findings here and sum up some stuff since none of the "old" fixes worked:
* Problem was that the touchscreen x and y axis are swapped after fresh installation on **RPi4** buster
* Turns out that using **Raspbian Lite** creates a few issues related to missing X Desktop session when using RPi4 as mentioned in this thread, [here](https://github.com/pimoroni/hyperpixel4/issues/39) and in a few other places (this should be indicated somewhere on the main page! Did I miss it?). Thats why hyperpixel4-rotate will create errors.
* `dtoverlay=hyperpixel4:rotate` does nothing in my case. Probably a X Desktop thing.
* Playing with `src/hyperpixel4.dts` and `touchscreen-size-x/y` did not improve anything
* If you want to rotate the screen using Raspbian Lite you need to remove `dtoverlay=vc4-fkms-v3d` from your `/boot/config.txt` . After that you can add `display_lcd_rotate=...` and the boot sequence will not crash.
* To fix the swapped and inverted touch controls you need to find the correct combination of `dtparam=touchscreen-swapped-x-y`, `dtparam=touchscreen-inverted-x` and `dtparam=touchscreen-inverted-y`
* I've only managed to fix the touch controls for `display_lcd_rotate=2` but not 0 or 4 (seem it does not accept the swap parameter in this case)

Hope this helps!

Gadgetoid, 2020-03-17:
Indeed the in-tree Goodix driver that we're currently using *does not* support `touchscreen-swapped-x-y` which makes it impossible to support low-level 90 and 270 degree rotations.

There's an alternate driver that's a pain to install and work with, but does enable these touch rotations, see here for more details: https://github.com/pimoroni/HyperPixel4TouchScreen/tree/master/driver

The mentioned `.deb` package is in releases here: https://github.com/pimoroni/HyperPixel4TouchScreen/releases

But since it's a kernel source package it requires the kernel headers to compile, which can also be a pain to install.

There's some discussion of this driver in another thread, here: https://github.com/pimoroni/hyperpixel4/issues/58#issuecomment-597896969

Since this alternate driver requires dkms and kernel headers I've not put together a one-line installer for it. Also from personal experience it's a pain to maintain out-of-tree modules. I guess the optimum solution here is to figure out how to get these fixes upstream.

metaColin, 2021-03-09:
*bump*

Does the Pimoroni crew have any plans to address this issue with a fix?

I *want* I would *prefer* to use Hyper Pixel displays in more of my Pi Projects...

I had such plans for this line of screens, but in practice, I've with a few projects now resorting to other displays that can rotate even in headless circumstances.

iainp999, 2021-04-28:
I've just received a hyperpixel 4.0 and having the same issue.

No matter what I try, just can't get the screen to rotate.

```
Rotating display
Can't open display
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ImportError: No module named screenlayout
Setting matrix: 1 0 0 0 1 0
Unable to connect to X server
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
```

iainp999, 2021-04-28:
Maybe `sudo apt-get install arandr` will help?

iainp999, 2021-04-28:
Then reboot

Gadgetoid, 2021-05-20:
> Does the Pimoroni crew have any plans to address this issue with a fix?

I don't have the bandwidth to look into trying to upstream kernel module fixes at the moment, it's a heck of an undertaking.

Regarding *display* rotation- if you're not using Raspberry Pi OS + Desktop then you're out in the dark- rotation while using `dtoverlay=vc4-fkms-v3d` but *not* Raspberry Pi OS + Desktop has been confirmed to be impossible and, it seems, unlikely to ever be implemented by Raspberry Pi. This is not something we can fix :(

This is the pain of small displays like this being intended for portrait devices like mobile phones and it frustrates me to no end that we don't have a good answer for this.

I'm nudging our engineers to find a *landscape* display in future, since portrait is more trouble than it's worth.

closed 2021-05-20 by Gadgetoid