pimoroni/hyperpixel4 #52: [How To] - Raspberry PI 4 -

Ninic0c0, 2019-11-07:
Hello :)

Ok so after wasting 2 days on it, i got a good workflow!

**!!!!Do not plug the screen yet!!!!**

### 1.Install Raspbian Buster full
Nothing more to do here.
**Doesn't enable i2c**
### 2. Update Raspbian
`sudo apt update && sudo apt upgrade -y`
### 3. hyperpixel4  script 
```
git clone https://github.com/pimoroni/hyperpixel4 -b pi4
cd hyperpixel4
sudo ./install
```
### 4. Power off
`sudo halt`

**!!!!Plug the screen!!!!**

### Edit /boot/config.txt for Landcape Mode:

`[all]
#dtoverlay=vc4-fkms-v3d
start_x=1
gpu_mem=128

dtoverlay=hyperpixel4
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

dtparam=touchscreen-swapped-x-y
dtparam=touchscreen-inverted-x
dtparam=touchscreen-inverted-y
`

Enjoy 👍

Gadgetoid, 2019-11-08:
I've just set up a fresh image and tested again.

For me it works with:

* Install Raspbian Buster Full (in my case I used win32diskimager)
* Go through the onboarding process to set up WiFi and install updates
* Open a terminal window while you're waiting for updates to install
* `git clone git://github.com/pimoroni/hyperpixel4.git -b pi4-i2c-fix` * **SEE NOTE 1** 
* `cd hyperpixel4`
* `sudo ./install.sh`
* Wait for updates to finish
* Reboot
* Open a terminal window and type `hyperpixel4-rotate left` or whatever orientation you're aiming for

## Notes
1. the `pi4-i2c-fix` branch is the same as `pi4` except it attempts to fix issues with touch by using two dtoverlays- one for each possible i2c address. This will be merged into `pi4` soon!
2. it shouldn't matter if you plug in the HyperPixel4 before or after the install process- I usually start with mine plugged in and HDMI connected just in case
3. Despite .2 above, having HDMI connected will completely mess up your touch input since touch is scaled to the entire screen area
4. This *requires* Raspbian Buster Full, since Lite does not have a desktop environment

On Raspbian Buster Full you should *never* need additional `/boot/config.txt` entries, it should suffice just to use `hyperpixel4-rotate` to get the rotation you want.

Ninic0c0, 2019-11-26:
After a full update same issue again and again ...

Gadgetoid, 2019-11-26:
What do you mean by "cannot run any display application"? Is that the issue you mention?

Ninic0c0, 2019-11-26:
Hello,

I mean it's a terrible mess, I have to benchmark this screen, i have several raspberry pi 4.
After same install process (with curl + bash method) my results:

Board 1 -> black screen
Board 2 -> touchscreen x,y seems really wired :(
Board 3 -> No trouble

Have you tested on your side with a clean raspbian ?

Thx for support :)
Cheers,

Ninic0c0, 2019-11-26:
Sorry but it's enough for me, I have spent too much time on this screen and I have found a similar screen easy to use and plug-and-play. I close the bug.

Cheers,

closed 2019-11-26 by Ninic0c0