pimoroni/hyperpixel4 #53: Rotation of display and touch is reset to normal after reboot

nielsclausen, 2019-12-01:
The rotation of the display set over SSH by:
`DISPLAY=:0.0 hyperpixel4-rotate right`
works like a charm. BUT after reboot, rotation is set back to normal again. Is this normal behavior? If so, how can i rotate the screen/touch permanently?

**System**
Rectangular Hyperpixel4-touch, and I've installed the driver via the install script.

Using Raspian Buster (sep 26th) on a RBPI4 rev:b03111. 
`uname -r` produces: 4.19.75-v7l+

Also installed kivy, and dependencies, but not much else.

**Extra debugging information**
pi@rc00:~ $ dmesg | grep Goodix
[    4.771178] Goodix-TS 7-0014: ID 911, version: 1060
[    4.794259] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-7/7-0014/input/input0

Gadgetoid, 2019-12-18:
The rotate script hooks into the Screen Configuration utility which uses a somewhat arcane method of persisting rotation settings across boots - it will write a file out to `/usr/share/dispsetup.sh` which is normally executed on boot, but I'm not actually sure *what* executes it. (This is exactly what Screen Configuration does if a user sets up their config with the GUI)

You should check if that file exists, if it's executable and if it actually runs. If it's okay and your particular setup just isn't executing it, then you can add it to crontab or as a systemd one-shot service according to your preference.

capgames, 2020-02-20:
a possible solution is to get it to automatically execute on start up

you can do this by doing
sudo nano /etc/rc.local

at the very bottom add the following line
/usr/bin/hyperpixel4-rotate right

replace 'right' for the way you want the screen to be
left - landscape, power/HDMI on bottom
right - landscape, power/HDMI on top
normal - portrait, USB ports on top
inverted - portrait, USB ports on bottom

reboot.
If the screen does not rotate just open the terminal and run the command
hyperpixel4-rotate right

again replace 'right' with the rotation you want and then reboot again and it should be fine and the touchscreen is not broken too

sscherbinski, 2020-09-02:
I get this as well, though the display rotates after I run it.    From the command line I get the following.  Hopefully it helps.

pi@raspberrypi:~ $ DISPLAY=:0.0 hyperpixel4-rotate right
Rotating display
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in __init__
    version_output = self._output("--version")
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _output
    raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: No protocol specified
Can't open display :0.0

Setting matrix: 0 1 0 -1 0 1
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules

travellingkiwi, 2021-01-12:
Nothing appears consistent... I'm trying to achieve this on an octopi install - RasPi-4 4GB with a HyperPixel4 Rectangular. I can see the display in portrait but when I try to use hyperpixel4-rotate I get

(With DISPLAY set and exported as 

export DISPLAY=:0.0

And the X apps work (i.e. I can start an xterm etc)

However trying to rotate I get

pi@octopi:~ $ /usr/bin/hyperpixel4-rotate left
Rotating display
xrandr: Failed to get size of gamma for output default
warning: output DSI-1 not found; ignoring
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ImportError: No module named screenlayout
Setting matrix: 0 -1 1 1 0 0
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
pi@octopi:~ $

And nothing happens... 

Manually changing config.txt as the Pi-# requires results in a rotated boot screen that doesn't scroll and doesn't show the X server.

jackrenter, 2021-01-14:
having a similar issue. have to do the "DISPLAY=:0.0 hyperpixel4-rotate left" with every reboot to get the correct landscape. is there any way to make this do this automatically on boot? ive tried like 10 different things and nothing seems to work. seems like there should be a simple way to execute one line of code. but i havent managed to find it, or do it properly.

pi@octopi:~ $ DISPLAY=:0.0 hyperpixel4-rotate left
Rotating display
[sudo] password for pi:
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ImportError: No module named screenlayout
Setting matrix: 0 -1 1 1 0 0
/usr/bin/hyperpixel4-rotate: line 10: xinput: command not found
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules

mspeculatrix, 2021-06-17:
> it will write a file out to `/usr/share/dispsetup.sh` which is normally executed on boot, but I'm not actually sure _what_ executes it.

I just added /usr/bin/hyperpixel4-rotate left to this file and it works. If I change the screen config it might get overwritten, I guess.

Gadgetoid, 2021-06-18:
Looks like I need to investigate the underlying xrandr commands to achieve rotation and set up some kind of launch script for usage where a standard Raspberry Pi OS setup isn't available. Would be nice to get the rotation early on in the boot process... I guess xorg.conf(.d). Which would probably be where input configuration should go, too. Right now we're at third party kludges built on top of third party kludges (because the latter would override the former otherwise) which is... "fine" for Pi OS but not so great for the wider Linux landscape.

closed 2026-08-25 by Gadgetoid