pimoroni/hyperpixel4 #58: Hyperpixel 4.0 on Volumio 2.7 (Debian Jessie)

Tom4c, 2020-03-02:
The display works perfectly on Volumio 2.7, using the Pimoroni driver.  Unfortunately, the touch interface does not.  I appreciate you only have time to support Raspbian, but any tips on where and how to start troubleshooting would be much appreciated.

If it's useful to anyone, I got the display working via the following process and SSH commands:

Install Volumio 2.713 and Touch Display 1.1.6 Plugin
```
sudo apt install apt-utils
sudo apt install device-tree-compiler
sudo apt-get install rpi.gpio
sudo git clone https://github.com/pimoroni/hyperpixel4 -b pi3
cd hyperpixel4
sudo ./install.sh
```
`sudo i2cdetect -7 3` shows the touchscreen is using i2c address 0x5d, and /dev/i2c-3 is present. I've just no idea how to point to it and use it.

Again, any advice on where to start the troubleshooting process would be much appreciated, thank you.

Gadgetoid, 2020-03-03:
It's likely Volumio is built down to a size and thus lacks the drivers required for the touch screen- that's more an educated guess than reality, though.

The good news is, adding the drivers should get you up and running.

The bad news? You might have to compile them from source.

A first point of call might be to check your kernel modules directory to see if there's a `goodix.ko`:

```
ls /lib/modules/`uname -r`/kernel/drivers/input/touchscreen
```

Or, just type `modprobe goodix` and you should get an `Operation not permitted` error if you're a normal user and the module is installed.

Tom4c, 2020-03-03:
Huge thanks for replying, that really cheered me up :-)

`goodix.ko` is present in the kernel modules directory.  However, `modprobe goodix` returns the result: `-bash: modprobe: command not found` .  Running the command `sudo modprobe goodix` runs without any success or failure messages and returns to the command line.  I've checked and modprobe is installed (kmod version 18).

BTW, the touchscreen is working fine on Raspbian, so it's not a hardware issue.

Many thanks again for your help.

Tom4c, 2020-03-06:
There's a very good summary of the issue here: [https://forums.pimoroni.com/t/hyperpixel-4-0-digitiser-not-working/12461](url)

Tom4c, 2020-03-09:
I've tried using 'make' to compile a new hyperpixel4.dtbo, and I get the following errors:
`volumio@volumio-study:~/hyperpixel4/src$ sudo make
dtc -I dts -O dtb -o hyperpixel4.dtbo hyperpixel4.dts
Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /fragment@2 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /fragment@3 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /fragment@3/__overlay__/i2c@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /fragment@4 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /fragment@5 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /fragment@6 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /fragment@7 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /__local_fixups__/fragment@0 has a unit name, but no reg property
volumio@volumio-study:~/hyperpixel4/src$ `
I copied the new .dtbo to `/boot/overlays/` and the new .dtbo works for the display (as before) but not for the touchscreen (as before).  Is this relevant?  Thanks for looking.

Gadgetoid, 2020-03-10:
Those warnings can be ignored.

What changes did you make to the dtbo?

If you're still running with a single dtbo file, it's possible you could be encountering the i2c address issue with the touch IC where it can be either 0x14 or 0x5d. Might be worth trying the installer from this branch: https://github.com/pimoroni/hyperpixel4/tree/pi4-i2c-fix

Tom4c, 2020-03-10:
Thanks for your help.  I didn't make any changes to the .dtbo, I just wondered if compiling it would establish the correct environment variables.  No luck unfortunately.
I have done a fresh install of Volumio, as per my previous process, and installed from the pi4-i2c-fix branch (despite it being a Pi3B+).  As before, the display works beautifully, and the touchscreen doesn't!  Commenting out dtparam=i2c_arm=on from /boot/config.txt seems to help with the i2c dmesg errors, but the situation remains the same.  If you can see any clues in the following I would be delighted!
Thanks again for your time.

```volumio@volumio-study:~$ dmesg | grep i2c
[   19.320443] i2c /dev entries driver
[   19.804762] gpio-10 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[   19.804838] gpio-11 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[   19.831747] i2c-gpio i2c@0: using lines 10 (SDA) and 11 (SCL)
[   19.915270] pinctrl-bcm2835 3f200000.gpio: pin gpio2 already requested by soc:fb; cannot claim for 3f804000.i2c
[   19.915289] pinctrl-bcm2835 3f200000.gpio: pin-2 (3f804000.i2c) status -22
[   19.915311] i2c-bcm2835 3f804000.i2c: Error applying setting, reverse things back
[   19.915338] i2c-bcm2835: probe of 3f804000.i2c failed with error -22
volumio@volumio-study:~$ sudo nano /boot/config.txt _[commented out dtparam=i2c_arm=on]_
volumio@volumio-study:~$ sudo reboot now           
...
volumio@volumio-study:~$ dmesg | grep i2c
[   11.966502] i2c /dev entries driver
[   12.340331] gpio-10 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[   12.340405] gpio-11 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[   12.340826] i2c-gpio i2c@0: using lines 10 (SDA) and 11 (SCL)``

volumio@volumio-study:~$ sudo lsmod | grep goodix
volumio@volumio-study:~$ sudo lsmod | grep evdev
volumio@volumio-study:~$ sudo lsmod | grep i2c*

i2c_gpio               16384  0 
i2c_algo_bit           16384  1 i2c_gpio
i2c_dev                20480  0 

volumio@volumio-study:~$ ls /dev/i2c-*
/dev/i2c-3
volumio@volumio-study:~$ sudo i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f`
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- 5d -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
volumio@volumio-study:~$ sudo udevadm info /sys/bus/i2c/devices/3-005d
Unknown device, --name=, --path=, or absolute path in /dev/ or /sys expected.

Gadgetoid, 2020-03-11:
The i2c dmesg errors regarding GPIO2 are fine, since that's presumably a side-effect of the system trying to set up the EEPROM i2c bus, which it can't do because those pins are required for DPI output.

Do you see any Goodix/GT911 related output in dmesg?

Could you paste the contents of your `/boot/config.txt`?

Tom4c, 2020-03-11:
Hi, thanks again.  There is no Goodix or GT911 output in dmesg.  My /boot/config.txt is below (there is nothing in the userconfig.txt).  Commenting out i2c_arm makes no difference to behaviour, other than the reported dmesg 'errors'...

```initramfs volumio.initrd
gpu_mem=32
max_usb_current=1
dtparam=audio=on
audio_pwm_mode=2
#dtparam=i2c_arm=on
disable_splash=1
hdmi_force_hotplug=1

include userconfig.txt

dtoverlay=hyperpixel4
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

Gadgetoid, 2020-03-11:
Did you modify your `hyperpixel4.dtbo` to use i2c address `0x5d` instead of `0x14`? Doesn't look like you're running the `pi4-i2c-fix` branch, so it's probably not picking up the touch.

The touch IC is prone to bouncing between both i2c addresses, so the i2c-fix branch is your best bet, but if you want to try modifying the overlay these are the lines which need to change:

https://github.com/pimoroni/hyperpixel4/blob/c958d35ee2c91af49c9c8e41feb4ef8508f8a81b/src/hyperpixel4.dts#L53-L55  (`14` to `5d` on 53 and 55)

Tom4c, 2020-03-11:
OOOOH!  Getting so much closer :-)  My mistake on the pi4-i2c-fix branch. I've uninstalled and now got it right using the command `sudo git clone https://github.com/pimoroni/hyperpixel4 -b pi4-i2c-fix`.  I now have a touchscreen that responds, albeit 90degrees out of rotation.  Running `sudo hyperpixel4-rotate normal` gives errors (see below).  I'll go off and do some more homework to see where to go from here.  Many many many thanks for your help and patience, it's been much appreciated.

```
volumio@volumio-study:~$ sudo hyperpixel4-rotate normal
Rotating display
Can't open display 
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ImportError: No module named screenlayout
Setting matrix: 1 0 0 0 1 0
/usr/bin/hyperpixel4-rotate: line 10: xinput: command not found
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules

gvolt, 2020-03-11:
@Tom4c Maybe instead try adding [touchscreen-swapped-x-y](https://github.com/pimoroni/hyperpixel4/tree/pi4-i2c-fix#touch-rotation) in /boot/config.txt:
`dtoverlay=hyperpixel4,touchscreen-swapped-x-y`

Possibly this needs to be enhanced by one of the options to invert x or y axis (`touchscreen-inverted-x` or `touchscreen-inverted-y`)

Tom4c, 2020-03-11:
PS, I have tried the command `sudo DISPLAY=:0.0 hyperpixel4-rotate normal` but I still get errors (see below).  Is this a Jessie versus Buster issue?
```
Rotating display
No protocol specified
Can't open display :0.0
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ImportError: No module named screenlayout
Setting matrix: 1 0 0 0 1 0
/usr/bin/hyperpixel4-rotate: line 10: xinput: command not found
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules

Gadgetoid, 2020-03-11:
In the absense of xinput you'll have to go with @gvolt's suggestion, which is to rotate things at the driver level.

Between `touchscreen-inverted-x`, `touchscreen-inverted-y`, and `touchscreen-swapped-x-y` you should be able to achieve any rotation.

gvolt, 2020-03-11:
IMHO correcting the touch orientation accordingly to the screen orientation at driver level would be the preferred way.

If the screen can be rotated by using `display_lcd_rotate` in /boot/config.txt the "Touch Display plugin" for Volumio would probably be able to do the rotation for screen as well as for touch without using `98-hyperpixel4-calibration.rules`.

Tom4c, 2020-03-11:
@gvolt @gadgetoid Thanks both, I really appreciate your help.  I can get the inversion commands to work, but not the `touchscreen-swappped-x-y` command.  Now I have the correct pi4-i2c-fix branch, there are three overlays in the `/boot/config.txt` .  I've appended `,touchscreen-swapped-x-y` to all of them, and then just the `hyperpixel4-common`, then just both `hyperpixel4-0x14` & `hyperpixel4-0x5d` dtoverlays.  The screen is portrait, with the USB sockets at the top.  The touchscreen is (I think) rotated 90 degrees clockwise, so touching top right registers as a touch top left.  When swiping down, the cursor moves to the right, when swiping right, the cursor moves up.  Interestingly, a full swipe top to bottom equals a full traverse right to left, and the same vertically. I.e. the long vertical swipe is being scaled down to the shorter horizontal width, and the cursor doesn't 'fall off' the screen.  Any ideas?!

gvolt, 2020-03-11:
You may try placing the parameters to dedicated lines like this:
```
dtoverlay=hyperpixel4
<other overlay options / parameters>
dtparam=touchscreen-swapped-x-y
dtparam=touchscreen-inverted-x
dtparam=touchscreen-inverted-y
```

Tom4c, 2020-03-11:
@gvolt Thankyou.  I gave it a try but sadly no luck.  It turns out that neither the swap nor invert commands work via either method.  The touchscreen is deaf to the changes and resolutely one orientation.  I have managed to use `display_lcd_rotate=1` to rotate the screen to match the touchscreen, which is a moderate victory :-).  However, I need the screen/touchscreen to be portrait.  I tried to rotate the whole lot using the Volumio "Touch Display plugin", but they do not rotate in unison and become unusable again.  Once again, many thanks for all your time and help these past few days.

Gadgetoid, 2020-03-11:
Curious `touchscreen-swapped-x-y` is a valid parameter, apparently, so I wonder if it's failing because the device-tree overlay is broken in some way, or because the driver isn't properly supporting it:

https://github.com/raspberrypi/linux/blob/2fab54c74bf956951e61c6d4fe473995e8d07010/Documentation/devicetree/bindings/input/touchscreen/goodix.txt#L24-L25

Oh... good grief... this may be related: https://forums.pimoroni.com/t/hyperpixel-4-touch-screen-dimensions-are-swapped/8198/32

Gadgetoid, 2020-03-11:
This is what happens when a problem comes up and is solved 2 years ago, but you're juggling so many things you forget it even happened and waste an hour trawling through Linux driver source, patches and other things trying to figure out if `swapped-x-y` is supported. Long story short: it's not.

However, persistent genius @AndrewCapon solved this by - if I remember correctly - customising and sideloading the manufacturer drivers. See: https://github.com/AndrewCapon/HyperPixel4TouchScreen/tree/master/driver

Oh boy, isn't this an adventure.

Maybe I should try to throw together - yet another - branch on this git repo to handle the install steps for non-standard installations.

Tom4c, 2020-03-12:
@Gadgetoid Thank you for looking, you're a star.  I've had a quick go at following the User instructions, but the result was `dpkg: error processing archive hyperpixel4-goodix-dkms_1.0_all.deb (--install): cannot access archive: No such file or directory`.  
Following the developer instructions had a few errors as well.  Is there much to do to bring the 2018 work up to date and compatible with the changes to the latest hyperpixel4 branch? Many thanks again.

Gadgetoid, 2020-03-12:
Ah- the instructions mention that the dpkg file is available in releases- you can grab it here: https://github.com/pimoroni/HyperPixel4TouchScreen/releases/download/v1.0/hyperpixel4-goodix-dkms_1.0_all.deb

Tom4c, 2020-03-12:
Thanks. So I've managed to fetch the file using `wget https://github.com/pimoroni/HyperPixel4TouchScreen/releases/download/v1.0/hyperpixel4-goodix-dkms_1.0_all.deb` and install using `sudo dpkg -i hyperpixel4-goodix-dkms_1.0_all.deb`.  That seemed to go work, although the final part of the output is `Module build for the currently running kernel was skipped since the
kernel source for this kernel does not seem to be installed.`.  Is that a problem?

Next up is setting the paramaters (I assume in /boot/config.txt).  Do I need to change the example from `dtoverlay=hyperpixel4:rotate_0` to `dtoverlay=hyperpixel4-common:rotate_0` to account for your extra overlays in the new pi4-i2c-fix branch?

There's a note: 

> both Portrait orientations must have framebuffer_width=480 and framebuffer_height=800 specified instead of the defaults in /boot/config.txt

 I don't believe that there are defaults in the confi.txt any more.  How should framebuffer_width etc. be implemented in the config.txt?  Just individual lines, or a `dtparam` etc?
Many thanks once again :-)

Gadgetoid, 2020-03-12:
`Module build for the currently running kernel was skipped since the kernel source for this kernel does not seem to be installed.` will probably be a problem.

You might want to `sudo apt install dkms` which will bring in the kernel headers. I'd hoped the .deb package would avoid this, but no such luck!

The `rotate_0` etc commands are only supported by Andy's version of `hyperpixel4.dts` - you might have some luck just using:

```
dtparam=touchscreen-swapped-x-y
dtparam=touchscreen-inverted-x
dtparam=touchscreen-inverted-y
```

Andy's dts uses the rotate commands to apply different touchscreen settings in a similar way:

https://github.com/AndrewCapon/HyperPixel4TouchScreen/blob/44439052d595e28686fe1f4da89fc82780a49d27/driver/hyperpixel4.dts#L117-L123

It might be possible to port his code and the relevant fragments into `hyperpixel4-common`.

Tom4c, 2020-03-12:
While I remember, the README.md at [https://github.com/pimoroni/hyperpixel4/tree/pi4-i2c-fix](url) suggests that the fix is only for Pi 4s despite it working with my Pi3B+ and Darioml's Pi2B.  the instructions are to use the `-b pi4` branch, it may be worth updating the readme to state the `-b pi4-i2c-fix`.
Currently, for the @AndrewCapon solution,  I'm having a problem installing `dkms`, but **I'll take that up on the Volumio forum**.  The error crops up right at the end of the install process (see below), supposedly I've run out of space!  the `cp` shell command makes the same claim.  It's a 64Gb SD card, but perhaps it's been partitioned frugally (understandable as it's supposed to be 'headless').  `apt-get clean` and `apt-get autoremove` haven't solved the problem.
Many thanks for the adventure so far :-)
```
Setting up linux-image-3.6-trunk-rpi (3.6.9-1~experimental.1+rpi7) ...
dpkg-divert: error: cannot copy '/usr/share/rpikernelhack/System.map-3.6-trunk-rpi' to '/boot/System.map-3.6-trunk-rpi.dpkg-divert.tmp': failed to write (No space left on device)
Running depmod.
depmod: FATAL: could not load /boot//System.map-3.6-trunk-rpi: No such file or directory
Error setting debconf flags in linux-image-3.6-trunk-rpi/postinst/depmod-error-initrd-3.6-trunk-rpi: Unsupported command "removing" (full line was "Removing 'diversion of /boot/vmlinuz-3.6-trunk-rpi to /usr/share/rpikernelhack/vmlinuz-3.6-trunk-rpi by rpikernelhack'") received from confmodule. at /var/lib/dpkg/info/linux-image-3.6-trunk-rpi.postinst line 495, <STDIN> line 3.
dpkg: error processing package linux-image-3.6-trunk-rpi (--configure):
 subprocess installed post-installation script returned error exit status 2
Setting up make (4.0-8.1) ...
Setting up patch (2.7.5-1+deb8u1) ...
Setting up binutils (2.25-5+deb8u1) ...
Setting up libgcc-4.9-dev:armhf (4.9.2-10+deb8u1) ...
Setting up gcc-4.9 (4.9.2-10+deb8u1) ...
Setting up gcc (4:4.9.2-2) ...
Setting up dkms (2.2.0.3-2) ...
Setting up libfakeroot:armhf (1.20.2-1) ...
Setting up fakeroot (1.20.2-1) ...
update-alternatives: using /usr/bin/fakeroot-sysv to provide /usr/bin/fakeroot (fakeroot) in auto mode
update-alternatives: warning: skip creation of /usr/share/man/es/man1/fakeroot.1.gz because associated file /usr/share/man/es/man1/fakeroot-sysv.1.gz (of link group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/es/man1/faked.1.gz because associated file /usr/share/man/es/man1/faked-sysv.1.gz (of link group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/fr/man1/fakeroot.1.gz because associated file /usr/share/man/fr/man1/fakeroot-sysv.1.gz (of link group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/fr/man1/faked.1.gz because associated file /usr/share/man/fr/man1/faked-sysv.1.gz (of link group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/sv/man1/fakeroot.1.gz because associated file /usr/share/man/sv/man1/fakeroot-sysv.1.gz (of link group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/sv/man1/faked.1.gz because associated file /usr/share/man/sv/man1/faked-sysv.1.gz (of link group fakeroot) doesn't exist
Processing triggers for libc-bin (2.19-18+deb8u10) ...
Errors were encountered while processing:
 linux-image-3.6-trunk-rpi
E: Sub-process /usr/bin/dpkg returned an error code (1)
volumio@volumio-study:~$

Gadgetoid, 2020-03-17:
I suspect Volumio is probably partitioned frugally- if cross-compiling weren't such a pain I'd suggest it. It would be nice to get the Goodix driver fixes upstream, but I understand that Andy's driver is basically *completely* different (it's the manufacturer supplied driver) from the driver that's available upstream.

AndrewCapon, 2020-03-17:
The boot partition doesn't have enough space, I bet there is some junk hanging around there.

What does:
```
df -h
```

and
```
ls -ltr /boot
```

show, paste it here.

Tom4c, 2020-03-17:
Thank you, I hadn't had time to look up the right commands.  It certainly looks tight in there!:
```volumio@volumio-study:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  2.2G  796M  1.3G  39% /imgpart
/dev/loop0      338M  338M     0 100% /static
overlay          55G  858M   51G   2% /
devtmpfs        471M     0  471M   0% /dev
tmpfs           485M   56M  430M  12% /dev/shm
tmpfs           485M  4.8M  481M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           485M     0  485M   0% /sys/fs/cgroup
tmpfs           485M   48K  485M   1% /tmp
tmpfs           485M     0  485M   0% /var/spool/cups
tmpfs            20M   52K   20M   1% /var/log
tmpfs           485M     0  485M   0% /var/spool/cups/tmp
/dev/mmcblk0p1   61M   61M   42K 100% /boot
tmpfs            97M     0   97M   0% /run/user/1000
volumio@volumio-study:~$ ls -ltr /boot
total 60982
-rw-rw-rw- 1 root root  3794600 Mar  2 23:49 start_x.elf
-rw-rw-rw- 1 root root  4857192 Mar  2 23:49 start_db.elf
-rw-rw-rw- 1 root root   688100 Mar  2 23:49 start_cd.elf
-rw-rw-rw- 1 root root  3536968 Mar  2 23:49 start4x.elf
-rw-rw-rw- 1 root root  4582856 Mar  2 23:49 start4db.elf
-rw-rw-rw- 1 root root   776256 Mar  2 23:49 start4cd.elf
-rw-rw-rw- 1 root root  2775396 Mar  2 23:49 start4.elf
-rw-rw-rw- 1 root root  2880420 Mar  2 23:49 start.elf
-rw-rw-rw- 1 root root  5751680 Mar  2 23:49 kernel7l.img
-rw-rw-rw- 1 root root  5418584 Mar  2 23:49 kernel7.img
-rw-rw-rw- 1 root root  5140112 Mar  2 23:49 kernel.img
-rw-rw-rw- 1 root root     9817 Mar  2 23:49 fixup_x.dat
-rw-rw-rw- 1 root root     9819 Mar  2 23:49 fixup_db.dat
-rw-rw-rw- 1 root root     2656 Mar  2 23:49 fixup_cd.dat
-rw-rw-rw- 1 root root     9185 Mar  2 23:49 fixup4x.dat
-rw-rw-rw- 1 root root     9183 Mar  2 23:49 fixup4db.dat
-rw-rw-rw- 1 root root     3097 Mar  2 23:49 fixup4cd.dat
-rw-rw-rw- 1 root root     6195 Mar  2 23:49 fixup4.dat
-rw-rw-rw- 1 root root     6745 Mar  2 23:49 fixup.dat
-rw-rw-rw- 1 root root    52304 Mar  2 23:49 bootcode.bin
-rw-rw-rw- 1 root root    40697 Mar  2 23:49 bcm2711-rpi-4-b.dtb
-rw-rw-rw- 1 root root    25249 Mar  2 23:49 bcm2710-rpi-cm3.dtb
-rw-rw-rw- 1 root root    26435 Mar  2 23:49 bcm2710-rpi-3-b.dtb
-rw-rw-rw- 1 root root    27054 Mar  2 23:49 bcm2710-rpi-3-b-plus.dtb
-rw-rw-rw- 1 root root    25394 Mar  2 23:49 bcm2710-rpi-2-b.dtb
-rw-rw-rw- 1 root root    25265 Mar  2 23:49 bcm2709-rpi-2-b.dtb
-rw-rw-rw- 1 root root    23643 Mar  2 23:49 bcm2708-rpi-zero.dtb
-rw-rw-rw- 1 root root    24379 Mar  2 23:49 bcm2708-rpi-zero-w.dtb
-rw-rw-rw- 1 root root    23719 Mar  2 23:49 bcm2708-rpi-cm.dtb
-rw-rw-rw- 1 root root    23938 Mar  2 23:49 bcm2708-rpi-b.dtb
-rw-rw-rw- 1 root root    24201 Mar  2 23:49 bcm2708-rpi-b-plus.dtb
-rw-rw-rw- 1 root root      334 Mar  2 23:50 cmdline.txt
-rw-rw-rw- 1 root root 19920866 Mar  2 23:56 volumio.initrd
drwxrwxrwx 2 root root      512 Mar 12 18:52 System Volume Information
-rw-rw-rw- 1 root root       90 Mar 12 18:55 userconfig.txt
drwxrwxrwx 2 root root    15872 Mar 12 20:09 overlays
-rw-rw-rw- 1 root root      398 Mar 12 20:09 config.txt
-rw-rw-rw- 1 root root  1820424 Mar 12 20:38 vmlinuz-3.6-trunk-rpi
-rw-rw-rw- 1 root root    75818 Mar 12 20:38 config-3.6-trunk-rpi
volumio@volumio-study:~$

rawdevice, 2020-07-09:
Hi Tom4C,

Were you ever able to resolve this?  I'm in a similar situation - Hyperpixel 4.0 touch on a Pi 3B+ Volumio 2.779.   have the display at -90 degrees (portrait) but the touch is ass-backward.  I actually need to rotate to landscape (with the micro-usb power port on top)  but so far no luck in rotating the display or the touch overlay.

RawDevice

closed 2026-08-25 by Gadgetoid