pimoroni/hyperpixel4 #59: FAQ: RPi4 Rect Touch not working out of box

neozenith, 2020-03-09:
**Describe the bug**

After running the one-liner and rebooting I'm not getting the screen to come up. 

It is backlit but no picture.

**To Reproduce**

I have a RPi3B+ and RPi4. 

```bash
sudo apt-get update
sudo apt-get upgrade -y
sudo apt dist-upgrade -y
sudo apt autoremove -y
``` 

I used the one-liner from the README and chose the correct option for each of my device platforms.

**Solutions:**
For both of them I had to edit `/boot/config.txt`.

```bash
sudo vim /boot/config.txt
```

**RPi3**: 
Uncomment the following:
`hdmi_safe=1`

**RPi4**:
According to this link: 
https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/

It would seem I had some rogue settings I needed to comment out from a previous project I used the board for that the installer missed warning about.

```ini
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on
```

Since the HyperPixel needs all GPIO these should have been commented out so the HyperPixel can self manage.

```ini
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
```

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular? **Rectangular**
2. Touch or non Touch? **Touch**
3. Approximately when was it purchased. **March-2020**
4. How is it connected to your Pi? **Attached to GPIO header with riser and screwed in with stand offs.**

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly: **`c03111`**
2. Which OS release are you using: 
 - `lsb_release --description`: **`Raspbian GNU/Linux 10 (buster)`**
 - `uname -r`: **`4.19.97-v7l`**
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular
* `dmesg | grep ft5` for HyperPixel 4.0" Square

And check i2c is working:

`ls /dev/i2c-*`

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Gadgetoid, 2020-03-09:
Insightful stuff- thank you! Will hopefully get a moment to look into/test this soon.

neozenith, 2020-03-09:
Thanks. I got mine working. But wanted to post this so that there is some documentation on troubleshooting the config.txt

Totally happy for you to close this issue as resolved. 

Or if there is a FAQ or docs page to add the DPI link I can open a pull request. I will need a hand pointing me to the right spot though.

Gadgetoid, 2020-03-10:
Thank you!

I have a "notice" label for posts that don't constitute an open issue, but that still warrant being visible to users. Happy to leave this open and tagged as such for now.

HyperPixel4, despite its apparent simplicity, has probably been one of the most difficult to support products we've made (I mean there are currently 3 active variations on the software for rectangular alone!) so I'm always eager for input and guidance where possible.

Perhaps some of this can be distilled into the main README.

Footleg, 2020-05-05:
Thank you. I ran into this same issue, except my hyperpixel4 touch worked on a Pi3b+ without needing to uncomment in the hdmi_safe option. On the Pi4 where I had installed the software first, I just got a blank hyperpixel screen with the backlight on (but lcd black). After testing on the Pi3 I found this post and sure enough my config.txt had the same options set:
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

Commenting all out and then running the hyperpixel4 install for Pi4, before shutting down and moving display and SD Card over to the Pi4 got just the Raspbian desktop splash screen to appear. Turned out this was because of some baggage left over from the Pi3 install of hyperpixel. I SSHed in and ran the uninstall for pi3, then the install script for pi4 again. On next reboot it all came up good apart from needing to rotate the display again using the hyperpixel4-rotate command. All working now.

My initial SD Card image was not a clean Raspbian, but a Buster image which I had then updated a few times and installed a few software packages on for my code club. Nothing exotic. Not clear if this was what set those options in my config.txt, or maybe me turning on a few things in raspi-config? I have an archive of my starting SD Card img if you need it.

closed 2026-08-25 by Gadgetoid