pimoroni/hyperpixel4 #60: Rotate "Inverted" not rotating touch capabilities

finmacd, 2020-03-10:
When I use the rotate command on my Pi4 "hyperpixel4-rotate inverted" the touch capabilities do not rotate, meaning that when I touch the top left of the screen, the cursor goes to the bottom right. Using the weirdly square hyperpixel 4 with raspbian buster full. Is there a solution for this?

Gadgetoid, 2020-03-10:
`hyperpixel4-rotate` doesn't currently support square- I should probably have deleted it from the repository to make that clear. I mean... it's square 😆 

That said, I don't see why it *couldn't*- you could try changing the `DEVICE=` line to the correct touchscreen device (find with `xinput --list`) and running it again.

https://github.com/pimoroni/hyperpixel4/blob/eebabef1c52e4e7ec5bf7944c7993624131abd13/dist/hyperpixel4-rotate#L4

Gadgetoid, 2020-03-10:
Okay, should be fixed in https://github.com/pimoroni/hyperpixel4/commit/a35de83bdd368f12e197970315cdae0c53f1442e

finmacd, 2020-03-10:
Great, working perfectly now. Had to rotate the orientation due to having a confined housing meaning that my power port had to face a specific direction. 

Thanks for the fix!

closed 2020-03-10 by finmacd

haukesand, 2020-06-16:
Fix does not seem to be working on my pi. Screen inverts but touch is not.

haukesand, 2020-06-16:
I manually cloned the folder instead of using the one-liner install script. rotation and touch work than until reboot