pimoroni/hyperpixel4 #63: Unable to use the provided I2C bus

varzenbot, 2020-03-28:
**Describe the bug**

I've ran the `sudo ln -s /dev/i2c-3 /dev/i2c-1` command 

```
pi@hyper:~ $ ls -t /dev/*i2c*
/dev/i2c-1  /dev/i2c-3
```
The devices detected are:
```
pi@hyper:~ $ i2cdetect -l
i2c-3	i2c       	i2c@0                           	I2C adapter
```
The address seems to be just 0x5D 
```
pi@hyper:~ $ i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

What's gone wrong?

**To Reproduce**

I've used the one line installer

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Rectangular
2. Touch
3. ~ 15th of Februrary
4. Via the gpio header

**Your Raspberry Pi**

1. `Revision	: 9020e0`
2. `Description:	Raspbian GNU/Linux 10 (buster)`
and 
`4.19.97-v7+`
3. Fresh buster install - followed by the display one-liner and then update, upgrade, dist-upgrade

**Extra debugging information**

The display and touch work fine, I am just unable to use the bus in order to connect to an Arduino/Teensy slave in order to communicate ( the uC is connected to several other inputs and should send their status once that is requested ).

My python code is as follows:

```
from smbus import SMBus

addr = 0x5d

bus =SMBus(1)

test_value = bus.read_byte(addr)
print(test_value)
```

The output is this:
```
Traceback (most recent call last):
  File "i2c_test.py", line 7, in <module>
    test_value = bus.read_byte(addr)
OSError: [Errno 16] Device or resource busy
```
Am I doing something wrong?
The touchscreen is not mandatory, but nice to have.
If I need to disable the touchscreen in order to communicate with my uC, I won't get mad.

Thank you in advance!

Gadgetoid, 2020-03-30:
How are you connecting your uC? `i2c-1` is completely disabled when using HyperPixel. Not because it's used for the touchscreen, but because those pins are used for the essential VSync and HSync signals.

HyperPixel sets up a software i2c bus for communicating with the touch interface (which is what you're seeing marked as UU, because the kernel touch driver reserves that address).

You should:

* Connect your device to the i2c breakout garden header on HyperPixel4
* use `SMBus(3`)`
* Find the correct address of your device, it's *not* the touch chip address

If you're not seeing your device when you `sudo i2cdetect -y 3` then it's not working, or not wired up correctly.

If you haven't already- I'd strongly recommend removing HyperPixel4 and all configuration from the equation, and verifying that your device works as expected under normal conditions.

varzenbot, 2020-04-01:
I've moved my teensy to an arduino uno, and managed to connect the two. even though I've had some issues with the initial code.
After moving the teensy back to the rpi, It still doesn't detect it, I'll do some more tests in the upcoming days and come back with an update, maybe it will help someone else who hits my issue.

please note that this is my first attempt at working with I2C.
One question: if my teensy (master) only requests/sends data when a button is pressed, do I have to keep that button pressed when running the i2cdetect command? or should I only press one button and then run the command? because it doesn't register any device when no button is pressed, and when the button is pressed, the i2cdetect command runs really slow

varzenbot, 2020-05-07:
Hi again, haven't managed to connect to the display.
I have tried different pi's ( 3B+, 3A+, I will try a 4 next ), and I still can't seem to use the I2C.

Does anyone have an answer for my previous question ?

Gadgetoid, 2020-07-31:
I'm afraid you lost me with this, so sorry for the late reply- if you're attempting to set up a microcontroller as an i2c device then you should remove the display from the equation and test with the Pi's normal i2c interface. Once you get that working, then you can bring HyperPixel 4 back into the equation and make the next step.

closed 2020-07-31 by Gadgetoid