pimoroni/hyperpixel4 #65: Hyperpixel4 STAYING in Portrait..... PLZ HELP !! :)

Jeffmc1, 2020-04-26:
I have a hyperpixel4 I bought a few mos ago..
Using a Pi4
Running full version of raspbian
After raspbian loaded I went to PREFS and set it to ROTATE screen... that worked for raspbian no problem.

Then I loaded using the console... PIHOLE
After that I loaded PADD, its a CONSOLE Window that comes up automatically showing me details.... (PIC BELOW) it is in PORTRAIT mode,,,, I can not get it to go into LANDSCAPE mode... this is what the problem I am having is.....

Ive tried the hyperpixel4rotate left
DIDNT WORK ! ;(

Im so bummed and frustrated... Dont know anything about Linux etc... but I got this far all working cept the display....

[image]

Gadgetoid, 2020-04-27:
This is a limitation of the Raspberry Pi 4 and how rotation is accomplished. I believe the `/boot/config.txt` settings mentioned here may work: https://github.com/pimoroni/hyperpixel4/issues/39#issuecomment-552190708

Pay particular attention to the `dtoverlay=vc4-fkms-v3d` line(s), which must be commented out with a preceeding `#` or it breaks rotation in horrible ways.

Jeffmc1, 2020-04-29:
> work: #39 (comment)

CRAP, Ughhhh, im really beginning to HATE this HYPERPIXEL 4, not working
right... I did the CODE in the confix like you posted and it FURTHER BROKE
my pi4.....

I know NOTHING about LINUX, here is the edited config.txt can you help me ?
I did the CODE at the END of the config and rebooted and the stupid pi4
just comes up with a pi desktop LOGO for raspbian and that is it.

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details
lcd_rotate=3
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if
console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size
minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work
in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking,
or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
dtoverlay=hyperpixel4-0x5d
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
display_rotate=1

-- 
*? ! @ # % *

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details
lcd_rotate=3
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
dtoverlay=hyperpixel4-0x5d
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
display_rotate=1

Gadgetoid, 2020-04-30:
You should drop an email to support@pimoroni.com and they should be able to guide you through this. Sorry to bounce you around, but GitHub issues are for reporting problems with the software and not for general support.

Cheers!

closed 2020-04-30 by Gadgetoid