pimoroni/hyperpixel4 #66: Display gets blurry

Matze121, 2020-04-28:
Hi,

If I use the display for some minutes it becomes blurry in the middle of the screen.
The picture shows the effect quite good.

I've added a 1MOhm resistor between 3v3 and GND as mentioned in a previous issue.
Otherwise the touch won't work after a while.

Is there a way to deal with this?

[image]

Gadgetoid, 2020-04-30:
Did you mean 3v3 and *Int* or 3v3 and *Gnd* since the latter might explain what's happening here.

Matze121, 2020-04-30:
Oh, of course 3v3 and Int is connected...

Gadgetoid, 2020-05-01:
Is that an official power supply you're using there? It could be a power issue.

I'll ask around our engineering and testing dept to see if they've seen something like this.

Matze121, 2020-05-01:
Thanks for your help!
I used a stable 5v/5a power supply.

Gadgetoid, 2020-05-15:
Unfortunately nobody has seen anything like this, excepting the effect you get when hot unplugging the power while running the screen.

I still strongly suspect some sort of power supply issue, since it's the wildcard here but I don't really have anything more substantial than a hunch to back that up. If you do have another supply to try, it may be worth a go.

If all else fails, ping the support team a message at support@pimoroni.com and they will- hopefully- be able to furbish you with a replacement.

98Matthias, 2020-05-15:
Ok, thanks! It‘s not so bad and you can recognize everything. That‘s enough for me.

closed 2020-05-15 by 98Matthias