pimoroni/hyperpixel4 #67: Touch not working on Pi 3B

luastoned, 2020-04-29:
**Describe the bug**

The I2C connection does not appear to work.

**To Reproduce**

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?

Installed via the one-line-installer.

**Your HyperPixel 4**

1. Is it Square or Rectangular? **Rectangular**
2. Touch or non Touch? **Touch**
3. Approximately when was it purchased? **June 2019**
4. How is it connected to your Pi? **Directly**

**Your Raspberry Pi**

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
2. Which OS release are you using: `lsb_release --description` and `uname -r`
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?

```
1. Revision: a02082
2. Description: Raspbian GNU/Linux 10 (buster)
3. 4.19.97-v7+
```

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

```
[    2.968454] i2c /dev entries driver
[    4.275606] gpio-10 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    4.275681] gpio-11 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    4.276348] i2c-gpio i2c@0: using lines 10 (SDA) and 11 (SCL)
[    5.279616] Goodix-TS 3-005d: i2c test failed attempt 1: -6
[    5.320321] Goodix-TS 3-005d: i2c test failed attempt 2: -6
```

Additional info:

```
18:19:52 pi@raspberrypi:/code > ls /dev/i2c-*
/dev/i2c-3
```

```18:21:02 pi@raspberrypi:/code > i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Gadgetoid, 2020-04-30:
Thanks for the super thorough issue report!

Looks like you've had it a while, has it ever worked or did you only just find time to try it?

At face value it looks like a total comms failure with the touch driver chip. Do you have any entries in `/boot/config.txt` or other software installed/running that might be interfering with pins BCM10 and BCM11?

luastoned, 2020-04-30:
I tried to use it last fall but the support for Buster was not there yet I think, so I just wrote it off for the time being.

```
14:45:44 pi@raspberrypi:/code> cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
enable_dpi_lcd=1
display_default_lcd=1
display_rotate=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

It's a fresh install, interestingly enough `i2cdetect -y 3` now yields a different output:

```
14:53:09 pi@pideck:/code> i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

luastoned, 2020-04-30:
After rebooting a couple of times:

```
pi@raspberrypi:/code $ dmesg | grep i2c
[    5.778133] i2c /dev entries driver
[   12.020072] gpio-10 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[   12.020185] gpio-11 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[   12.027356] i2c-gpio i2c@0: using lines 10 (SDA) and 11 (SCL)
[   23.449973] Goodix-TS 3-005d: i2c test failed attempt 1: -6
[   23.493254] Goodix-TS 3-005d: i2c test failed attempt 2: -6
```

```
pi@raspberrypi:/code $ i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- 5d -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

thirdr, 2025-01-08:
Closing this to clear our decks for the new year, if it's still troubling you please re-open! :)

closed 2025-01-08 by thirdr