pimoroni/hyperpixel4 #70: Is it possible to right click?

fauno, 2020-05-01:
I've been looking around but couldn't find anything.  Is it possible to right click (to open menus, for instance)?  I've found projects like [twofing](https://github.com/Plippo/twofing) and [fusuma](https://github.com/iberianpig/fusuma/) but the first one doesn't have an udev rule for the touchpad and fusuma starts but fails to read gestures from the touchpad as soon as I touch the screen.

fauno, 2020-05-01:
I guess the actual questions would be:

* What kind of gestures does the Goodix touchpad support?

* What would be needed to make those projects support the touchpad?

Gadgetoid, 2021-05-20:
It's a touchscreen so unlike a touchpad gesture support is implemented in the OS layer and is... from experience... totally arbitrary and application specific. IE: a browser might support two finger scroll, but this will have no effect anywhere else in the OS.

"twofing" used to work for me, but I don't know how well it plays with Raspberry Pi OS these days. The udev rules don't do anything special and you can just throw together your own. Something like (the touchscreen name may be wrong here):

```
KERNEL=="event*",ATTRS{name}=="Goodix Touchscreen",SYMLINK+="twofingtouch",RUN+="/bin/chmod a+r /dev/twofingtouch"
```

closed 2021-05-20 by Gadgetoid