pimoroni/hyperpixel4 #72: Touchscreen inversion issue on a fresh Rasbian/hyperpixel4 installation.

lotek-rng, 2020-05-07:
**Describe the bug**

What's gone wrong?
_**-I installed a fresh version of raspbian-buster on my rpi4 and i installed the drivers specifically for the rpi4, and im having some issues with the touchscreen. In the default orientation if i move my finger from left to right the cursor travels bottom to top. Also, if i move my finger top to bottom it moves right to left Also, i don't have a minihdmi yet(although one's coming in the mail) so i added a ssh and wpa_supplicant.conf file to the boot directory. I installed the drivers over ssh**_

**To Reproduce**

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?
_**-I used the one line installer and nothing else. I eventually want to rotate it to the right, but first i need to figure out what the problem is in it's default orientation.**_
**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?
_**-Square**_
2. Touch or non Touch?
_**-Touch**_
3. Approximately when was it purchased
_**-About 7ish months ago. Now i finally have a use for it, and im having this issue.**_
4. How is it connected to your Pi
_**-Standard GPIO connection with riser screws.**_
**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
_**c03111 rpi4 4gb model**_
2. Which OS release are you using: `lsb_release --description` and `uname -r`
_**- Raspbian GNU/Linux 10 (buster) - 4.19.97-v7l+**_
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
_**-Completely stock setup. The only thing i did was include a ssh and wpa_supplicant.conf file to the boot directory, as i have no hdmi display connected to it.**_
**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular

**_[    4.256205] Goodix-TS 7-0014: ID 911, version: 1060
[    4.277945] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-7/7-0014/input/input1_**

* `dmesg | grep ft5` for HyperPixel 4.0" Square

And check i2c is working:

`ls /dev/i2c-*`

**_/dev/i2c-7_**

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
_**-My i2c shows exactly what's above**_

_**Thanks <3**_

EtoTen, 2020-05-26:
Just tried with a rpi4, same exact behaviour with a fresh install.

68gt500, 2020-06-05:
Exactly same issue here -  rpi4B 2GB Raspian Buster --> 4.19.97-v7l+

Please Pimoroni - HELP!

closed 2026-08-25 by Gadgetoid