pimoroni/hyperpixel4 #73: OS Support

bretonics, 2020-05-09:
There is no indication on product descriptions about the HyperPixel 4.0 support of OS, but looking at the source code for the one-line-installer (as I just bought 2 and was about to run the install), it looks like this only supports Raspbian (likely all other Debian distros?). Is this correct?

If so, is there or will there be support for other operating systems like CentOS?

Otherwise, recommendations for install on other Linux distros?

Gadgetoid, 2020-05-15:
See: https://github.com/pimoroni/hyperpixel4/issues/21

We should probably specify in the description that only Raspbian is supported for the one-line installer.

Other operating systems tend to work, but with mixed results. We're ultimately at the behest of how the videocore handles the DPI interface, and how X handles rotation and touch input. It's... a nightmare.

In your case if you're running a Pi 4 you should be able to install the dtoverlays and relevant config.txt entries to get up and running.

We *really* need to gets some step-by-step instructions for this, but broadly:

Use the `dtc` tool to compile `hyperpixel4.dts` from https://github.com/pimoroni/hyperpixel4/tree/pi4/src and place it into `/boot/overlays/`:

```
dtc -I dts -O dtb -Wno-unit_address_vs_reg -o hyperpixel4.dtbo hyperpixel4.dts
cp hyperpixel4.dtbo /boot/overlays
```

Then ensure the following lines are added to `/boot/config.txt`:

```
dtoverlay=hyperpixel4
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

KodeIn, 2020-06-16:
On a Hyperpixel 4.0 square, I had to create a 01-touchscreen.conf in xorg.conf.d to get the touchscreen in the same orientation as the screen :
```
Section "InputClass"
   Identifier "touchscreen"
   MatchIsTouchScreen "true"
   Option "TransformationMatrix" "-1 0 1 0 -1 1 0 0 1"
EndSection
```

closed 2020-07-15 by Gadgetoid

coreybruce, 2021-02-06:
> dtc -I dts -O dtb -Wno-unit_address_vs_reg -o hyperpixel4.dtbo hyperpixel4.dts

I ran into this issue 
```FATAL ERROR: Couldn't open "hyperpixel4.dts": No such file or directory```