pimoroni/hyperpixel4 #74: NanoPi M4v2/RK3399 Driver

3s10, 2020-05-13:
Is it possible to have a branch for the NanoPi M4v2? It has a Mali-T864 GPU (it supports OpenGL® ES1.1/2.0/3.0/3.1, OpenCL™, DX11, and AFBC) and Rock Chip RK3399 CPU. It also has a 40 Pin GPIO header. The HyperPixel 4.0 is a great pairing with the RK3399 boards as they are both attractive to people looking for a stronger, higher end SBC experience. When I tried to install the drivers for the RPI 3 it gives an error that the chip is not supported.

Gadgetoid, 2020-05-15:
I'm afraid I can only support the Raspberry Pi here- there simply aren't enough hours in the day to open the floodgates to other devices.

In my experience, precious few of these 40-pin headers have a 1:1 match to the Pi's GPIO anyway, and I'd be surprised if it had a matching DPi interface. The M4v2, for example, makes no mention of supporting DPI out and has no documentation on its existence. It might be *theoretically* possible to bitbang it in software, but I don't see the result being useful.

closed 2020-05-15 by Gadgetoid

3s10, 2020-05-15:
I completely understand, thank you so much!