pimoroni/hyperpixel4 #75: screen not working at all

AeyeBALL, 2020-05-23:
I have just received my hyperpixle 4.0 and also A brand new raspberry pi 4 model B. I have a brand new image of raspbian installed. I have the non-touch version of the rectangle display. The screed does not display anything but the backlight turns on. I tried to install the drivers using the one-line installer and it has not worked. I also tried to install for the pi 4 specifically but that also does not work. I can't figure it out for the life of me.

Gadgetoid, 2020-05-25:
Does the one-line installer fail with any errors at all?

AeyeBALL, 2020-05-25:
No, both installers don’t fail. I have plugged in a second monitor to one of the HDMI ports and it considers the HDMI to be a second screen so it’s registering it as a screen.

Gadgetoid, 2020-05-26:
Curious. PCB pushed firmly down onto the Pi's GPIO?

Any other software running at all?

This may be a question for support@pimoroni.com since they've got some experience diagnosing issues and will be empowed to sort you out with a replacement if it's broken.

jpreculeau, 2020-05-27:
Same here. I managed to get screen work but the touch screen was not right (and not possible to get it work properly). Since then I've reinstall a clean octopi and now I can get this screen to work.

AeyeBALL, 2020-05-27:
No, it's a brand new image of raspbian, and I pushed it down all the way. I will also see if support@pimoroni.com knows whats up.

AeyeBALL, 2020-06-15:
Nothing seemed to work so i am going to try to return it.

closed 2021-05-20 by Gadgetoid