pimoroni/hyperpixel4 #76: Camera preview not rotating with display

seanpcook, 2020-05-25:
Now that I have the display in landscape and touch controls are working well I want to be able to preview the camera preview in landscape as well. With these new settings it previews off axis by 90 degrees as it still thinks 0,0 is the original corner. The -roi camera command is not good for this because it rotates the actual image. Any idea if there is a fix for this or where I could look for more information?

Gadgetoid, 2020-05-26:
I'm afraid I don't know- I guess you're using a Pi 4?

I suspect the camera overlay is generated by the Videocore GPU, whereas the rotation of the desktop is happening in software. This combination means that neither has any awareness of what rotation the other is in.

This might be something worth raising upstream against the relevant Raspberry Pi software project, but I'm not entirely sure which that might be.

Disabling `dtoverlay=vc4-fkms-v3d` in `/boot/config.txt` *might* help. This breaks the normal method of display rotation, though, so you will additionally have to use `display_rotate`.

seanpcook, 2020-05-26:
Yes, sorry for my noobness and not posting details about my devices. And thank you for all the hard work you put into figuring out the landscape display and touch. When I finally got it working properly it was exciting. 

This is my first adventure into Raspberry Pi so please excuse anything that doesn’t make sense.  I’m using a RPi 4b with the typical full OS installation. I’ve got the HQ camera attached with the ribbon and the HyperPixel 4 on the GPIO. I see what your saying about the the difference in how the display can be rotated and will try your suggestion.

Gadgetoid, 2021-05-20:
Closing stale issue. Feel free to re-open if you're still having problems.

closed 2021-05-20 by Gadgetoid