pimoroni/hyperpixel4 #77: Unable to get correct orientation for touch matrix on RPI4

jpreculeau, 2020-05-28:
**Describe the bug**
Unable to get touch screen working

What's gone wrong?
First : in order to have the right orientation I have to modify /boot/config.txt this way :
display_rotate=1 
this ligne dtoverlay=hyperpixel4:rotate_1 never worked. This "hyperpixel4-rotate right" also never worked. Even with the notice solution with sudo nano /etc/rc.local and at the very bottom adding the following line /usr/bin/hyperpixel4-rotate right. 
After that I get a working screen with correct orientation but touch panel is reverse on Y. Not always. If i boot It could be reverse on both axis.
I get it to work properly only one time (until shutdown the RPI) with this configuration on config.txt
dtoverlay=hyperpixel4:touchscreen-swapped-x-y
but with this exact same configuration I get a reverse Y touch panel.

**To Reproduce**

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?
Install on an Octopi 1.7.0 distribution with one-line installer (same with installation from RPI4 branch)

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular? Rectangular
2. Touch or non Touch? Touch
3. Approximately when was it purchased : 1 week ago
4. How is it connected to your Pi : GPIO with you adapter

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
Revision        : c03112

2. Which OS release are you using: `lsb_release --description` and `uname -r`
Description:    Raspbian GNU/Linux 10 (buster)
4.19.118-v7l+

3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
Octoprint was running since a month on a fresh install.
Followed an tutorial to add Octoscreen app with this very screen on RPI4.

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular
[    4.350738] Goodix-TS 7-005d: i2c test failed attempt 1: -6
[    4.389922] Goodix-TS 7-005d: i2c test failed attempt 2: -6
[    4.427776] Goodix-TS 7-005d: I2C communication failure: -6
[    4.429778] Goodix-TS 7-0014: ID 911, version: 1060
[    4.450823] Goodix-TS 7-0014: Checking firmware
[    4.451068] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-7/7-0014/input/input0

And check i2c is working:

`ls /dev/i2c-*`
got : /dev/i2c-7

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
[config.txt](https://github.com/pimoroni/hyperpixel4/files/4697285/config.txt)

jpreculeau, 2020-05-29:
So I've try on another sdcard a fresh raspbian install the largest one desktop + app.
install from unique command line with option 1.
I does not work either. "sudo hyperpixel4-rotate right" do nothing. I manage to have a correct orientation if I put display_rotate=1 in config.txt but then the touch is not aligned.
Will try again some other method described here and there on this github.

Gadgetoid, 2021-05-20:
Any luck? Sorry it's been quite a year and I haven't been able to get on top of GitHub issues at all :grimacing: