pimoroni/hyperpixel4 #79: Burn in effect on Pi weirdly square.

haukesand, 2020-06-15:
We have three brand new square hyperpixels with the latest raspbian installed. 

Drivers are installed via the install script for raspberry pi4. We use official power supply. 

Quickly after some time at the edges colors degrade and change. 

I guess this is because the backlight does not turn off automatically? 

So far when powering off the pi completely it was fine at next bootup.

haukesand, 2020-06-15:
```

## Platform Information

Revision	: b03111
Description:	Raspbian GNU/Linux 10 (buster)
4.19.97-v7l+

## Touchscreen logs

Rectangular: Goodix
Square: ft5
[    4.138749] input: generic ft5x06 (11) as /devices/platform/i2c@0/i2c-7/7-0048/input/input7

## I2C Devices and Mappings

/dev/i2c-7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
50: -- -- -- 53 -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

## Boot Config

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
dtparam=spi=off

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
framebuffer_width=720
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f226
hdmi_timings=720 0 15 15 15 720 0 10 10 10 0 0 0 60 0 35113500 6
[all]
#dtoverlay=vc4-kms-v3d

dpi_timings=720 0 15 15 15 720 0 10 10 10 0 0 0 60 0 35113500 6
```

Gadgetoid, 2020-06-26:
We believe this is because the screen is not being shut down when not being actively drive - ie: are in "sleep" mode - and are currently looking into solutions. For some reason this seems to have become considerably more pronounced with recent OS versions. It's like looking at a vintage photograph!

haukesand, 2020-06-29:
I can confirm that it happens when the screen blanks / goes to sleep.

eliliam, 2023-05-30:
Bumping this as I am experiencing similar issues. Any update on how to counteract this or prevent further burn in?

closed 2026-08-25 by Gadgetoid